Ysurac/openmptcprouter #1200: Glorytun not available in VPN menu un Wizard

matthieu71000, 2020-09-04:
## Expected Behavior
After 0.55.3 update, Glorytun not available in VPN menu un Wizard

## Context (Environment)
Same Problem With RPI3 and X86_64

## Specifications

  - OpenMPTCProuter version: 0.55.3
  - OpenMPTCProuter VPS version: 0.1017 5.4.52-mptcp
  - OpenMPTCProuter platform: x86_64 or RPI3

Thanks

Ysurac, 2020-09-04:
What VPNs are available?
You tested on 2 platforms ? In this case, what do you have in ```/var/log/daemon.log``` on the VPS ?

matthieu71000, 2020-09-04:
A Dead Simple VPN and OpenVPN TCP are available.

On 2 platforms after restore my backup config.

If I don't restore backup Glorytun is available.

matthieu71000, 2020-09-04:
No daemon.log file in /var/log/

Ysurac, 2020-09-04:
Debian is not used on the VPS ? Ubuntu ?

matthieu71000, 2020-09-04:
Sorry "on the VPS"

Sep  4 13:24:01 vps683897 ss-server[465]: remote recv: Connection reset by peer
Sep  4 13:24:03 vps683897 ss-server[465]: remote recv: Connection reset by peer
Sep  4 13:24:04 vps683897 ss-server[465]: remote recv: Connection reset by peer
Sep  4 13:24:08 vps683897 ss-server[465]: remote recv: Connection reset by peer
Sep  4 13:25:11 vps683897 ss-server[465]: getpeername: Transport endpoint is not connected
Sep  4 13:25:13 vps683897 ss-server[465]: getpeername: Transport endpoint is not connected
Sep  4 13:25:14 vps683897 ss-server[465]: getpeername: Transport endpoint is not connected
Sep  4 13:26:43 vps683897 ss-server[465]: getpeername: Transport endpoint is not connected
Sep  4 13:28:16 vps683897 ss-server[465]: getpeername: Transport endpoint is not connected
Sep  4 13:29:46 vps683897 ss-server[465]: getpeername: Transport endpoint is not connected
Sep  4 13:30:15 vps683897 ss-server[465]: getpeername: Transport endpoint is not connected
Sep  4 13:30:16 vps683897 ss-server[465]: getpeername: Transport endpoint is not connected
Sep  4 13:31:18 vps683897 ss-server[465]: getpeername: Transport endpoint is not connected
Sep  4 13:32:50 vps683897 ss-server[465]: getpeername: Transport endpoint is not connected
Sep  4 13:34:22 vps683897 ss-server[465]: getpeername: Transport endpoint is not connected

matthieu71000, 2020-09-04:
Is Glorytun only available after a first Wizard ?

Ysurac, 2020-09-04:
The log is a bit short, I would like log related to omr-admin.
After first run, Glorytun is available when set as available by the VPS. You can try to validate again the wizard with another VPS to retrieve updated VPN list.

matthieu71000, 2020-09-04:
After Wizard with OpenVPN still no glorytun in wizard

Sep  4 14:04:55 vps683897 ss-server[465]: getpeername: Transport endpoint is not connected
Sep  4 14:05:04 vps683897 ss-server[465]: remote recv: Connection reset by peer
Sep  4 14:05:31 vps683897 ss-server[465]: getpeername: Transport endpoint is not connected
Sep  4 14:05:32 vps683897 ss-server[465]: getpeername: Transport endpoint is not connected
Sep  4 14:05:43 vps683897 ovpn-tun0[475]: TCP connection established with [AF_INET]37.164.44.220:19574
Sep  4 14:05:43 vps683897 ovpn-tun0[475]: 37.164.44.220:19574 TLS: Initial packet from [AF_INET]37.164.44.220:19574, sid=0a8962c9 58f93c03
Sep  4 14:05:43 vps683897 ovpn-tun0[475]: 37.164.44.220:19574 WARNING: Failed to stat CRL file, not (re)loading CRL.
Sep  4 14:05:44 vps683897 ovpn-tun0[475]: 37.164.44.220:19574 VERIFY OK: depth=1, CN=ChangeMe
Sep  4 14:05:44 vps683897 ovpn-tun0[475]: 37.164.44.220:19574 VERIFY OK: depth=0, CN=openmptcprouter
Sep  4 14:05:44 vps683897 ovpn-tun0[475]: 37.164.44.220:19574 peer info: IV_VER=2.4.9
Sep  4 14:05:44 vps683897 ovpn-tun0[475]: 37.164.44.220:19574 peer info: IV_PLAT=linux
Sep  4 14:05:44 vps683897 ovpn-tun0[475]: 37.164.44.220:19574 peer info: IV_PROTO=2
Sep  4 14:05:44 vps683897 ovpn-tun0[475]: 37.164.44.220:19574 peer info: IV_LZ4=1
Sep  4 14:05:44 vps683897 ovpn-tun0[475]: 37.164.44.220:19574 peer info: IV_LZ4v2=1
Sep  4 14:05:44 vps683897 ovpn-tun0[475]: 37.164.44.220:19574 peer info: IV_LZO=1
Sep  4 14:05:44 vps683897 ovpn-tun0[475]: 37.164.44.220:19574 peer info: IV_COMP_STUB=1
Sep  4 14:05:44 vps683897 ovpn-tun0[475]: 37.164.44.220:19574 peer info: IV_COMP_STUBv2=1
Sep  4 14:05:44 vps683897 ovpn-tun0[475]: 37.164.44.220:19574 peer info: IV_TCPNL=1
Sep  4 14:05:44 vps683897 ovpn-tun0[475]: 37.164.44.220:19574 Outgoing Data Channel: Cipher 'AES-256-CBC' initialized with 256 bit key
Sep  4 14:05:44 vps683897 ovpn-tun0[475]: 37.164.44.220:19574 Outgoing Data Channel: Using 160 bit message hash 'SHA1' for HMAC authentication
Sep  4 14:05:44 vps683897 ovpn-tun0[475]: 37.164.44.220:19574 Incoming Data Channel: Cipher 'AES-256-CBC' initialized with 256 bit key
Sep  4 14:05:44 vps683897 ovpn-tun0[475]: 37.164.44.220:19574 Incoming Data Channel: Using 160 bit message hash 'SHA1' for HMAC authentication
Sep  4 14:05:44 vps683897 ovpn-tun0[475]: 37.164.44.220:19574 Control Channel: TLSv1.3, cipher TLSv1.3 TLS_AES_256_GCM_SHA384, 2048 bit RSA
Sep  4 14:05:44 vps683897 ovpn-tun0[475]: 37.164.44.220:19574 [openmptcprouter] Peer Connection Initiated with [AF_INET]37.164.44.220:19574
Sep  4 14:05:44 vps683897 ovpn-tun0[475]: openmptcprouter/37.164.44.220:19574 MULTI_sva: pool returned IPv4=10.255.252.6, IPv6=(Not enabled)
Sep  4 14:05:44 vps683897 ovpn-tun0[475]: openmptcprouter/37.164.44.220:19574 MULTI: Learn: 10.255.252.6 -> openmptcprouter/37.164.44.220:19574
Sep  4 14:05:44 vps683897 ovpn-tun0[475]: openmptcprouter/37.164.44.220:19574 MULTI: primary virtual IP for openmptcprouter/37.164.44.220:19574: 10.255.252.6
Sep  4 14:05:45 vps683897 ovpn-tun0[475]: openmptcprouter/37.164.44.220:19574 PUSH: Received control message: 'PUSH_REQUEST'
Sep  4 14:05:45 vps683897 ovpn-tun0[475]: openmptcprouter/37.164.44.220:19574 SENT CONTROL [openmptcprouter]: 'PUSH_REPLY,route 10.255.252.1 255.255.255.255,route 10.255.252.1,topology net30,ping 10,ping-restart 240,ifconfig 10.255.252.6 10.255.252.5,peer-id 0' (status=1)
Sep  4 14:05:51 vps683897 ss-server[465]: remote recv: Connection reset by peer
Sep  4 14:05:52 vps683897 systemd[1]: Stopping OMR6IN4 on user0...
Sep  4 14:05:52 vps683897 systemd-networkd[213]: omr-6in4-user0: Lost carrier
Sep  4 14:05:52 vps683897 systemd[1]: omr6in4@user0.service: Succeeded.
Sep  4 14:05:52 vps683897 systemd[1]: Stopped OMR6IN4 on user0.
Sep  4 14:05:52 vps683897 systemd[1]: Starting OMR6IN4 on user0...
Sep  4 14:05:53 vps683897 systemd-networkd[213]: omr-6in4-user0: Gained IPv6LL
Sep  4 14:05:53 vps683897 systemd-networkd[213]: omr-6in4-user0: Gained carrier
Sep  4 14:05:53 vps683897 systemd[1]: Started OMR6IN4 on user0.
Sep  4 14:05:53 vps683897 systemd-udevd[29938]: link_config: autonegotiation is unset or enabled, the speed and duplex are not writable.
Sep  4 14:05:53 vps683897 systemd[1]: Reloading Shorewall IPv4 firewall.
Sep  4 14:05:54 vps683897 shorewall[29953]: Compiling using Shorewall 5.2.3.2...
Sep  4 14:06:12 vps683897 omr-service[464]: Error: Nexthop has invalid gateway.
Sep  4 14:06:18 vps683897 shorewall[29953]: Processing /etc/shorewall/params ...
Sep  4 14:06:18 vps683897 shorewall[29953]: Processing /etc/shorewall/shorewall.conf...
Sep  4 14:06:18 vps683897 shorewall[29953]: Loading Modules...
Sep  4 14:06:19 vps683897 shorewall[29953]: Compiling /etc/shorewall/zones...
Sep  4 14:06:19 vps683897 shorewall[29953]: Compiling /etc/shorewall/interfaces...
Sep  4 14:06:19 vps683897 shorewall[29953]: Determining Hosts in Zones...
Sep  4 14:06:19 vps683897 shorewall[29953]: Locating Action Files...
Sep  4 14:06:19 vps683897 shorewall[29953]: Compiling /etc/shorewall/policy...
Sep  4 14:06:19 vps683897 shorewall[29953]: Adding Anti-smurf Rules
Sep  4 14:06:19 vps683897 shorewall[29953]: Adding rules for DHCP
Sep  4 14:06:19 vps683897 shorewall[29953]: Compiling TCP Flags filtering...
Sep  4 14:06:19 vps683897 shorewall[29953]: Compiling Kernel Route Filtering...
Sep  4 14:06:19 vps683897 shorewall[29953]: Compiling Martian Logging...
Sep  4 14:06:19 vps683897 shorewall[29953]: Compiling Accept Source Routing...
[... cut ...]
Sep  4 14:07:53 vps683897 omr-service[464]: Error: Nexthop has invalid gateway.
Sep  4 14:08:03 vps683897 omr-service[464]: Error: Nexthop has invalid gateway.
Sep  4 14:08:03 vps683897 ss-server[465]: remote recv: Connection reset by peer
Sep  4 14:08:14 vps683897 omr-service[464]: Error: Nexthop has invalid gateway.
Sep  4 14:08:24 vps683897 omr-service[464]: Error: Nexthop has invalid gateway.
Sep  4 14:08:34 vps683897 omr-service[464]: Error: Nexthop has invalid gateway.
Sep  4 14:08:44 vps683897 omr-service[464]: Error: Nexthop has invalid gateway.

Ysurac, 2020-09-04:
Strange...
You can, via SSH on the router, run ```uci delete openmptcprouter.vps.available_vpn && uci commit```, this should reset VPN list.

matthieu71000, 2020-09-04:
Parfait !!
Glorytun appeared !

Merci beaucoup / Thanks
Bon Week-end

closed 2020-09-04 by Ysurac