Ysurac/openmptcprouter #1201: OpenMPTCProuter v0.56 alpha

Ysurac, 2020-09-04:
This issue is opened for OpenMPTCProuter 0.56.

New features:
* V2Ray support for TCP and UDP (proxy can be selected in wizard between shadowsocks and v2ray)
* Domain support as VPS address

cr3ative, 2020-10-27:
Hello! The index page for me has two blank squares, and this file is missing when the page tries to load it - perhaps that's why?

`http://192.168.100.1/luci-static/resources/protocol/wwan.js`

Here is a screenshot:

[image]

Ysurac, 2020-10-27:
one square is for DHCP and the other is for wifi. They are blank when no data.
The wwan.js never existed, I have no idea why it's called (but it doesn't make any issue).

cr3ative, 2020-10-27:
Ah okay! That makes sense. Thank you for clarifying!

JeedHome44, 2020-10-29:
> Ah okay! That makes sense. Thank you for clarifying!

WARNING cr3ative! We can see your external IP adress to your VPS... Delete screenshot ;)

cr3ative, 2020-10-29:
> > Ah okay! That makes sense. Thank you for clarifying!
> 
> WARNING cr3ative! We can see your external IP adress to your VPS... Delete screenshot ;)

Ehhh, maybe someone will be my friend this way. ;)

JeedHome44, 2020-10-29:
is it used to switch to 0.56 if we have done 0.55 recently?

Ysurac, 2020-10-30:
Changes are available on https://www.openmptcprouter.com/

TalalMash, 2020-11-03:
I don't see that "MPTCP over VPN" OVPN protocol has been changed to udp from tcp, was it always udp? I've seen much better performance and no bufferbloat on heterogenous links with the latest release.

Ysurac, 2020-11-03:
MPTCP over VPN always used OpenVPN in UDP mode.

closed 2021-03-30 by Ysurac