Ysurac/openmptcprouter #1202: DPDK in a futur release ?

krissfr, 2020-09-04:
Hello Yannick,

Maybe you already know [DPDK](https://www.dpdk.org/) wich is an Open source software project managed by the Linux Foundation, it boost packet processing on many kind of CPU architecture (arm, x86, ..) and hypervisors.

they have a documentation to enable it in openwrt :
[https://doc.dpdk.org/guides/howto/openwrt.html](https://doc.dpdk.org/guides/howto/openwrt.html)

i think it could be awesome on our little OMR devices (and also on the VPS side).

we could expect: 
- less load on router and VPS
- better latency
- better thoughtput

Have you already did some test with DPDK ?

Did you plan to enable DPDK in a futur release ?

Thank you

Christophe

Adorfer, 2020-09-04:
sounds like spam to me. 
the is no any issue or proposal apart from _"hey look what we have got"_

my experience with fastd (userland) and userland-pppoe (freebsd) is that i do not like that approach due to lack of performance, since it is not disciplined by the kernel.

Ysurac, 2020-09-05:
From what I understand, DPDK is a framework and changes are needed on VPS/Proxy to use it.
I don't know if network drivers they provide can be used directly and can help for the router.
I will maybe test that if I can do a package to install it.

krissfr, 2020-09-05:
@Adorfer : i'm definitly not a spammer ;-)

I juste discover DPDK few days ago when i did search on SRIOV .

I'm currently limited by my Proxmox machine and i can't have more than 1Gb between my LAN and my WAN1 and WAN2 network with OpenMPTCPRouter (my Freebox Delta in WAN1 is also a NAS with SSDs in Raid10 and i would have more than 1 Gbps for file transferts between WAN1 and LAN).

I ordered somes news hardware components which are SRIOV capable. So i will expose a Virtual Functions of my 10Gbps NIC to my OMR Virtual machine NIC. That for the hidden part (from OMR point of view).

Beside that i discover DPDK wich can give also more gain inside the Virtual Machine (and for baremetal installation too). During my search i discover that the usage of DPDK in Network software ( ie in OpenVSwitch ) is growing up. So thats why i talk about it here.

@Ysurac : DPDK could be install only in the OMR part (for my use case, transferts between my Freebox Delta in WAN1 and my LAN). I think that the installation on the VPS side will benefit only for the VPN traffic. 

I really don't know if it's just install and play or if it necessite to adapt many things. But as i saw some people trying to install it for Openwrt, i thought that maybe there is no need to made lot of change.

When i will receive my new parts (intel X520-DA2) i will try with the SRIOV optimisation, maybe it will be enought for my use case.
I'm just curious about the gain that DPDK could give to OMR, so if you test @Ysurac  , please share your results :)

closed 2021-01-04 by github-actions[bot]