Ysurac/openmptcprouter #1203: How to debug OMR-bypass?

Adorfer, 2020-09-04:
## Expected Behavior

OMR bypass either "to dedicated modem" (lan-port), or "to master port" (if bypassuplink left empty)

- traceroute going not via VPS
- latency (ping) to service at provider x should be as expected for 'via uplink x'
- service web-ui should show for logged in client 'IP of uplink x' instead of VPS-ip

## Current Behavior
OMR does not seem to work
- based on LAN mac
- based on LAN IP
- based on DST ASN
- based on DST IP
- based on DST FQDN

verified by mtr and traceroute
client uses DNS (v4, v6) of OMR
client is win10 and ubuntu18.04
had been working on 0.55beta3(?). Setup was not NOT upgraded, but fresh install on 0.55.1

## Possible Solution

switching off VPS routes directly via uplink

## Steps to Reproduce the Problem

unknown how do describe. 
please provide instruction how to debug locally.

## Context (Environment)

OMR with 3 uplinks (2x VDSL, 1x DOCSIS)

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.55.3 r0+14198-da98603597 / LuCI Master git-20.219.57394-35e2735
  - OpenMPTCProuter VPS version: Version 0.1017 5.4.52-mptcp
  - OpenMPTCProuter platform: x64-ext4-EFI, i5

note: had been working on 0.55beta3(?). Setup was not NOT upgraded, but fresh install on 0.55.1

Ysurac, 2020-09-04:
Simple check, add domain ```ifconfig.co``` in domain bypass, and from the router via SSH ```curl ifconfig.co``` do you get VPS IP or WAN IP ?

Adorfer, 2020-09-04:
wan-ip!

Ysurac, 2020-09-04:
check in State->System log if there is no firewall/iptables error.

Adorfer, 2020-09-04:
> logread |grep "error"

exclusivly strings like 

> Thu Sep  3 00:05:58 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found 

and one about a problem in vnstat-settings.

Ysurac, 2020-09-04:
what is the result of ```iptables-save``` and ```ipset list``` via SSH from the router ?

Adorfer, 2020-09-04:
https://termbin.com/ku78
and 
https://termbin.com/mah3

Ysurac, 2020-09-04:
I don't think this will fix the problem, but there is errors:
```
omr-bypass.@ips[4].ip='217.10.68.145 '
omr-bypass.@ips[6].ip=' 94.79.131.221'
```
(the space after first IP and space before last ip)

Adorfer, 2020-09-04:
fixed those issues, but as expected, did not resolve the issue.

Ysurac, 2020-09-04:
ifconfig.co ip is 172.67.133.228 and 2606:4700:3032::681c:125e both are in ipset.
94.79.131.221 is also in ipset.
What is the result of ```ip rule``` ?

Adorfer, 2020-09-04:
ip rule : 
https://termbin.com/r6b6

Ysurac, 2020-09-04:
And what is the result of ```ip r show table 991337``` and ```ip r show table 1``` ?

Adorfer, 2020-09-04:
not much: 

```
root@LES-OpenMPTCP:~# ip r show table 991337
default via 192.168.12.1 dev eth3
root@LES-OpenMPTCP:~# ip r show table 1
Dump terminated
```

only table 0 is a bit more: 
https://termbin.com/0u07

Ysurac, 2020-09-04:
I should have asked table 4, 5 and 6. They shouldn't be empty.
But as table 991337 is not empty all should be ok here.

What is the result of ```iptables -vL -t nat``` ?

Adorfer, 2020-09-04:
Viola! 

ip r show table 4
https://termbin.com/pv5x0

ip r show table 5 
https://termbin.com/na2b

ip r show table 6
https://termbin.com/h1f5

iptables -vL -t nat
https://termbin.com/pqkc

Ysurac, 2020-09-04:
Strange, all seems to work here.

Adorfer, 2020-09-04:
What can i do? do a fresh install?

Ysurac, 2020-09-05:
You can try yes, because I have no more idea.
You don't use multiple public IPs on the VPS ?

Adorfer, 2020-09-05:
single IPv4 (/32) and IPv6 is /64 on the vps.

Adorfer, 2020-09-11:
i did fresh install on a completly new device. 
only OMR-rule is "ifconfig.co"

still the same, 
 
    curl http://ifconfig.co

returns the vps-ip. 

[image]

Ysurac, 2020-09-12:
Where do you run the command ?
You get VPS IPv4 or IPv6 ? If it's IPv6, try ```curl -4 ifconfig.co```

Adorfer, 2020-09-12:
i run the command on the OMR shell, and the result is IPv4 of the VPS.

closed 2021-01-04 by github-actions[bot]