Ysurac/openmptcprouter #1204: Cannot install RPI 3+, stuck at "Adding VLAN0 to HW filter"

Merrick28, 2020-09-04:
## Expected Behavior
Hello,

I just bought a new rpi 3+, and I want to install image taken here : https://www.openmptcprouter.com/download
I flash the image (tested with etcher, dd, ...), and then I boot the rpi with the new card

## Current Behavior
The rpi boots, but hangs at "Adding VLAN0 to HW filter", and nothing happens.
Tested on another rpi that works, same issue.
Tested with 32 bits image, same issue

## Context (Environment)
As the rpi won't boot, I can't use it...

## Specifications

  - OpenMPTCProuter version: openmptcprouter-v0.55.3-r0+14198-da98603597-bcm27xx-bcm2710-rpi-3-ext4-factory.img.gz 
(tested also with squashfs)

Ysurac, 2020-09-04:
"Adding VLAN0..." is in most case latest boot messages.
Hit "Enter" key to show login prompt and access web interface via http://192.168.100.1/

Merrick28, 2020-09-04:
Oh... My... God... I'm sorry, I was just waiting for the prompt. Please accept my apologies.

closed 2020-09-04 by Merrick28