Ysurac/openmptcprouter #1205: Debian 10 Buster compile source can't get started: openssl-1.0-dev not located

PCRyan, 2020-09-04:
## Expected Behavior
Needed packages per instructions should be installed.

## Current Behavior
Error that openssl-1.0-dev can't be located, so environment installation aborts.

## Possible Solution
Tell me how to fix, hopefully. Then clarify or correct current documentation, I suppose.

## Steps to Reproduce the Problem
  1. On AWS, create Debian instance and initialize to Debian 10 (Buster)

  2. Get ready for package installation with `sudo apt update` then `sudo apt upgrade`

  3. Run following as described in the primary OMR Readme.md 
`sudo apt install build-essential git unzip ncurses-dev libz-dev libssl-dev openssl-1.0-dev python python3-dev python3.5 libelf-dev subversion gettext gawk wget curl rsync perl`

4. Following output received

Linux ip-172-31-47-186 4.19.0-10-amd64 #1 SMP Debian 4.19.132-1 (2020-07-24) x86_64

Last login: Fri Sep  4 16:19:50 2020 from 185.69.xxx.xxx
admin@ip-172-31-47-186:$ sudo apt install build-essential git unzip ncurses-dev libz-dev libssl-dev openssl-1.0-dev python python3-dev python3.5 libelf-dev subversion gettext gawk wget curl rsync perl
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'libncurses-dev' instead of 'ncurses-dev'
Note, selecting 'zlib1g-dev' instead of 'libz-dev'
Note, selecting 'libpython3.5-stdlib' for regex 'python3.5'
Note, selecting 'libpython3.5-minimal' for regex 'python3.5'
E: Unable to locate package openssl-1.0-dev
E: Couldn't find any package by glob 'openssl-1.0-dev'
E: Couldn't find any package by regex 'openssl-1.0-dev'

## Context (Environment)
Am preparing environment to build OMR myself, with a view towards then customizing to support a currently unsupported target.

## Specifications
  - OpenMPTCProuter version: none since haven't gotten that far, but it will be the current master version.
  - OpenMPTCProuter VPS version: not applicable
  - OpenMPTCProuter platform: not applicable

Ysurac, 2020-09-04:
The line was for Debian 9. Now it's libssl-dev package.

Ysurac, 2020-09-04:
This https://github.com/Ysurac/openmptcprouter/wiki/Create-image-for-unsupported-platform should be up to date.

PCRyan, 2020-09-04:
Thanks. I'll let you close this when you see fit (e.g. after readme update)

closed 2021-01-04 by github-actions[bot]