Ysurac/openmptcprouter #1207: Feature: VPN path manager.

dragon2611, 2020-09-04:
Where the underlaying VPN's support doing so it would be nice if possible to expose the path management functions in the UI.

For instance I'd like to be able to say all traffic using the VPN uses only WAN2 unless it's down.

This would be useful where you have WANs of differing latencies where you want the TCP to be bonded (Via MPTCP) but where you you want latency sensitive traffic to go over the WAN with the lowest latency.

The reason to use the VPN here instead of direct egress out of the WAN (OMR-BYPASS) is if the IP need to remain the same between the UDP and TCP traffic (Such as is the case with some games where the login is over TCP and the actual ingame traffic is over UDP, they often can't deal with The UDP traffic coming from a different source IP)

Ysurac, 2020-09-05:
mptcp can be disabled on glorytun TCP, this can maybe really disable it.
Else I will not add this features for now, but a custom VPN can be configured, like wireguard that is based on UDP and doesn't support multipath.

dragon2611, 2020-09-05:
Ok I think It maybe can be done with the CLI e.g glorytun-udp path

closed 2021-01-04 by github-actions[bot]