Ysurac/openmptcprouter #1208: IPv6 VPS

dragon2611, 2020-09-04:
I enabled IPv6 on openmptcprouter and assigned a /60 that was routed to the VPS but it didn't work (Note IPv6 was added later as originally it wasn't available) however the VPS wasn't forwarding the V6 to it's upstream gateway.

Looks like only the ethernet interface had IPv6 forwarding enabled.

Running 

`sysctl -w net.ipv6.conf.all.forwarding = 1` on the VPS enabled the IPv6 subnet to route.

Ideally the admin script needs to do this when you enable IPv6 on the openmptcprouter side.

Ysurac, 2020-09-05:
IPv6 must be enabled on the VPS (even without public IP) before running VPS script (a shadowsocks config change is made by the script if IPv6 is disabled).
If Public IPv6 is set after the VPS script was launched, the VPS must be rebooted (or shorewall6 restarted) this will configure forwarding.

dragon2611, 2020-09-05:
Rebooted the VPS to be sure, SS is passing IPv6 but the vpn is not, needed to set the sysctl value again (I haven't persisted it yet)

< OpenMPTCProuter VPS 0.1017 > on Ubuntu 18.04

closed 2021-01-04 by github-actions[bot]