Ysurac/openmptcprouter #1209: OMR build failing on stock Debian 10 system

PCRyan, 2020-09-06:
## Expected Behavior
Under a stock and prepped-for-purpose Debian 10 (Buster) system, target rpi3 should compile without error.

## Current Behavior
rpi3 compilation throws error when compiling the cmGeneratorTarget.cxx module of the cmake tool.

## Possible Solution
Pick your brains on what the underlying issue is and work from there. At worst, I'd have to slog through a debug myself to narrow down the problem space, leading to pinpointing the cause and its solution. Normal stuff.

## Steps to Reproduce the Problem
  1. On AWS, configure a new instance to Debian 10. (If using a cheap-compute AWS platform like "tiny," tick the "unlimited CPU" option so that AWS won't slow down compilation midway as CPU credits are used up.)
  2. Update and upgrade apt. Install OMR build environment per instructions on OMR's "Create image for unsupported platform" page
  3. Run verbose build with 
`OMR_TARGET="rpi3" ./build.sh -j1 V=s`
  4. After some minutes (20 or so in my case with AWS "tiny") the following tail output is observed, showing where the error arose, while making what I assume to be CMakeLib:

time: tools/mklibs/compile#0.03#0.01#0.06
make[3]: Entering directory '/home/paul/openmptcprouter/rpi3/source/tools/libressl'
make[3]: Leaving directory '/home/paul/openmptcprouter/rpi3/source/tools/libressl'
time: tools/libressl/compile#0.06#0.03#0.16
make[3]: Entering directory '/home/paul/openmptcprouter/rpi3/source/tools/cmake'
CFLAGS="-O2 -I/home/paul/openmptcprouter/rpi3/source/staging_dir/host/include " CPPFLAGS="-I/home/paul/openmptcprouter/rpi3/source/staging_dir/host/include " CXXFLAGS="" LDFLAGS="-L/home/paul/openmptcprouter/rpi3/source/staging_dir/host/lib " make  -C /home/paul/openmptcprouter/rpi3/source/build_dir/host/cmake-3.17.0
make[4]: Entering directory '/home/paul/openmptcprouter/rpi3/source/build_dir/host/cmake-3.17.0'
make[5]: Entering directory '/home/paul/openmptcprouter/rpi3/source/build_dir/host/cmake-3.17.0'
make[6]: Entering directory '/home/paul/openmptcprouter/rpi3/source/build_dir/host/cmake-3.17.0'
make[6]: Leaving directory '/home/paul/openmptcprouter/rpi3/source/build_dir/host/cmake-3.17.0'
[  1%] Built target cmsys_c
. . . . (lines omitted by me for brevity)
 52%] Built target cmlibuv
make[6]: Entering directory '/home/paul/openmptcprouter/rpi3/source/build_dir/host/cmake-3.17.0'
make[6]: Leaving directory '/home/paul/openmptcprouter/rpi3/source/build_dir/host/cmake-3.17.0'
[ 56%] Built target cmForm
make[6]: Entering directory '/home/paul/openmptcprouter/rpi3/source/build_dir/host/cmake-3.17.0'
make[6]: Leaving directory '/home/paul/openmptcprouter/rpi3/source/build_dir/host/cmake-3.17.0'
make[6]: Entering directory '/home/paul/openmptcprouter/rpi3/source/build_dir/host/cmake-3.17.0'
[ 56%] Building CXX object Source/CMakeFiles/CMakeLib.dir/cmGeneratorTarget.cxx.o
    g++: fatal error: Killed signal terminated program cc1pluscompilation terminated.
make[6]: *** [Source/CMakeFiles/CMakeLib.dir/build.make:1263: Source/CMakeFiles/CMakeLib.dir/cmGeneratorTarget.cxx.o] Error 1
make[6]: Leaving directory '/home/paul/openmptcprouter/rpi3/source/build_dir/host/cmake-3.17.0'
make[5]: *** [CMakeFiles/Makefile2:1124: Source/CMakeFiles/CMakeLib.dir/all] Error 2
make[5]: Leaving directory '/home/paul/openmptcprouter/rpi3/source/build_dir/host/cmake-3.17.0'
make[4]: *** [Makefile:180: all] Error 2
make[4]: Leaving directory '/home/paul/openmptcprouter/rpi3/source/build_dir/host/cmake-3.17.0'
make[3]: *** [Makefile:51: /home/paul/openmptcprouter/rpi3/source/build_dir/host/cmake-3.17.0/.built] Error 2
make[3]: Leaving directory '/home/paul/openmptcprouter/rpi3/source/tools/cmake'
time: tools/cmake/compile#16.96#19.24#1270.77
make[2]: *** [tools/Makefile:163: tools/cmake/compile] Error 2make[2]: Leaving directory '/home/paul/openmptcprouter/rpi3/source'
make[1]: *** [tools/Makefile:159: /home/paul/openmptcprouter/rpi3/source/staging_dir/host/stamp/.tools_compile_yyyyyynnyyynyyyyyynyynnyyyynyyyyyyyyyyyyyyynynynnyyynnyyyy] Error 2
make[1]: Leaving directory '/home/paul/openmptcprouter/rpi3/source'
make: *** [/home/paul/openmptcprouter/rpi3/source/include/toplevel.mk:235: world] Error 2
paul@ip-172-31-16-210:~/openmptcprouter$

## Context (Environment)
I am looking to get a baseline OMR build environment working before trying to extend it for another, yet unsupported device.

## Specifications
  - OpenMPTCProuter version: 0.55.3 (according to `git tag --sort=committerdate | tail -1`), i.e. latest from develop branch
  - OpenMPTCProuter VPS version: not applicable
  - OpenMPTCProuter platform: RPI3

Adorfer, 2020-09-06:
my guess is on "not enough ram/swap".

Ysurac, 2020-09-06:
Check with ```dmesg``` if the process was not killed for memory usage or any other reason.

PCRyan, 2020-09-08:
Correct. Thanks for the suggestion. This here was indeed memory exhaustion. Compile has gotten much further, but not without multiple issues. I will report back when I have stuff synthesized. We can leave this issue open to receive my further input, if that works for you.

PCRyan, 2020-09-08:
On second thought, given the nice template you have for describing issues, I will open a new one for my next foray, and close this one with my memory-related findings:
1) 0.5GB RAM fails as described when building that particular CMake module.
2) 1GB RAM is also inadequate, as it gets depleted when building the golang package/module.
Therefore, on AWS at least a t2.small instance model is required for the build, as it has 2GB RAM. Furthermore I've found that more than 20GB storage is required for a normal RPi3 build, at least (or especially) if a verbose build is needed with stdout saved for analysis. 30GB has seemed adequate thus far.

closed 2020-09-08 by PCRyan