Ysurac/openmptcprouter #1211: Router Install Page clarification please

adamb, 2020-09-08:
## Expected Behavior

Maybe a diagram that shows how things can be connected?

(I would be happy to update the documentation after I get mine working...)

Some detailed descriptions on how to install OMR on RPI and Virtualbox.

## Current Behavior

Trying to understand how to install OMR on a Pi 4 which has one ethernet port.  The [router install page](https://github.com/Ysurac/openmptcprouter/wiki/Router-install) says **"You don't need ethernet adapters, only a switch is enough. Modems doesn't need to be directly connected to RPI ethernet port."**

Not sure what this means.  I have  LB1120 and a GL-X750 modems which both have ethernet.  Does this mean I can use a switch and plug both those ethernet connections into a switch along with the Pi?  

## Possible Solution

Maybe a diagram or picture showing how these can be connected.  (I'm happy to contribute once I get mine working.  Very excited about this project!) 

```ascii
  RPI4 ------> switch ------ > Eero Mesh Wifi
  LB1120------->
  GL-X750 -->
````

Also not sure how to install the software on the Pi either.  While I'm waiting for my Pi to arrive, I thought I might try installing OMR under Virtualbox on my Mac for now.  

## Steps to Reproduce the Problem

Go to https://github.com/Ysurac/openmptcprouter/wiki/Router-install and try to understand how to setup a Raspberry Pi and some modems to use OMR.

## Context (Environment)

I'm trying to install OMR on a RPI4 and use it with a couple LTE modems.  An LB1120 and a GL-X750.  The LB1120 has only an ethernet connection.   The LAN will feed an Eero mesh system, which also has an ethernet connection.  

How do I connect these?   The config page seems to indicate I can plug in all 4 devices to a switch.  This sounds great since it simplifies things.

## Specifications

  - OpenMPTCProuter version:  not sure yet, suggestions?
  - OpenMPTCProuter VPS version:  not sure yet, will use Linode Dallas probably
  - OpenMPTCProuter platform:  RPI4

Ysurac, 2020-09-08:
Yes you can set all modems and RPI to same switch. OpenMPTCProuter use MacVLAN to create some "virtual" interfaces based on the physical interface. DHCP must be disabled on each modems in this case (else this will conflict) and static IP must be used.

To install on RPI, you can use etcher to copy image to a SD card, like in the doc. better to choose 64bits image.

adamb, 2020-09-17:
My modem doesn't have a way to turn off the built in DHCP server.  It's a GL-x750.  I could maybe set the range of IPs to something like 192.168.10.255 to 255 maybe?

Ysurac, 2020-09-18:
No. Check if modem have a bridge mode that can be set. Else you need to use vlan if you have a manageable switch or usb to ethernet.

adamb, 2020-09-18:
I was able to turn off DHCP on the gl-x750.   Everything is working now!  I'm using LTE for one path and a UBNT Nanostation connected to a wifi hotspot for the other path.  

I plan to add a 3rd path with a second LTE connection eventually.  Thanks for all the support!  Amazing project.

closed 2020-09-18 by adamb