Ysurac/openmptcprouter #1212: Glorytun VPN Down

rnunes77, 2020-09-08:
## Expected Behavior

When starting the software, the GLORYTUN VPN is in a DOWN status
## Current Behavior

The VPN (glorytun) starts with the status inactive and it is necessary to restart it manually so that its status is with the green arrow.
## Possible Solution

It is necessary to restart it manually so that its status is with the green arrow. Can I put a script to perform this task automatically after startup?
## Steps to Reproduce the Problem

  1.
  2.
  3.

## Context (Environment)

I'm using a Raspberry Pi 3, running on version (v0.55.1 r0 +), kernel version 5.4.52, and three 4G modens. I have a single problem, the VPN (glorytun) starts with the status inactive and it is necessary to restart it manually so that its status is with the green arrow. Can someone help me? Can I put a script to perform this task automatically after startup?

## Specifications

  - OpenMPTCProuter version: v0.55.1
  - OpenMPTCProuter VPS version: 0.1017
  - OpenMPTCProuter platform: RPI3

Ysurac, 2020-09-08:
Check in the wizard, advenced settings checkbox that glorytun TCP is the VPN selected and save the wizard.

rnunes77, 2020-09-08:
Perfect!! After change to TCP the VPN working like a breeze! Thanks!!

closed 2020-09-08 by rnunes77