Ysurac/openmptcprouter #1213: Perte de connexion dans les jeux en ligne ( ex: TESO, Eve Online)

Randdrick, 2020-09-08:
## Expected Behavior
Jouer normalement, sans être déconnecté du serveur de jeu au bout de 10 minutes (ou moins).

## Current Behavior
Depuis le passage de la version 0.54.7 à la version 0.55.3, jouer à des jeux en ligne devient très compliqué

## Possible Solution
Un ping -t ( en continue donc) du VPS empêche la déconnexion du client avec le serveur de jeu, ou configurer le PCs en OMR by Pass.

## Steps to Reproduce the Problem

  1. Lancer un jeu en ligne (ex : TESO)
  2. Jouer quelques minutes
  3. Attendre la déconnexion et vérifier dans le journal système que l'erreur suivante apparaît bien : daemon.err /usr/bin/ss-redir[5835]: server recv: Operation timed out

## Context (Environment)
La connexion entre le client et le serveur du jeu est perdue au moindre problème sur le VPS. La route entre le VPS et le serveur du jeu semble être perdue.

## Specifications

- OpenMPTCProuter version: 0.55.3
- OpenMPTCProuter VPS version: 0.1017
- OpenMPTCProuter platform: Raspberry Pi 3 Model B Rev 1.2- X64

Ysurac, 2020-09-09:
S'agit il d'une mise à jour depuis une 0.54 ou d'une nouvelle installation ?
Le temps pourrait correspondre avec le timeout de ss-redir dans Services->Shadowsocks-libev...

Randdrick, 2020-09-09:
Il s’agit d’une nouvelle installation pour laquelle j’ai récupéré la sauvegarde de la version 0.54.7.
J’y ai pensé pour le timeout. Cependant il me semble que c’était le même pour la version 0.54.7, (qui est pour moi de 1000 ms.) ? Et je n’avais pas ce problème de déconnexion.

Randdrick, 2020-09-09:
Je viens de faire une nouvelle installation sans restauration de la sauvegarde.
Le problème est identique sur les deux serveur VPS que je possède (un est hébergé chez First Heberg, l'autre chez OVH) et ce quelque soit le serveur maître.

Ysurac, 2020-09-10:
Le ping utilise le VPN alors que shadowsocks ne sert que pour le TCP, hors Eve Online n'utilise que le TCP.
Il faudrait voir si il n'y a pas d'autres erreurs dans les logs.

Randdrick, 2020-09-10:
J'ai bien regardé. A part les kernels Warnings en masse (voir un de mes posts précédent) même sur une réinstallation propre, il n'y a pas d'erreurs autres que celle indiquée. 
A contrario de la version 05.55.3, dans la version 0.54.7, lorsqu'un time out se produit au niveau du VPS, cela se traduit par une augmentation visible (mais temporaire) de la latence, mais la connexion n'est pas perdue avec le serveur de jeu.

Ysurac, 2020-09-10:
Et en remettant le "Planificateur Multipath TCP" à défaut dans Réseau->MPTCP (il est en Blest depuis la 0.55) ? (et en redémarrant le routeur pour être sur)

Randdrick, 2020-09-10:
Je viens d'essayer cette manipulation. Cela ne résout pas le problème. Je suis déconnecté du serveur dès qu'un time out se produit.

Ysurac, 2020-09-11:
J'essaye avec la branche develop pour voir si le soucis existe toujours.

Randdrick, 2020-09-11:
Malheureusement, même problème avec la version 0.55.1-104-ge391fb2f (Branche Develop)

Ysurac, 2020-09-11:
J'ai pour le moment beaucoup moins souvent un "server recv: Operation timed out".
Après difficile pour moi de tester avec ces jeux car pas disponible sous Linux, et pour TESO même si il peut fonctionner sous Linux avec Steam il est payant.

Ysurac, 2020-09-11:
Le soucis est toujours présent dans Système->OpenMPTCProuter, puis onglet Paramètres avancés, "Selection de l'interface maître" est mis à "Pas de changements" ?

Randdrick, 2020-09-11:
Oui même problème si je met à "Pas de changement". Je suis déconnecté du serveur de jeu ( TESO ou Eve OnLine) au moindre server recv: Operation timed out

Ysurac, 2020-09-12:
Un des seuls "timeout" changé est le tcp_retries2, pour remettre comme avant pour la session via SSH sur le routeur ```sysctl -w net.ipv4.tcp_retries2=15``` (pour garder définitivement après reboot, c'est dans ```/etc/sysctl.d/zzz_openmptcprouter.conf```).

Randdrick, 2020-09-12:
[quote]sysctl -w net.ipv4.tcp_retries2=15[/quote]
Cela résout le problème. Je n'ai plus de problèmes de déconnexion ni même de "server recv: Operation timed out". avec la version 0.55.3.

Merci pour ton aide.

Ysurac, 2020-09-12:
Vraiment ?
Si tu me confirme ça après 24h je changerais donc le paramètre (et je serais content d'avoir trouver le soucis)

dettofatto, 2020-09-12:
I'm interested too...

Randdrick, 2020-09-12:
To resolve this problème : 
type on your router consol : **sysctl -w net.ipv4.tcp_retries2=15**
and save in the file zzz_openmptcprouter.conf  under /etc/sysctl.d/, this line : **net.ipv4.tcp_retries2=15**

Randdrick, 2020-09-13:
Plus de problèmes d'accès aux serveurs de jeu (TESO, Eve Online). La solution est efficace pour moi.
---
No more problems accessing game servers (TESO, Eve Online). The solution is effective for me

closed 2021-01-04 by github-actions[bot]