Ysurac/openmptcprouter #1214: 2 keys usb 4G

CoSysopK, 2020-09-09:
## Actual Behavior

Par défaut, clé-usb1 se connecte sur wan1 et clé-usb2 se connecte sur wan2
Lorsque le routeur redémarre, volontairement ou à cause d'une coupure involontaire.
clé-usb1 se connecte sur wan2 et clé-usb2 se connecte sur wan1.
Comment dois-je faire pour que clé-usb1 se connecte toujours sur wan1 et clé-usb2 se connecte toujours sur wan2 ?
Y a-t-il un appareil à acquérir ?

## Steps to Reproduce the Problem

 1. Install image for Raspberry PI 3/3B+ (64-bit)
 2. Install VPS Debian 10 Buster x86_64
 3. Configure the router with the wan1 static and wan2 static
 4. Add the IP address VPS and, all the informations retrived on the VPS
 5. Two keys 4G on usb PI 3B

## Specifications

  - OpenMPTCProuter version: Version 0.55.3
  - OpenMPTCProuter VPS version: Version 0.1017
  - OpenMPTCProuter platform: RPI3B+ 64 bits

Ysurac, 2020-09-09:
C'est en cours dév et test sur la branche develop.
C'est quoi comme clefs ?

CoSysopK, 2020-09-09:
2 clés 4G Huawei E8372

Ysurac, 2020-09-09:
En mode hi-link ? C'est ce que j'ai donc ça fonctionnera normalement correctement.

CoSysopK, 2020-09-09:
Non, je n'ai pas ce mode par défaut sur mes clés 4G Huawei E8372.
Avez-vous un lien URL pour procéder à cette installation si cela s'avère possible?

Ysurac, 2020-09-09:
Elles sont utilisées avec NCM alors ?

CoSysopK, 2020-09-09:
Je viens de vérifier sur l'interface. Il n'y a rien de tout ça.
Je vous fournis quelques captures écrans si toutefois, ces réglages sont bons.

[image]
[image]

CoSysopK, 2020-09-09:
Y a-t-il un firmware à mettre à jour ?

Ysurac, 2020-09-09:
Vu qu'il y a une interface, elles sont déjà en mode Hi-link :)
Donc quand elles sont branchées, une interface eth* est créée directement.

CoSysopK, 2020-09-09:
Oui, j'ai mis eth1 sur wan1 et eth2 sur wan2. Mais lors d'un redémarrage du routeur, eth1 va se mettre sur wan2, et cela pose problème car je n'ai pas de réseau. Alors je dois faire un redémarrage manuel en laissant juste la clé 4G wan1 eth1, pour que le réseau fonctionne, alors je peux rebrancher la clé 4G wan2 eth2, et tout se configure bien comme il faut.
Le soucis, lorsque je suis absent, cela ne va pas se remettre bien comme il faut.

Ysurac, 2020-09-09:
Oui, comme je l'ai dit la solution est en cours de développement dans la branche develop.

CoSysopK, 2020-09-09:
d'accord, impeccable.

closed 2020-09-09 by CoSysopK