Ysurac/openmptcprouter #1215: Port forwarding issue ?

jhackeur, 2020-09-09:
## Expected Behavior
Port forwarding not working as expected

## Current Behavior
I configure port forwarding based on this document:
https://github.com/Ysurac/openmptcprouter/wiki/Port-forwarding

However I'm not able to reach my webserver from my VPS public IP...

VPN is up
Redirects all ports from server to this router is checked
OpenMPTCP router IP: 192.168.89.101
WebServer IP: 192.168.89.33
[image]
[image]

## Possible Solution
Maybe a NAT issue ?

## Context (Environment)
I try to reach my web server from my public IP, currently trying with public port 8000 redirected to 80

## Specifications

  - OpenMPTCProuter version: v0.55.3 r0+14198-da98603597
  - OpenMPTCProuter VPS version: 0.1017 5.4.52-mptcp
  - OpenMPTCProuter platform: ESXI

Ysurac, 2020-09-09:
Did you do what is in https://github.com/Ysurac/openmptcprouter/wiki/Port-forwarding#debug ?

jhackeur, 2020-09-09:
Yes for all step!

Do we need to reboot or restart somes process ?
I can try to reboot both router and server later if needed ?

kevin39, 2020-09-09:
SNAT too ?

Ysurac, 2020-09-09:
So you see packets with tcpdump ? You can also run tcpdump on eth0 and see if it's correctly forwarded.

jhackeur, 2020-09-09:
Yes and yes!

I performed a reboot of my openmptcp router, and it's working fine now. Maybe a NAT table has to be cleared or something like that, anyways, works fine :)

closed 2020-09-09 by jhackeur