Ysurac/openmptcprouter #1216: documentation request about log message

zulu14, 2020-09-09:
hi again thanx for this awesome software, 

i get this error: 
Wed Sep  9 19:27:04 2020 daemon.info glorytun[7370]: STOPPED tun0
Wed Sep  9 19:27:05 2020 daemon.err glorytun[7370]: XXXXX.65001: connected
Wed Sep  9 19:27:05 2020 daemon.err glorytun[7370]: XXXXX.65001: key exchange failed
Wed Sep  9 19:27:05 2020 daemon.info glorytun[7370]: STOPPED tun0
Wed Sep  9 19:27:06 2020 daemon.err glorytun[7370]: XXXXX.65001: connected
Wed Sep  9 19:27:06 2020 daemon.err glorytun[7370]: XXXXX.65001: key exchange failed

can you tell me how is the location files make the _log function of the error ?  (RPI4) 
im curious about how its works :p
have a nice day to you !

zulu14, 2020-09-09:
hi , okay i have fix my issues, its not a general issues, d'ont worry, 
just question where the omr retrive all key by the api of vps ? (on the controller wizard_add, the key are get by the html , but you have a get request to vps (/config) with he token get by /token with the user and pass , im right?

Ysurac, 2020-09-09:
"Key exchange failed" occurs with Glorytun TCP when the key is incorrect or sometimes when IP address change.

Yes, keys are retrieved from the VPS API via /config with a token.

zulu14, 2020-09-09:
okay thanx for you kick answer, wich file make the request ?

zulu14, 2020-09-10:
hi Ysurac I allow myself to revive you

Ysurac, 2020-09-10:
openmptcprouter-vps service.

closed 2021-01-04 by github-actions[bot]