Ysurac/openmptcprouter #1217: Building for RPI3, unincluded dependency on wireguard-tools

PCRyan, 2020-09-09:
## Expected Behavior
Under a stock and prepped-for-purpose Debian 10 (Buster) system, target RPI3 should compile without error.

## Current Behavior
The build seems to near completing, reaching the stage of installing & configuring built package-items, only to fail over something to do with wireguard, specifically a dependency on wireguard-tools (see below). I've watched RAM consumption as the failure point approaches, but it remains low, so that shouldn't be the issue. Also dmesg shows nothing suspicious.

## Possible Solution
Hoping you all can diagnose the issue so I can bring this further. Else I'll be struggling to figure out how to adjust the environment to get the build to complete.

Note: I have no form of wireguard installed on the build system.

## Steps to Reproduce the Problem
  1. On AWS, configure a new instance to Debian 10 with minimum 2GB RAM and 30 GB storage. If using a cheap-compute AWS platform like "small," tick the "unlimited CPU" option so that AWS won't slow down compilation midway as CPU credits are used up.
  2. As usual, first update and upgrade apt. Then install the OMR build environment per instructions on OMR's "Create image for unsupported platform" page. If the "apt install" command shown there doesn't include the"gettext" package, add it to that command, as the build of OMR's "whois" package-item requires it and it's not present on stock Debian 10.
  3. For the build command, run it instead verbosely with 
`OMR_TARGET="rpi3" ./build.sh -j1 V=s`
  4. After many minutes (75 or so in my case with AWS "small") the following tail output is observed, showing where the error arose: 

Configuring luci-proto-ncm.
Configuring luci-i18n-https-dns-proxy-hi.
Configuring luci-i18n-shadowsocks-libev-it.
Configuring kmod-net-rtl8192su.
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for wireguard:
 * 	wireguard-tools
 * opkg_install_cmd: Cannot install package wireguard.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-proto-wireguard:
 * 	wireguard-tools
 * opkg_install_cmd: Cannot install package luci-proto-wireguard.
 * resolve_conffiles: Existing conffile /home/admin/openmptcprouter/rpi3/source/build_dir/target-aarch64_cortex-a53_musl/root-bcm27xx/etc/config/glorytun is different from the conffile in the new package. The new conffile will be placed at /home/admin/openmptcprouter/rpi3/source/build_dir/target-aarch64_cortex-a53_musl/root-bcm27xx/etc/config/glorytun-opkg.

make[2]: *** [package/Makefile:68: package/install] Error 255
make[2]: Leaving directory '/home/admin/openmptcprouter/rpi3/source'
make[1]: *** [package/Makefile:108: /home/admin/openmptcprouter/rpi3/source/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/admin/openmptcprouter/rpi3/source'
make: *** [/home/admin/openmptcprouter/rpi3/source/include/toplevel.mk:235: world] Error 2

## Context (Environment)
I am looking to get a baseline OMR build environment working before trying to extend it for another, yet unsupported device.

## Specifications
  - OpenMPTCProuter version: 0.55.3, i.e. latest from develop branch
  - OpenMPTCProuter VPS version: not applicable
  - OpenMPTCProuter platform: RPI3

Ysurac, 2020-09-09:
You are building from develop branch, it's not 0.55.3, not stable and sometimes can't be compiled directly.
To solve the issue, you can remove wireguard.

PCRyan, 2020-09-09:
>> You are building from develop branch, it's not 0.55.3 <<
Ah, so what version should I have put there? I thought it must be right because it was the output of 
`git tag --sort=committerdate | tail -1` 
which I found in build.sh.

I do see now that I missed the OMR_FEED_SRC="master" input to the build as noted on the "Create image for unsupported platform" page. I will try again starting from there and report back.

PCRyan, 2020-09-10:
My build now works as expected. Thank you! I will leave this open in anticipation of an answer to my question above about how I should have specified the OMR version when building from the develop branch. Just name it "develop"?

Ysurac, 2020-09-10:
Specifing OpenMPTCProuter feeds based version is done via OMR_FEED_SRC, this can be master, develop or a tag like v0.55.3

closed 2020-09-10 by PCRyan