Ysurac/openmptcprouter #1218: multisites

tycoons, 2020-09-10:
Hello,
a small message to have a little more info concerning the setting up of this project in multi site version with only one vps.
is this possible?
exemple:
site A:
2 line one adsl and one fiber
site B
2 line one adsl and one fiber
site a and b both connect to the same vps
I wanted to know if this is possible or if my vps will quickly be on their knees
thank you for your patience and your feedback on this subject

Ysurac, 2020-09-10:
This is supported by adding a new user to the VPS API, but it's not publicly documented at this time.

tycoons, 2020-09-11:
Hello,
Thank you for that answer. Do you think this will be documented soon?
if you have a little doc i'm a taker
thank you

mhohman, 2020-09-12:
I have this same request, Thanks in advance.

Ysurac, 2020-09-12:
Duplicate of https://github.com/Ysurac/openmptcprouter/issues/1058
Same answer.

closed 2020-09-12 by Ysurac

abpprkonsalting, 2024-09-21:
Hello, I've seen in multiple answers to this question that some work had been done in this issue, but it is not documented. At the same time it looks as if @Ysurac does not consider this a priority. Anyway for me it could be, and I would be happy to help in any task that could bring this to production stage, or published, or whatever.