Ysurac/openmptcprouter #1219: MTU issue

jhackeur, 2020-09-10:
## Expected Behavior
I'm currently facing to MTU issue. Without OpenMPTCP, I'm able to use MTU of 1480 for my 3 internet access.
With OpenMPTCP, I need to use less of 900...

## Current Behavior
VPN not working through OpenMPTCP

## Possible Solution
Absolutely no idea of what is the issue

## Steps to Reproduce the Problem
We can see the issue with a ping:
[image]

## Specifications

    OpenMPTCProuter version: v0.55.3 r0+14198-da98603597
    OpenMPTCProuter VPS version: 0.1017 5.4.52-mptcp
    OpenMPTCProuter platform: ESXI

Ysurac, 2020-09-10:
Check MTU of all interfaces in status page. Check also OMRVPN interface MTU.

jhackeur, 2020-09-10:
Thanks for this prompt reply!

On status page I have all working interfaces with more than 1440:
[image]
If I do the same test (ping) with MTU as 1440, it's working fine with all my internet access.

How can I check this point ?
" OMRVPN interface MTU" ?

Ysurac, 2020-09-10:
Using ```ip a``` or ```ip link``` via SSH on the router. MTU should be on interface tun0 in most case.

jhackeur, 2020-09-10:
[image]

So 1500 - how this is possible with my at 1447 on one internet access ?

Ysurac, 2020-09-10:
Hum, in fact I have same result when pinging google.fr, but no problem with 1.1.1.1. Can you check too ?

jhackeur, 2020-09-10:
Same issue with 1.1.1.1:
[image]

My main problem is about establishing an IPSEC VPN inside OpenMPTCP, it's not related to pinging some random IP with specified MTU :)
So I suspect this issue is related to all destination IP (at least for me)...

Ysurac, 2020-09-11:
You can try another VPN in System->OpenMPTCProuter wizard tab, and check advanced settings box.

jhackeur, 2020-09-11:
I try this yesterday evening (A dead simple vpn and openvpn) -> exactly the same behavior.

I've already try to deploy another router and another server (on ESX hosted in another VPS provider) -> same behavior.
Unfortunately, when I disconnect all my 4G access and keep only my ADSL, I got the same behavior too.

I suspect the issue is related to OpenMPTCP router -> we got no answer when MTU is too large (ex: 900):
[image]

Do you have the same behavior ?

Ysurac, 2020-09-11:
I have same behavior, but it's also the case from any server without OpenMPTCProuter. Google seems to limit ICMP packet size.

jhackeur, 2020-09-11:
Sorry but not really, as you can see here:
[image]

Ysurac, 2020-09-11:
Check from the VPS.

jhackeur, 2020-09-11:
Huummmm correct, seems to have an issue here...
Let me check with my provider and I come back to you with the feedback.

Ysurac, 2020-09-11:
No issue with provider, I have the same result on all VPS/server from many provider. Google seems to limit ICMP packet size from some source, but it's not a problem. No issue with 1.1.1.1 (from cloudfare) for example.
So I don't think this is related to your IPsec problem.

jhackeur, 2020-09-11:
Yes but with my VPS provider (Milkywan), I got this:
[image]
Same result witj google, 1.1.1.1, and all other website I've tested.

And of course same result to my VPN destination, that's why I suspect this as my issue !

jhackeur, 2020-09-13:
Ok so now it's solved for MTU part:
[image]

My VPN is bringing up, I'm able to ping inside the tunnel, but no ways to got an tcp connection running fine inside this tunnel...
There is something to fine tune to be able to used IPSEC vpn inside OpenMPTCP link ?

Ysurac, 2020-09-13:
The tunnel client is not installed on OpenMPTCProuter ? in this case nothing should be needed.

jhackeur, 2020-09-14:
There is no real "client": this is a Site to site IPSEC VPN.

I've try few thing:
1/ initiate this VPN between Home (ADSL) -> Remote pfSense VPN -> Working fine
2/ initiate this VPN between Home (4G) -> Remote pfSense VPN -> Working fine
3/ initiate this VPN between Home (openMPTCP router) -- (openMPTCP server) -> Remote pfSense VPN -> VPN are bringed up, but TCP flow are not working

Do you have already experienced this king of issue ?
Maybe an issue with multipath and IPSEC/GRE ?

Ysurac, 2020-09-16:
I just made a test with L2TP/IPsec and I have no issues.
There is no public server for GRE over IPsec, so more difficult to test.
You can ping over VPN ? UDP over the VPN is working ? TCP doesn't work at all ?

jhackeur, 2020-09-16:
Yes I can ping, and UDP works fine but with slow bandwidth.
TCP doesn't work at all.

I can manage to provide you a termination server on my VPS if you want to proceed with more test ?

Ysurac, 2020-09-16:
I think it's a firewall problem.

jhackeur, 2020-09-16:
Maybe. However why I can get this working fine with my ADSL and 4G access but not via OpenMPTCP ?
I'm not saying that OpenMPTCP is in cause, I'm juste trying to find a solution.

Do you have an idea how troubleshoot this ?

Ysurac, 2020-09-16:
Because same firewall is not used.
You can try to be really permissive in firewall and try remove syn flood protection in OpenMPTCProuter firewall.

jhackeur, 2020-09-17:
I will try this and keep you informed.

Another strange point. Usually I use speed.ui.com to run a speedtest.
Similar behavior: it's working fine with all my internet access, but when I used it via OpenMPTCP I got this issue at the end:
[image]

Do you have the same symptom ?
For you record, I have 2 OpenMPTCP router and 2 server : 1 on VM hosted by Milkywan, one other on a private server hosted at OneProvider -> same issue on both

Ysurac, 2020-09-17:
I have no issue on a scaleway VM. Check in your browser developer tools, network tab what is the error code.

jhackeur, 2020-09-17:
[image]

However no internet issue during the test, continuous ping still working fine...
I will try to order a scaleway VM for testing, hope this can solved all my issue, because I have few similar issue like this with other website (like my banking account disconnect me, ...) so not really stable for now.

closed 2021-01-04 by github-actions[bot]

ragefan, 2021-04-14:
I have the same issue with Cisco VPN client.