Ysurac/openmptcprouter #1220: Soucis : Timeout

AdrienDssy, 2020-09-10:
## Current Behavior
Bonsoir,

J'ai eu une amélioration de ma connexion 4G depuis quelques jours (avec une bien meilleure stabilité l'après midi et soir)
Mais je constate que j'ai souvent des timeout (sur pas mal de mes services, et souvent un petit coup de "impossible de trouver l'adresse ip du serveur", quand je navigue sur internet) avec discord (ça m'est arriver aussi avec NPMJS)...
[image]
[image]

## Specifications

  - OpenMPTCProuter version: (0.55.3) 
  - OpenMPTCProuter VPS version: (0.1017 5.4.52-mptcp)
  - OpenMPTCProuter platform: (x86_64)

AdrienDssy, 2020-09-10:
> Thu Sep 10 18:12:35 2020 daemon.info dnsmasq-dhcp[5212]: DHCPACK(eth0) 192.168.100.230 00:26:f2:b8:33:cd
Thu Sep 10 18:13:29 2020 daemon.err /usr/bin/ss-redir[6123]: server recv: Operation timed out
Thu Sep 10 18:13:42 2020 daemon.err /usr/bin/ss-redir[6122]: server recv: Host is unreachable
Thu Sep 10 18:18:21 2020 daemon.err /usr/bin/ss-redir[6123]: server recv: Operation timed out
Thu Sep 10 18:18:26 2020 daemon.err /usr/bin/ss-redir[6123]: server recv: Operation timed out
Thu Sep 10 18:18:28 2020 daemon.err /usr/bin/ss-redir[6122]: server recv: Operation timed out
Thu Sep 10 18:18:28 2020 daemon.err /usr/bin/ss-redir[6122]: server recv: Operation timed out
Thu Sep 10 18:18:31 2020 daemon.err /usr/bin/ss-redir[6122]: server recv: Operation timed out
Thu Sep 10 18:18:32 2020 daemon.err /usr/bin/ss-redir[6123]: server recv: Operation timed out
Thu Sep 10 18:18:33 2020 daemon.err /usr/bin/ss-redir[6123]: server recv: Operation timed out
Thu Sep 10 18:18:33 2020 daemon.err /usr/bin/ss-redir[6123]: server recv: Operation timed out
Thu Sep 10 18:18:33 2020 daemon.err /usr/bin/ss-redir[6123]: server recv: Host is unreachable
Thu Sep 10 18:18:33 2020 daemon.err /usr/bin/ss-redir[6123]: server recv: Operation timed out
Thu Sep 10 18:18:34 2020 daemon.err /usr/bin/ss-redir[6122]: server recv: Operation timed out
Thu Sep 10 18:18:34 2020 daemon.err /usr/bin/ss-redir[6123]: server recv: Operation timed out
Thu Sep 10 18:18:35 2020 daemon.err /usr/bin/ss-redir[6122]: server recv: Operation timed out
Thu Sep 10 18:18:35 2020 daemon.err /usr/bin/ss-redir[6123]: server recv: Operation timed out
Thu Sep 10 18:18:35 2020 daemon.err /usr/bin/ss-redir[6122]: server recv: Operation timed out
Thu Sep 10 18:18:36 2020 daemon.err /usr/bin/ss-redir[6122]: server recv: Operation timed out
Thu Sep 10 18:18:37 2020 daemon.err /usr/bin/ss-redir[6122]: server recv: Operation timed out
Thu Sep 10 18:18:37 2020 daemon.err /usr/bin/ss-redir[6123]: server recv: Operation timed out
Thu Sep 10 18:18:37 2020 daemon.err /usr/bin/ss-redir[6123]: server recv: Operation timed out
Thu Sep 10 18:18:38 2020 daemon.err /usr/bin/ss-redir[6122]: server recv: Operation timed out
Thu Sep 10 18:18:38 2020 daemon.err /usr/bin/ss-redir[6122]: server recv: Operation timed out
Thu Sep 10 18:18:39 2020 daemon.err /usr/bin/ss-redir[6122]: server recv: Operation timed out
Thu Sep 10 18:18:40 2020 daemon.err /usr/bin/ss-redir[6122]: server recv: Operation timed out
Thu Sep 10 18:18:41 2020 daemon.err /usr/bin/ss-redir[6122]: server recv: Operation timed out
Thu Sep 10 18:20:07 2020 daemon.err /usr/bin/ss-redir[6123]: server recv: Operation timed out
Thu Sep 10 18:20:44 2020 daemon.err /usr/bin/ss-redir[6123]: server recv: Operation timed out
Thu Sep 10 18:21:19 2020 daemon.err /usr/bin/ss-redir[6123]: server recv: Operation timed out
Thu Sep 10 18:21:25 2020 daemon.err /usr/bin/ss-redir[6123]: server recv: Operation timed out
Thu Sep 10 18:21:58 2020 daemon.err /usr/bin/ss-redir[6122]: server recv: Operation timed out
Thu Sep 10 18:22:25 2020 daemon.err /usr/bin/ss-redir[6122]: server recv: Host is unreachable
Thu Sep 10 18:22:27 2020 daemon.err /usr/bin/ss-redir[6123]: server recv: Operation timed out
Thu Sep 10 18:22:53 2020 daemon.err /usr/bin/ss-redir[6122]: server recv: Operation timed out
Thu Sep 10 18:22:58 2020 daemon.err /usr/bin/ss-redir[6122]: server recv: Operation timed out
Thu Sep 10 18:23:45 2020 daemon.err /usr/bin/ss-redir[6122]: server recv: Host is unreachable
Thu Sep 10 18:28:28 2020 daemon.err /usr/bin/ss-redir[6122]: server recv: Operation timed out
Thu Sep 10 18:34:51 2020 daemon.notice netifd: wan1 (3495): udhcpc: sending renew to 192.168.10.1
Thu Sep 10 18:34:51 2020 daemon.notice netifd: wan1 (3495): udhcpc: lease of 192.168.10.114 obtained, lease time 86400
Thu Sep 10 18:35:08 2020 daemon.info dnsmasq-dhcp[5212]: DHCPREQUEST(eth0) 192.168.100.177 ec:41:18:1c:fd:f5
Thu Sep 10 18:35:08 2020 daemon.info dnsmasq-dhcp[5212]: DHCPACK(eth0) 192.168.100.177 ec:41:18:1c:fd:f5 MiWiFi-R4CM
Thu Sep 10 18:37:20 2020 daemon.err /usr/bin/ss-redir[6122]: server recv: Operation timed out
Thu Sep 10 18:37:23 2020 daemon.err /usr/bin/ss-redir[6122]: server recv: Operation timed out
Thu Sep 10 18:38:39 2020 daemon.err /usr/bin/ss-redir[6123]: server recv: Operation timed out
Thu Sep 10 18:38:39 2020 daemon.err /usr/bin/ss-redir[6122]: server recv: Operation timed out
Thu Sep 10 18:38:40 2020 daemon.err /usr/bin/ss-redir[6122]: server recv: Operation timed out
Thu Sep 10 18:38:40 2020 daemon.err /usr/bin/ss-redir[6123]: server recv: Operation timed out
Thu Sep 10 18:38:40 2020 daemon.err /usr/bin/ss-redir[6122]: server recv: Operation timed out
Thu Sep 10 18:38:40 2020 daemon.err /usr/bin/ss-redir[6123]: server recv: Operation timed out
Thu Sep 10 18:38:41 2020 daemon.err /usr/bin/ss-redir[6123]: server recv: Operation timed out
Thu Sep 10 18:38:41 2020 daemon.err /usr/bin/ss-redir[6123]: server recv: Operation timed out
Thu Sep 10 18:38:41 2020 daemon.err /usr/bin/ss-redir[6123]: server recv: Operation timed out
Thu Sep 10 18:38:41 2020 daemon.err /usr/bin/ss-redir[6122]: server recv: Operation timed out
Thu Sep 10 18:38:42 2020 daemon.err /usr/bin/ss-redir[6123]: server recv: Operation timed out
Thu Sep 10 18:38:42 2020 daemon.err /usr/bin/ss-redir[6123]: server recv: Operation timed out
Thu Sep 10 18:38:43 2020 daemon.err /usr/bin/ss-redir[6122]: server recv: Operation timed out
[... cut ...]
Thu Sep 10 19:44:22 2020 daemon.err /usr/bin/ss-redir[6123]: server recv: Operation timed out
Thu Sep 10 19:44:22 2020 daemon.err /usr/bin/ss-redir[6123]: server recv: Operation timed out
Thu Sep 10 19:44:22 2020 daemon.err /usr/bin/ss-redir[6123]: server recv: Operation timed out
Thu Sep 10 19:44:23 2020 daemon.err /usr/bin/ss-redir[6123]: server recv: Operation timed out
Thu Sep 10 19:44:23 2020 daemon.err /usr/bin/ss-redir[6122]: server recv: Operation timed out
Thu Sep 10 19:44:24 2020 daemon.err /usr/bin/ss-redir[6123]: server recv: Operation timed out

Ysurac, 2020-09-11:
J'essaye de reproduire le soucis sur la branche develop. J'ai le même avec la version courante (en beaucoup moins courant cependant).

AdrienDssy, 2020-09-11:
D'accord, donc ça viens bien du routeur si je comprends bien 🤔

Ysurac, 2020-09-11:
C'est possible. Je cherche.
Si c'est bien le soucis, la prochaine version sera disponible très rapidement.

AdrienDssy, 2020-09-11:
D'accord, merci !

closed 2020-09-11 by AdrienDssy