Ysurac/openmptcprouter #1221: Network / MPTCP / Bandwith strange behavior

Corrado56, 2020-09-11:
Hi, openmptcp 0.55.3 on Linksys 3200

This is the result of a speedtest:
[image]
but after 3 minutes
[image]
the blue connection WAN2 is running quicker than the others.

Regards
Corrado

Ysurac, 2020-09-12:
The issue is the graph or the speed ?

Corrado56, 2020-09-12:
The issue is with the graph.
the blue area is moving left quicker that the other two. 
I have done only one speedtest. 
The graph above seems correct (3 connection bonded at the same time).  
in the graph below the blue connection appears to be done before the other two.
I am doing nothing only look at the graph for three minutes
It's an UI interface error, not a technical one.

closed 2021-01-04 by github-actions[bot]