Ysurac/openmptcprouter #1223: Problème sur les protocoles et services d'OMR by Pass

Randdrick, 2020-09-12:
## Expected Behavior
On doit pouvoir définir une application listée dans l'OMR by Pass et l'appliquer a une interface particulière.

## Current Behavior
Certaine application ( comme Teamspeak par exemple) ne fonctionne pas. D'autres (Comme World Of Warcraft par exemple) provoquent une erreur.

## Possible Solution
Ouvrir les ports nécessaires soit-même.

## Steps to Reproduce the Problem

  1. Aller dans le menu omr-by-pass
  2. Dans protocole et services, choisir l'application et l'affecter à une interface (ou ne pas choisir d'interface ; le problème est identique)
  3. Lancer l'application

## Context (Environment)
- Teamspeak : impossible de se connecter (même au serveur public)
- World of Warcarft : voir erreur ci-après.
- Netflix : OK

## Specifications

- OpenMPTCProuter version: 0.55.3
- OpenMPTCProuter VPS version: 0.1017
- OpenMPTCProuter platform: Raspberry Pi 3 Model B Rev 1.2- X64

Sat Sep 12 21:37:01 2020 daemon.err omr-tracker[14063]: Segmentation fault
Sat Sep 12 21:37:04 2020 daemon.err omr-tracker[14066]: Segmentation fault
Sat Sep 12 21:37:04 2020 daemon.err omr-tracker[14064]: Segmentation fault
Sat Sep 12 21:37:07 2020 daemon.err omr-tracker[14063]: Segmentation fault
Sat Sep 12 21:37:10 2020 daemon.err omr-tracker[14066]: Segmentation fault
Sat Sep 12 21:37:10 2020 daemon.err omr-tracker[14064]: Segmentation fault
Sat Sep 12 21:37:13 2020 daemon.err omr-tracker[14063]: Segmentation fault
Sat Sep 12 21:37:16 2020 daemon.err omr-tracker[14066]: Segmentation fault
Sat Sep 12 21:37:16 2020 daemon.err omr-tracker[14064]: Segmentation fault
Sat Sep 12 21:37:19 2020 daemon.err omr-tracker[14063]: Segmentation fault
Sat Sep 12 21:37:22 2020 daemon.err omr-tracker[14066]: Segmentation fault
Sat Sep 12 21:37:22 2020 daemon.err omr-tracker[14064]: Segmentation fault
Sat Sep 12 21:37:25 2020 daemon.err omr-tracker[14063]: Segmentation fault
Sat Sep 12 21:37:28 2020 daemon.err omr-tracker[14064]: Segmentation fault

Ysurac, 2020-09-13:
You have the error when you run world of warcraft or even without it launched ?
Do you have something related in State->kernel log ?

Randdrick, 2020-09-13:
Yes. For World of Warcraft, i can't connect to my account. Without OMR by pas, WOW work fine.

Ysurac, 2020-09-15:
You have the issue even with account on the website or only in the app ?
(As it's a windows game I can't easily test that)

Randdrick, 2020-09-15:
Only in the app.

closed 2021-01-04 by github-actions[bot]