Ysurac/openmptcprouter #1224: Bug : Lors d'un ajout d'une interface

AdrienDssy, 2020-09-13:
## Current Behavior
Quand j'ajoute une interface (peux importe que ce soit un nouveau WAN ou un MPTCP over VPN) ça retire les "anciennes interfaces" du pare-feu. Donc il faut les rajouter manuellement pour que ça re-fonctionne.
[image]
[image]
[image]
Toute fois le fait de le rajouter fait un autre "bug" j'ai l'impression
## Specifications

 - OpenMPTCProuter version:  (0.55.3) 
 - OpenMPTCProuter VPS version:  (0.1017 5.4.52-mptcp)  
 - OpenMPTCProuter platform:  (x86_64)

Ysurac, 2020-09-13:
L'interface est ajoutée via le wizard quand une seule interface physique est disponible ?

AdrienDssy, 2020-09-13:
Via le Wizard

Ysurac, 2020-09-13:
Je n'arrive pas à reproduire le soucis.

AdrienDssy, 2020-09-13:
Je ne sais pas si c'est le fait que c'est du à des partages de connexion via mes téléphone mais j'y crois moyen

Ysurac, 2020-09-13:
Par défaut une interface de type macvlan est crée, l'interface physique choisie est eth0 ou une autre interface ?

AdrienDssy, 2020-09-16:
Selon l'appareil c'est soit eth5 pour l'IPhone ou usb0 pour mon Honor

tboome33, 2020-09-17:
Perso, j'ai jamais réussi à ajouter le partage de connection d'un Iphone. Le DHCP ésotérique de l'Iphone semble metter la pagaille dans toutes les interfaces avec un retour en arrière impossible !!!

AdrienDssy, 2020-09-18:
J'ai pas ce soucis personnellement mais j'ai plutôt l'impression que l'agrégation ce fait pas quand c'est un partage de connexion via USB (je constate que rien ne change alors que ça débite sur le forfait (4G Illimité le we))

Ysurac, 2020-09-18:
Il faut vérifer dans Réseau->Interfaces que les interfaces ne sont pas de type macvlan (car pas besoin dans ce cas) et voir si l'interface est bien dans la zone "wan" niveau firewall.

AdrienDssy, 2020-09-18:
Ils sont dans le wan j'ai revérifier et je dois aller où pour voir si c'est du macvlan ?

Ysurac, 2020-09-18:
Dans l'onglet "Paramètres physiques" de l'interface.

AdrienDssy, 2020-09-18:
C'est en mode normal

AdrienDssy, 2020-09-22:
up :)

closed 2020-09-25 by AdrienDssy