Ysurac/openmptcprouter #1225: [FeatureRequest] Connect to OMR to VPS via IPv6 if available (for DSlite-WANs)

Adorfer, 2020-09-13:
## Expected Behavior

one of my LAN connections is "DSlite", which is basically "encapsulationg IPv4 through Ipv6 tunnel protocoll, gooing via a CarrierGradeNAT at the broadband provider. 

this is not ideal in terms of 

    a) MTU-size (tunnel in a tunnel) 
    b) reliability, since CGN-Gateways tend to "rotate" every few hours, giving a new exit-IPv4 for the OMR via this WAN.

## Current Behavior
at the moment the VPS is systematically connected via ipv4 by the wizzard.
Only IPv4 is used to connect OMR to the VPS

## Possible Solution

  1. offer to enter IPv6 additional to IPv4 of the VPS
  2. determine on each WAN/Modem if IPv6 is available/has better MTU

## Steps to Reproduce the Problem

get a broadband connection "DSlite-only", see what happens to mtu-size and sporadic reconnect.

## Context (Environment)

Boradband connection via VDSL 1&1-dE, DOCSIS connection via Vodafone-DE

## Specifications

  - OpenMPTCProuter version: 0.55.3
  - OpenMPTCProuter VPS version:  0.1017 5.4.52-mptcp
  - OpenMPTCProuter platform: intel  pentium 4405U

Ysurac, 2020-09-15:
If I understand, you want a IPv6 and IPv4 support in same time ?
Possible but I don't think it's possible to aggregate IPv6+IPv4 connections (at least I don't see any easy way to do that).

Adorfer, 2020-09-15:
it would be nice to have a selection "per WAN line" if either IPv4 or IPv6 is used to connect to the VPS. 
(i am not asking to use ipv4 and ipv6 on the same WAN at the same time)

Adorfer, 2020-09-17:
Just as an annotatio, to visualize the imbalance of MTU-sizes here: 
[image]

closed 2021-01-04 by github-actions[bot]