Ysurac/openmptcprouter #1226: no server ping response after 1 seconds

bigcachetx, 2020-09-15:
## Expected Behavior
no server ping response after 1 seconds

## Current Behavior
Googled and it said this info just a notice and system will ok, but not really right

I run the speed test, only 1 - ATT works, the whole link lost when I unplug the ATT

## Possible Solution
I checked each line and can access internet 

## Steps to Reproduce the Problem

  1.
  2.
  3.

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: 0.55.3
  - OpenMPTCProuter VPS version: 0.1017 5.4.52
  - OpenMPTCProuter platform: x86_64

bigcachetx, 2020-09-15:
[image]

Ysurac, 2020-09-15:
Check that there is no firewall enabled on these devices.
You can also check using Network->MPTCP and MPTCP support check tab, or using via SSH on the router ```omr-tracebox <interface>```, if multipath is not blocked.

bigcachetx, 2020-09-15:
Thanks, I found a modem lost signal, after reset, all come back.

But I found the MPT works on fast.com but not youtube.com, even slower than 1 line.

Ysurac, 2020-09-16:
MPT ?
How do you test speed on youtube ?

bigcachetx, 2020-09-16:
"stats for nerds" will give you the connection speed to youtube.

Normally it should be the same as fast.com but it's not, much slower than that.

[image]
[image]

Ysurac, 2020-09-16:
There is no reason that speed from youtube can be the same as fast.com
Connection Speed provided by youtube is not accurate and is not mean to be.
Youtube also use Quic, when browser support it, that is based on UDP, so VPN must be up and work well.

bigcachetx, 2020-09-17:
I just found I submit same issues "Streaming issues #1030 "

I run shadowsocks on computer, the youtube stream reach to 90Mbps.
Or I use only 1 WAN, the speed will went up. Don't know where is the problem.

US wireless carrier limit the streaming speed, but normally VPN will solve the problem, this is very weird!

[image]

Ysurac, 2020-09-17:
What do you mean by "I run shadowsocks on computer" ?

bigcachetx, 2020-09-17:
I run shadowsocks proxy on computer, connect to VPS2 through OpenMPTCProuter.

Is this topo clear?
[image]

bigcachetx, 2020-09-17:
With 2 or more WAN:
Youtube 5-15Mbps by  blue route
90Mbps by yellow route.

With only 1 WAN
Both route will reach maximum of WAN bandwidth

[image]

Ysurac, 2020-09-17:
And same issue if VPS 2 is used instead of VPS 1 ?

bigcachetx, 2020-09-17:
Yes, same issue.

In fact, I have VPS3 earlier, after I submitted the ticket, I replaced VPS 3 with VPS1. But the same problem.

bigcachetx, 2020-09-18:
I almost find out why!
I install ssr onVPS2 by myself which include BBR accelerate.
Is this the reason?

Ysurac, 2020-09-19:
I don't think so.
Encryption is not disabled on shadowsocks in OpenMPTCProuter ?

bigcachetx, 2020-09-19:
OK, finally I found different:

If I deploy shadowsocks client on router (anotherone openwrt router), the same youtube result;
If I run shadowsocks client on computer, then it runs well.

I believe it's shadowsocks not OpenMPTCProuter problem.

Do you take a test or not?

Ysurac, 2020-09-19:
I don't have this issue, youtube use aggregation in my case, so it's difficult to test.
Did you try with another browser ?

closed 2021-01-04 by github-actions[bot]