Ysurac/openmptcprouter #1227: OMR bypass to a non mptcp wan

Rwestrem, 2020-09-15:
## Expected Behavior

Should be able to create a wan outside of openmptcp on the router and use OMR bypass to that wan

## Current Behavior

When the non openmptcp wan is selected in omr bypass it still bypasses to the first wan in the openmptcp bound wans instead.
## Possible Solution

Not sure

## Steps to Reproduce the Problem

  1.  Create the wan in the wizard, then disable it as a wan to use in openmptcp.  Select that wan in omr bypass and it still only bypasses to a wan I didn't select.
  2.  Create the wan on the interface page instead of in the wizard and same results.
  

## Context (Environment)

Trying to have Netflix hulu etc...bypassed to a wan not involved with openmptcpt 

## Specifications

  - OpenMPTCProuter version: 
55.3
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter platform:

Ysurac, 2020-09-15:
What is the result, via SSH on the router, of ```uci show network``` ?

Rwestrem, 2020-09-15:
Connecting to 192.168.100.1 port 22, please wait...
login as: 
root@192.168.100.1's password: 

BusyBox v1.31.1 () built-in shell (ash)

   ___                 __  __ ___ _____ ___ ___              _           
  / _ \ _ __  ___ _ _ |  \/  | _ \_   _/ __| _ \_ _ ___ _  _| |_ ___ _ _ 
 | (_) | '_ \/ -_) ' \| |\/| |  _/ | || (__|  _/ '_/ _ \ || |  _/ -_) '_|
  \___/| .__/\___|_||_|_|  |_|_|   |_| \___|_| |_| \___/\_,_|\__\___|_|  
       |_|                                                               
------------------------------------------------------------------------------
   (r0+14198-da98603597)
------------------------------------------------------------------------------
-----------------------------------------------------
 PACKAGE:     openmptcprouter
 VERSION:     v0.55.3

 BUILD REPO:  https://github.com/Ysurac/openmptcprouter.git
 BUILD DATE:  Sat 29 Aug 2020 05:59:41 PM UTC
-----------------------------------------------------
root@OpenMPTCProuter:~# uci show network
network.loopback=interface
network.loopback.ifname='lo'
network.loopback.proto='static'
network.loopback.ipaddr='127.0.0.1'
network.loopback.netmask='255.0.0.0'
network.loopback.multipath='off'
network.loopback.macaddr='00:00:00:00:00:00'
network.loopback.metric='1'
network.globals=globals
network.globals.multipath='enable'
network.globals.mptcp_path_manager='fullmesh'
network.globals.congestion='bbr'
network.globals.mptcp_checksum='0'
network.globals.mptcp_debug='0'
network.globals.mptcp_syn_retries='2'
network.globals.mptcp_fullmesh_num_subflows='1'
network.globals.mptcp_fullmesh_create_on_err='1'
network.globals.mptcp_ndiffports_num_subflows='1'
network.globals.ula_prefix='fd32:7ba4:387e::/48'
network.globals.mptcp_scheduler='default'
network.lan=interface
network.lan.proto='static'
network.lan.ipaddr='192.168.100.1'
network.lan.netmask='255.255.255.0'
network.lan.ip6assign='60'
network.lan.delegate='0'
network.lan.multipath='off'
network.lan.ip4table='lan'
network.lan.metric='2'
network.lan.macaddr='a0:04:60:11:8b:b6'
network.lan.ifname='eth1.2 eth1.3 eth1.4 eth1.5'
network.lan.type='bridge'
network.@switch[0]=switch
network.@switch[0].name='switch0'
network.@switch[0].reset='1'
network.@switch[0].enable_vlan='1'
network.@switch_vlan[0]=switch_vlan
network.@switch_vlan[0].device='switch0'
network.@switch_vlan[0].vlan='1'
network.@switch_vlan[0].ports='0t 5t 6t'
network.@switch_vlan[0].vid='10'
network.@switch_vlan[1]=switch_vlan
network.@switch_vlan[1].device='switch0'
network.@switch_vlan[1].vlan='2'
network.@switch_vlan[1].vid='2'
network.@switch_vlan[1].ports='1 6t'
network.@switch_vlan[2]=switch_vlan
network.@switch_vlan[2].device='switch0'
network.@switch_vlan[2].vlan='3'
network.@switch_vlan[2].vid='3'
network.@switch_vlan[2].ports='2 6t'
network.@switch_vlan[3]=switch_vlan
network.@switch_vlan[3].device='switch0'
network.@switch_vlan[3].vlan='4'
network.@switch_vlan[3].vid='4'
network.@switch_vlan[3].ports='3 6t'
network.@switch_vlan[4]=switch_vlan
network.@switch_vlan[4].device='switch0'
network.@switch_vlan[4].vlan='5'
network.@switch_vlan[4].vid='5'
network.@switch_vlan[4].ports='4 6t'
network.lan_rule=rule
network.lan_rule.lookup='lan'
network.lan_rule.priority='100'
network.wan1=interface
network.wan1.ip4table='wan'
network.wan1.multipath='master'
network.wan1.defaultroute='0'
network.wan1.ipv6='0'
network.wan1.metric='3'
network.wan1.peerdns='0'
network.wan1.ifname='eth0.10'
network.wan1.macaddr='a0:04:60:11:8b:b7'
network.wan1.proto='dhcp'
network.wan1.label='.10'
network.wan2=interface
network.wan2.ip4table='wan'
network.wan2.multipath='on'
network.wan2.defaultroute='0'
network.wan2.ipv6='0'
network.wan2.metric='4'
network.wan2.peerdns='0'
network.wan2.macaddr='a0:04:60:11:8b:b7'
network.wan2.ifname='eth0.20'
network.wan2.proto='dhcp'
network.wan2.label='.20'
network.wan3=interface
network.wan3.ip4table='wan'
network.wan3.defaultroute='0'
network.wan3.ipv6='0'
network.wan3.metric='5'
network.wan3.peerdns='0'
network.wan3.macaddr='a0:04:60:11:8b:b7'
network.wan3.ifname='eth0.30'
network.wan3.proto='dhcp'
network.wan3.label='.30'
network.wan3.multipath='backup'
network.wan4=interface
network.wan4.ifname='eth0.4'
network.wan4.ip4table='wan'
network.wan4.multipath='on'
network.wan4.defaultroute='0'
network.wan4.ipv6='0'
network.wan4.peerdns='0'
network.wan4.label='mc7455'
network.wan4.proto='static'
network.wan4.metric='6'
network.wan4.device='/sys/devices/platform/soc/100f8800.usb3/10000000.dwc3/xhci-hcd.1.auto/usb3/3-1/3-1:1.8/net/wwan0'
network.wan4.macaddr='a0:04:60:11:8b:b7'
network.omrvpn=interface
network.omrvpn.ifname='tun0'
network.omrvpn.ip4table='vpn'
network.omrvpn.multipath='off'
network.omrvpn.leasetime='12h'
network.omrvpn.type='tunnel'
network.omrvpn.txqueuelen='1000'
network.omrvpn.ipv6='0'
network.omrvpn.metric='7'
network.omrvpn.proto='none'
network.omr6in4=interface
network.omr6in4.proto='6in4'
network.omr6in4.ip4table='vpn'
network.omr6in4.multipath='off'
network.omr6in4.ipaddr='10.255.255.2'
network.omr6in4.peeraddr='10.255.255.1'
network.omr6in4.auto='0'
network.omr6in4.metric='8'
network.omr6in4.ip6addr='fe80::a00:2/126'
network.omr6in4.gateway='fe80::a00:1/126'
network.@switch_vlan[5]=switch_vlan
network.@switch_vlan[5].device='switch0'
network.@switch_vlan[5].vlan='6'
network.@switch_vlan[5].ports='0t 5t 6t'
network.@switch_vlan[5].vid='20'
network.@switch_vlan[6]=switch_vlan
network.@switch_vlan[6].device='switch0'
network.@switch_vlan[6].vlan='7'
network.@switch_vlan[6].ports='0t 5t 6t'
network.@switch_vlan[6].vid='30'
root@OpenMPTCProuter:~#

Ysurac, 2020-09-15:
Same for ```uci show omr-bypass``` and ```ip rule```

Rwestrem, 2020-09-15:
Connecting to 192.168.100.1 port 22, please wait...
login as: 
root@192.168.100.1's password: 

BusyBox v1.31.1 () built-in shell (ash)

   ___                 __  __ ___ _____ ___ ___              _           
  / _ \ _ __  ___ _ _ |  \/  | _ \_   _/ __| _ \_ _ ___ _  _| |_ ___ _ _ 
 | (_) | '_ \/ -_) ' \| |\/| |  _/ | || (__|  _/ '_/ _ \ || |  _/ -_) '_|
  \___/| .__/\___|_||_|_|  |_|_|   |_| \___|_| |_| \___/\_,_|\__\___|_|  
       |_|                                                               
------------------------------------------------------------------------------
   (r0+14198-da98603597)
------------------------------------------------------------------------------
-----------------------------------------------------
 PACKAGE:     openmptcprouter
 VERSION:     v0.55.3

 BUILD REPO:  https://github.com/Ysurac/openmptcprouter.git
 BUILD DATE:  Sat 29 Aug 2020 05:59:41 PM UTC
-----------------------------------------------------
root@OpenMPTCProuter:~# uci show omr-bypass
omr-bypass.all=interface
omr-bypass.m6replay=proto
omr-bypass.m6replay.url='m6web.fr' '6play.fr' '6cloud.fr'
omr-bypass.mycanal=proto
omr-bypass.mycanal.url='mycanal.fr' 'canal-plus.com'
omr-bypass.minecraft=proto
omr-bypass.minecraft.url='authserver.mojang.com'
omr-bypass.lesnumeriques=proto
omr-bypass.lesnumeriques.url='lesnumeriques.com' 'botscorner.com' 'app.botscorner.com'
omr-bypass.disneyplus=proto
omr-bypass.disneyplus.url='bamgrid.com' 'disney-plus.net'
omr-bypass.@dpis[0]=dpis
omr-bypass.@dpis[0].proto='amazonvideo'
omr-bypass.@dpis[0].interface='eth0.30'
omr-bypass.@dpis[1]=dpis
omr-bypass.@dpis[1].proto='hulu'
omr-bypass.@dpis[1].interface='eth0.30'
omr-bypass.@dpis[2]=dpis
omr-bypass.@dpis[2].proto='netflix'
omr-bypass.@dpis[2].interface='eth0.30'
omr-bypass.@dpis[3]=dpis
omr-bypass.@dpis[3].proto='disneyplus'
omr-bypass.@dpis[3].interface='eth0.30'
omr-bypass.@macs[0]=macs
omr-bypass.@domains[0]=domains
omr-bypass.@domains[0].name='netflix.com'
omr-bypass.@domains[0].interface='eth0.30'
omr-bypass.@domains[1]=domains
omr-bypass.@domains[1].name='nflxext.com'
omr-bypass.@domains[1].interface='eth0.30'
omr-bypass.@domains[2]=domains
omr-bypass.@domains[2].name='nflximg.com'
omr-bypass.@domains[2].interface='eth0.30'
omr-bypass.@domains[3]=domains
omr-bypass.@domains[3].name='nflximg.net'
omr-bypass.@domains[3].interface='eth0.30'
omr-bypass.@domains[4]=domains
omr-bypass.@domains[4].name='nflxso.net'
omr-bypass.@domains[4].interface='eth0.30'
omr-bypass.@domains[5]=domains
omr-bypass.@domains[5].name='nflxvideo.net'
omr-bypass.@domains[5].interface='eth0.30'
omr-bypass.lo=interface
omr-bypass.lo.id='1'
omr-bypass.tun0=interface
omr-bypass.tun0.id='7'
omr-bypass.wwan0=interface
omr-bypass.wwan0.id='6'
root@OpenMPTCProuter:~# ip rule
0:  from all lookup local
0:  from 10.0.2.128 lookup 9
0:  from 10.0.1.128 lookup 3
0:  from 10.0.0.128 lookup 4
0:  from 10.0.2.128 lookup 5
1:  from all fwmark 0x539 lookup 991337
1:  from all fwmark 0x5393 lookup 3
1:  from all fwmark 0x5394 lookup 4
1:  from all fwmark 0x5395 lookup 5
1:  from all fwmark 0x5397 lookup 7
1:  from all fwmark 0x5399 lookup 9
1:  from all fwmark 0x5391 lookup 1
1:  from all fwmark 0x5392 lookup 2
100:  from all lookup lan
10000:  from 192.168.100.1 lookup lan
20000:  from all to 192.168.100.1/24 lookup lan
32766:  from all lookup main
32767:  from all lookup default
90020:  from all iif lo lookup lan
root@OpenMPTCProuter:~#

Ysurac, 2020-09-15:
and after a ```/etc/init.d/omr-bypass restart```, what is the result of ```ip rule``` and it's working or no changes ?

Ysurac, 2020-09-15:
And what is the result of ```ifstatus wan3``` if it's not working ?

Rwestrem, 2020-09-15:
> And what is the result of `ifstatus wan3` if it's not working ?

Still not working.

Connecting to 192.168.100.1 port 22, please wait...
login as: 
root@192.168.100.1's password: 

BusyBox v1.31.1 () built-in shell (ash)

   ___                 __  __ ___ _____ ___ ___              _           
  / _ \ _ __  ___ _ _ |  \/  | _ \_   _/ __| _ \_ _ ___ _  _| |_ ___ _ _ 
 | (_) | '_ \/ -_) ' \| |\/| |  _/ | || (__|  _/ '_/ _ \ || |  _/ -_) '_|
  \___/| .__/\___|_||_|_|  |_|_|   |_| \___|_| |_| \___/\_,_|\__\___|_|  
       |_|                                                               
------------------------------------------------------------------------------
   (r0+14198-da98603597)
------------------------------------------------------------------------------
-----------------------------------------------------
 PACKAGE:     openmptcprouter
 VERSION:     v0.55.3

 BUILD REPO:  https://github.com/Ysurac/openmptcprouter.git
 BUILD DATE:  Sat 29 Aug 2020 05:59:41 PM UTC
-----------------------------------------------------
root@OpenMPTCProuter:~# ip rule
0:  from all lookup local
0:  from 10.0.2.128 lookup 9
0:  from 10.0.1.128 lookup 3
0:  from 10.0.0.128 lookup 4
0:  from 10.0.2.128 lookup 5
1:  from all fwmark 0x539 lookup 991337
1:  from all fwmark 0x5393 lookup 3
1:  from all fwmark 0x5394 lookup 4
1:  from all fwmark 0x5395 lookup 5
1:  from all fwmark 0x5397 lookup 7
1:  from all fwmark 0x5399 lookup 9
1:  from all fwmark 0x5391 lookup 1
1:  from all fwmark 0x5392 lookup 2
1:  from all fwmark 0x5396 lookup 6
100:  from all lookup lan
10000:  from 192.168.100.1 lookup lan
20000:  from all to 192.168.100.1/24 lookup lan
32766:  from all lookup main
32767:  from all lookup default
90020:  from all iif lo lookup lan
root@OpenMPTCProuter:~# ifstatus wan3
{
  "up": true,
  "pending": false,
  "available": true,
  "autostart": true,
  "dynamic": false,
  "uptime": 7365,
  "l3_device": "eth0.30",
  "proto": "dhcp",
  "device": "eth0.30",
  "updated": [
    "addresses",
    "routes",
    "data"
  ],
  "metric": 5,
  "dns_metric": 0,
  "delegation": true,
  "ipv4-address": [
    {
      "address": "10.0.2.128",
      "mask": 24
    }
  ],
  "ipv6-address": [
    
  ],
  "ipv6-prefix": [
    
  ],
  "ipv6-prefix-assignment": [
    
  ],
  "route": [
    
  ],
  "dns-server": [
    
  ],
  "dns-search": [
    
  ],
  "neighbors": [
    
  ],
  "inactive": {
    "ipv4-address": [
      
    ],
    "ipv6-address": [
      
    ],
    "route": [
      {
        "target": "0.0.0.0",
        "mask": 0,
        "nexthop": "10.0.2.1",
        "source": "10.0.2.128/32"
      }
    ],
    "dns-server": [
      "10.0.2.1"
    ],
    "dns-search": [
      "local.tld"
    ],
    "neighbors": [
      
    ]
  },
  "data": {
    "hostname": "OpenMPTCProuter",
    "leasetime": 43200
  }
}
root@OpenMPTCProuter:~#

Ysurac, 2020-09-15:
What is the result of ```ip r show table 5``` ?
And ```uci show dhcp.@dnsmasq[0].ipset``` ?

Did you try a reboot of the router ?

Rwestrem, 2020-09-15:
> What is the result of `ip r show table 5` ?
> And `uci show dhcp.@dnsmasq[0].ipset` ?
> 
> Did you try a reboot of the router ?

Yes.  Many reboots.  No change.

Connecting to 192.168.100.1 port 22, please wait...
login as: 
root@192.168.100.1's password: 

BusyBox v1.31.1 () built-in shell (ash)

   ___                 __  __ ___ _____ ___ ___              _           
  / _ \ _ __  ___ _ _ |  \/  | _ \_   _/ __| _ \_ _ ___ _  _| |_ ___ _ _ 
 | (_) | '_ \/ -_) ' \| |\/| |  _/ | || (__|  _/ '_/ _ \ || |  _/ -_) '_|
  \___/| .__/\___|_||_|_|  |_|_|   |_| \___|_| |_| \___/\_,_|\__\___|_|  
       |_|                                                               
------------------------------------------------------------------------------
   (r0+14198-da98603597)
------------------------------------------------------------------------------
-----------------------------------------------------
 PACKAGE:     openmptcprouter
 VERSION:     v0.55.3

 BUILD REPO:  https://github.com/Ysurac/openmptcprouter.git
 BUILD DATE:  Sat 29 Aug 2020 05:59:41 PM UTC
-----------------------------------------------------
root@OpenMPTCProuter:~# ip r show table 5
default via 10.0.2.1 dev eth0.30 
10.0.2.0/24 dev eth0.30 scope link 
root@OpenMPTCProuter:~# uci show dhcp.@dnsmasq[0].ipset
dhcp.cfg01411c.ipset='/googlevideo.com/omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/steamcontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1' '/netflix.com/omr_dst_bypass_eth0.30,omr6_dst_bypass_eth0.30' '/nflxext.com/omr_dst_bypass_eth0.30,omr6_dst_bypass_eth0.30' '/nflximg.com/omr_dst_bypass_eth0.30,omr6_dst_bypass_eth0.30' '/nflximg.net/omr_dst_bypass_eth0.30,omr6_dst_bypass_eth0.30' '/nflxso.net/omr_dst_bypass_eth0.30,omr6_dst_bypass_eth0.30' '/nflxvideo.net/omr_dst_bypass_eth0.30,omr6_dst_bypass_eth0.30' '/AmazonVideo:s3.ll.dash.row.aiv-cdn.net/omr_dst_bypass_eth0.30,omr6_dst_bypass_eth0.30' '/s3-dub.cf.dash.row.aiv-cdn.net/omr_dst_bypass_eth0.30,omr6_dst_bypass_eth0.30' '/AmazonVideo:dmqdd6hw24ucf.cloudfront.net/omr_dst_bypass_eth0.30,omr6_dst_bypass_eth0.30' '/d25xi40x97liuc.cloudfront.net/omr_dst_bypass_eth0.30,omr6_dst_bypass_eth0.30' '/AmazonVideo:.aiv-delivery.net/omr_dst_bypass_eth0.30,omr6_dst_bypass_eth0.30' '/aiv-cdn.net/omr_dst_bypass_eth0.30,omr6_dst_bypass_eth0.30' '/AmazonVideo:1s3.lvlt.dash.us.aiv-cdn.net.c.footprint.net/omr_dst_bypass_eth0.30,omr6_dst_bypass_eth0.30' '/s.loris.llnwd.net/omr_dst_bypass_eth0.30,omr6_dst_bypass_eth0.30' '/AmazonVideo:atv-ext.amazon.com/omr_dst_bypass_eth0.30,omr6_dst_bypass_eth0.30' '/c.media-amazon.com/omr_dst_bypass_eth0.30,omr6_dst_bypass_eth0.30' '/Hulu:.hulu.com/omr_dst_bypass_eth0.30,omr6_dst_bypass_eth0.30' '/NetFlix:netflix.com/omr_dst_bypass_eth0.30,omr6_dst_bypass_eth0.30'
root@OpenMPTCProuter:~#

Rwestrem, 2020-09-16:
Is there any other things I can show that would help me figure this issue out?

Rwestrem, 2020-09-19:
does any of this explain my bypass issues?

config interface 'loopback'
	option ifname 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'
	option multipath 'off'
	option macaddr '00:00:00:00:00:00'
	option metric '1'

config globals 'globals'
	option multipath 'enable'
	option mptcp_path_manager 'fullmesh'
	option congestion 'bbr'
	option mptcp_checksum '0'
	option mptcp_debug '0'
	option mptcp_syn_retries '2'
	option mptcp_fullmesh_num_subflows '1'
	option mptcp_fullmesh_create_on_err '1'
	option mptcp_ndiffports_num_subflows '1'
	option ula_prefix 'fd32:7ba4:387e::/48'
	option mptcp_scheduler 'default'

config interface 'lan'
	option proto 'static'
	option netmask '255.255.255.0'
	option ip6assign '60'
	option ifname 'eth1.5'
	option delegate '0'
	option multipath 'off'
	option ip4table 'lan'
	option metric '2'
	option macaddr 'a0:04:60:11:8b:b6'
	option ipaddr '192.168.1.1'
	option type 'bridge'

config switch
	option name 'switch0'
	option reset '1'
	option enable_vlan '1'

config switch_vlan
	option device 'switch0'
	option vlan '1'
	option ports '0t 5t'
	option vid '10'

config switch_vlan
	option device 'switch0'
	option vlan '2'
	option ports '0t 5t'
	option vid '20'
	option description 'wan2'

config switch_vlan
	option device 'switch0'
	option vlan '3'
	option vid '30'
	option ports '0t 5t'

config switch_vlan
	option device 'switch0'
	option vlan '4'
	option ports '0t 5t'
	option vid '40'

config switch_vlan
	option device 'switch0'
	option vlan '5'
[... 1734 lines cut ...]
	option name 'disneyplus.com.ssl.sc.omtrdc.net'

config domains
	option name 'dssott.com'

config domains
	option name 'dssott.com.akamaized.net'

config domains
	option name 'netflix.com'

config domains
	option name 'nflxext.com'

config domains
	option name 'nflximg.com'

config domains
	option name 'nflximg.net'

config domains
	option name 'nflxso.net'

config domains
	option name 'nflxvideo.net'

Ysurac, 2020-09-19:
I don't really understand what you want now: wan3 is down with multipath enabled and omr-bypass is set to use master interface.

Rwestrem, 2020-09-19:
> I don't really understand what you want now: wan3 is down with multipath enabled and omr-bypass is set to use master interface.

At this point I'm just trying to get OMR bypass to work any which way I can.  I set it to master (wan1) and yet ip will be from wan2.  Then I try setting OMR bypass to wan2 and ip will be the VPS.  Then I reboot and it all switches around.  Sometimes will be wan1, sometimes wan2, sometimes VPS.  Different devices get omr bypassed to different wans or not bypassed at all for the same sites.

Ysurac, 2020-09-19:
So it's not anymore initial issue...
You don't have same output when contacting same site ? How do you check that ?

Rwestrem, 2020-09-19:
> So it's not anymore initial issue...
> You don't have same output when contacting same site ? How do you check that ?

It's still original issue as it doesn't work any which way at all with any wan.  I checked by putting ifconfig.co in domains on omr bypass.  When undefined it will show ip of wan1 sometimes, wan2 sometimes, and VPS ip sometimes.  Rebooting normally always makes it go over VPS ip.  

I also checked that some devices can use Netflix and some can't.  All are using OMR ip as DNS.   Also if i add Netflix proto instead of individual domains it will not connect at all.  I get timeouts when trying to visit the site.  I should at least get the "you are using a vpn" warning.

Ysurac, 2020-09-19:
No it's not original issue that was only bypass to a non MPTCP wan.

I don't see ifconfig.co in the config. Try to set it to a fixed wan.

Netflix proto doesn't work in current release, this will be fixed in next release.

So the issue omr-bypass doesn't work well when output interface is unset ?

Rwestrem, 2020-09-19:
> No it's not original issue that was only bypass to a non MPTCP wan.
> 
> I don't see ifconfig.co in the config. Try to set it to a fixed wan.
> 
> Netflix proto doesn't work in current release, this will be fixed in next release.
> 
> So the issue is omr-bypass doesn't work well when output interface is unset ?

Kinda.  It has more of a chance of at least bypassing if I leave it unset.  If I set it to any other interface it will be unpredictable.  

I'm not convinced it is entirely working correctly when unset as well.

Ysurac, 2020-09-19:
ok I will check that. I made change on omr-bypass in develop branch, this may be a fixed issue.

Rwestrem, 2020-09-19:
> ok I will check that. I made change on omr-bypass in develop branch, this may be a fixed issue.

Thank you so much for the help.  I'm not great with knowing what to look for or test in order to explain my issues.  Can I compile with the developmental branch or is it not available?

closed 2021-01-04 by github-actions[bot]