Ysurac/openmptcprouter #1228: agrégation fail

zulu14, 2020-09-15:
hello, i have juste added my last modem number 8, and my mptcp borken, my trafic only pass on WAN 1 (its not a bypass )

into my mptcp connection i have this : 
[image]

my status page say all are ok, the syslog not tel an key echange failure or other, 

my mptcp settings are: 
[image]

thank you !

zulu14, 2020-09-15:
update: after reboot of OMR, only one modem are into the mptcp connection, 

[image]

zulu14, 2020-09-15:
the status page are always all green for all modem

Ysurac, 2020-09-15:
Please use template. I need to know release of router and VPS used.

zulu14, 2020-09-15:
sorry,

my omr ver: 0.55.3
my vps ver: 0.1017

Ysurac, 2020-09-15:
It's working with 7 modems but not with 8 ?
What do you have in Network->MPTCP, MPTCP Fullmesh tab ?

zulu14, 2020-09-16:
[image]

Ysurac, 2020-09-16:
Wrong tab.

zulu14, 2020-09-16:
sorry  for you:
[image]

zulu14, 2020-09-16:
result mptcp_check of a agrégation fail modem (wan2) :
[image]

zulu14, 2020-09-16:
(i have extend my syslog page to 300 line, but, i cannot show you the syslog, its spam by this)
[image]

Ysurac, 2020-09-16:
I see only 7 modems in MPTCP fullmesh tab. What is the result of ```multipath``` command via SSH on the router ? And what is the missing interface in fullmesh tab ?

For the log error, it's strange. CPU frequency shouldn't change. Check in System->OpenMPTCProuter, advanced settings tab that Scaling governor is set to "performance" and that "Minimum scaling CPU frequency" and "Maximun scaling CPU frequency" are both set at "1500000".

zulu14, 2020-09-16:
(when i apply its not update the CPUI params, (after F5 infact)) can i use the uci set and commit ? 

i have delete the modem number 8 to check if its rework but its no resume the agragtion, 
 

Multipath cmd: 
[image]

zulu14, 2020-09-16:
SS just fell now,  ( Shadowsocks: Rules DOWN)  and re up few second after ( Shadowsocks: Rules UP)

and i have found this in log: 
Wed Sep 16 19:22:08 2020 kern.err kernel: [84573.859068] xhci_hcd 0000:01:00.0: Ring expansion failed

Wed Sep 16 19:23:35 2020 user.notice post-tracking-post-tracking: Can't find Shadowsocks, restart it...

Ysurac, 2020-09-16:
After a "Ring expansion failed", that is a know RPI issue, USB is rebooted.
Try to reboot the router to fix aggregation.

zulu14, 2020-09-16:
i have reboot, and its not solve the issues, (i have only wan1 traffic )

zulu14, 2020-09-16:
[image]

Ysurac, 2020-09-16:
When I check previous screen, I see that Multipath TCP path-manager is not set to fullmesh. It must be set to fullmesh for aggregation.

zulu14, 2020-09-16:
okay i try it,

zulu14, 2020-09-16:
great, its working, the issues are so dump ,
thanx you, 

juste a liitle question, somme time the key are not retrive , i need to manualy restart curseboxmaster-vps, i have open file an i do not understand wich function will be run when you restart the service into the controller or a the first boot

Ysurac, 2020-09-16:
I don't know what you are talking about.

zulu14, 2020-09-16:
sorry wrong post,
into the openmptcprouter-vps with function are run bu the restart  (/etc/init.d/openmptcprouter-vps restart)

Ysurac, 2020-09-16:
If key are not retrieved at first launch, this service is restarted until it run correctly.
For info on what it does, check the source ;)

zulu14, 2020-09-16:
okay thanks !

zulu14, 2020-09-16:
good day to you !

closed 2020-09-16 by zulu14