Ysurac/openmptcprouter #1230: Impossible de faire fonctionner openmptcprouter

skyfr54, 2020-09-17:
Hello, I followed all the tutorials as needed, I installed the vps and prepared the two routers (a livebox in 192.168.1.1 and a tplink mr600 in 192.168.3.1). The network card of the pc is correctly configured. Then I run openmptcprouter on raspberry and it gives me what is on the photos, is there something I forgot ??? If you need more info I will give it ... thank you in advance for helping me Nb I made a mistake for the name for vps but nothing changes and both wan's ping...Si vous pouviez me répondre en français ça serait top. Merci!

[image]
[image]
[image]
[image]
[image]

Ysurac, 2020-09-17:
Ca aurait été mieux avec le template, notamment pour avoir les versions du VPS et du routeur ;)

Il faudrait via SSH sur le routeur, le résultat de ```ip a```, ```ip r``` et de ```ping 192.168.1.1```

skyfr54, 2020-09-17:
C'est ok j'ai tout revu, reste à tester....
[image]

closed 2021-01-04 by github-actions[bot]