Ysurac/openmptcprouter #1231: Comparing the Archtecture of Speedify and OpenMpTCPRouter

rifaterdemsahin, 2020-09-17:
I am about to buy a router
[image]

which is costly and i wanna aggregate 5 connections.

I see this guy trying to do that on openvpn and he mentions there is not much gain.
https://www.youtube.com/watch?v=I08A4-PWawk&feature=youtu.be

As the aggregation works at the speed of the lowest speed in the chain.

This is true for speedify and openvpn but in openmptcprouter is that the same case as well ?

What is the benefits of this architecture ? ( please exclude the costs / I am only here for the raw performance  )

tboome33, 2020-09-17:
Hi,

Cheap solution and better :
https://www.amazon.fr/LABISTS-Raspberry-Alimentation-Interrupteur-Dissipateur/dp/B084KNKYYP/
+
https://www.amazon.fr/gp/product/B00JZFT3VS/
+
https://www.amazon.fr/TP-Link-TL-SF1005D-100Mbps-Bo%C3%AEtier-Plastique/dp/B000FNFSPY/

send me the price difference :-))))

Roland

Ysurac, 2020-09-18:
> As the aggregation works at the speed of the lowest speed in the chain.

OpenMPTCProuter do real aggregation, not just bonding as in the video. So it will work at full speed of all links aggregated.

Adorfer, 2020-09-18:
The design of OpenMPTCProuter is explained on https://github.com/Ysurac/openmptcprouter/wiki

If that's not clear enough, please create a specific issue. 

Please do not mix mptcp with bonding (LACP, 1ad...) 

https://en.wikipedia.org/wiki/Multipath_TCP
https://en.wikipedia.org/wiki/Link_aggregation

dragon2611, 2020-09-20:
> > As the aggregation works at the speed of the lowest speed in the chain.
> 
> OpenMPTCProuter do real aggregation, not just bonding as in the video. So it will work at full speed of all links aggregated.

The only caveat I would say there is if one of the links is highly variable or has wildly different latency to the others it may cause some issues.

The LTE was so crap around here it was faster to leave it out.

Ysurac, 2020-09-20:
True, it's possible to increase syn retries, change scheduler and TCP congestion, but if too many packets are lost there is nothing that can be really done. This can also be a problem when wifi is used (even more if wifi is used to connect to a LTE router :) ).

dragon2611, 2020-09-20:
I think it was more the 30s (yes I really meant seconds!) or so buffering that was going on somewhere.

Ysurac, 2020-09-20:
30s latency ? It's far more than satellite connection. This may be better to use "IP over Avian Carriers" https://fr.wikipedia.org/wiki/IP_over_Avian_Carriers (not supported by OpenMPTCProuter :) )

dragon2611, 2020-09-20:
I think congested cell or channel saturation and some rather large buffers were at play, it was certainly weird.

Adorfer, 2020-09-21:
Talking about links with 5000ms+ latency is purely annecdotal in this issue IMO. 
It will NOT help to determine the answer to the question. 
Links with a latency of 5s plus will never work for any type of aggregation on layer2 or layer3. There might be some kind of application layer offloading for the "ocean freighter full of microSD card" scenario. but that's clearly out of scope.

rifaterdemsahin, 2020-09-22:
> send me the price difference :-))))

lol I dont have the time to go over a crappy hardware. 

I am trying to upload to google drive > wevideo > youtube using a NAS synology solution behind the router. I wanna get rid of connection lost issues and speed up the uploads/downloads. I need a workhorse who can do the job.

Peplink is ooutof my reach, speedify work on some clients and does not work in the router. ( maybe connectify and speedify combo in a windows box )

rifaterdemsahin, 2020-09-22:
> OpenMPTCProuter do real aggregation, not just bonding as in the video. So it will work at full speed of all links aggregated.

that is the main focus and the value proposition for me so I would go ahead and build the system. 

The initial systems with ESXI where not reliable due to the hardware. I need the fast reboot times when things go sideways.

rifaterdemsahin, 2020-09-22:
> If that's not clear enough, please create a specific issue.

my concern is with google drive like uploads if they turn into UDP they I would slow down to my slowest link.

most google drive syncs use tcp so i should be covered
example:http://www.t1shopper.com/tools/port-number/37483/

> dont mix mptcp with bonding 
I think the mptcp is a better tech as the bonding solutions always slows down and mptcp has better chances interms of performance as multipaths are getting used.

rifaterdemsahin, 2020-09-22:
> The only caveat I would say there is if one of the links is highly variable or has wildly different latency to the others it may cause some issues.

i am attaching 5 networks (3 LTE ,1 DSL and 1 FTTC ) so my case is like a lottery. I see them like little turtles and some packets will reach the shore not all of them. My worry is if the mptcp architecture would be able to keep the speed if some of the turtles fail to reach the shore. ( Shore is the GCP VPN located in London where i am connecting from Cambridge )

rifaterdemsahin, 2020-09-22:
> it's possible to increase syn retries, change scheduler and TCP congestion

how do we do this ?

rifaterdemsahin, 2020-09-22:
> Links with a latency of 5s plus will never work for any type of aggregation on layer2 or layer3

so for me to get the starlink from spacex does not make real sense than ?
https://www.starlink.com/

Ysurac, 2020-09-22:
Starlink will have a latency less than 100ms, so no problem here. Current satellite connection have a latency of 500ms.

For syn retries, scheduler,... it's in Network->MPTCP. BBR TCP congestion work well with good connections, but Cubic seems better when connections are bad. BLEST MPTCP scheduler should be better with LTE connections.

rifaterdemsahin, 2020-11-16:
I got the router and the speeds are great still there are issues on the first principles outlines in this video. I am curious about if I am on the right path with the architecture.

https://youtu.be/l3RErkPrQyg

Ysurac, 2020-11-16:
There is now more configuration possible, you can test in System->OpenMPTCProuter, "advanced settings" checkbox, the proxy v2ray instead of the default shadowsocks, can be better... or not.
I never tried streaming, so it's possible that some default settings are not the best for that and some optimizations can be done.

closed 2021-02-21 by github-actions[bot]