Ysurac/openmptcprouter #1232: Raspberry Pi 4, Wifi and Ethernet at the same time?

adamb, 2020-09-17:
## Expected Behavior

I'm trying to setup my Pi4 using the wifi for one path and ethernet over an LTE modem for the second.  I use the wizard to setup the LTE path over the ethernet and it comes up all green.  Then I use the wizard to add the built in wifi for the second path (select wlan and then "Add an interface").  The Pi is already on my wifi so I use the IP address that it got from my wifi router.  After I add this the status goes red for the LTE router and the VPS server.

If I delete the wlan interface, the wlan is gone but everything stays red in the status diagram.

## Current Behavior

Adding wlan on the settings wizard page sends system into bad state.  Restoring from backup doesn't seem to help either.

## Possible Solution

Maybe can't use RP4 wifi as a path?

## Context (Environment)

I'm trying to setup my Pi4 using the built in wifi for one path and ethernet over an LTE modem for the second.  

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.55.3 r0+14198-da98603597 / LuCI Master git-20.219.57394-35e2735
  - OpenMPTCProuter VPS version: Not sure, installed a few days ago from the script, running at Linode Dallas
  - OpenMPTCProuter platform: RPI4

Ysurac, 2020-09-18:
This can be used (even if not always a good idea, wifi is losing many packets) but wifi must not be set as a bridge, this need to be like this : https://github.com/Ysurac/openmptcprouter/wiki/Configure-Wifi (with mode set as client).

adamb, 2020-09-18:
I tried multiple times and couldn't get it to work.  I would have things working with one connection via the ethernet, then when I added the wifi, the status would show everything is offline.  I tried to store from backup but that didn't work.  I guess because it couldn't get to the server at that point.  I tried removing the wifi interface and rebooting and still nothing would connect.  I just re-flashed the RPi4 and started over.  In the end I decided not to use the built in RPi4 wifi.

Mutanter, 2020-09-21:
This is interesting. I had similar observations this weekend regarding the integration of a WLAN interface in OpenMPTCP and already suspected a bug. As soon as I added the WLAN interface via the OpenMPTCP Wizard, the other interfaces lost their firewall zone and could no longer be accessed. I was able to reproduce the whole thing several times. In the end I added the WLAN network as MPTCP path, using the normal LuCI interface management and everything worked fine.

closed 2021-01-04 by github-actions[bot]