Ysurac/openmptcprouter #1233: info error

ascomputer, 2020-09-18:
hello everyone, as the title suggests someone can tell me what caused this error?
Thank you

02:35 2020 kern.err kernel: [ 1842.723552] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Fri Sep 18 14:02:38 2020 kern.err kernel: [ 1845.724721] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Fri Sep 18 14:02:41 2020 kern.err kernel: [ 1848.725024] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Fri Sep 18 14:02:44 2020 kern.err kernel: [ 1851.722930] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Fri Sep 18 14:02:47 2020 kern.err kernel: [ 1854.722751] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Fri Sep 18 14:02:50 2020 kern.err kernel: [ 1857.728031] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Fri Sep 18 14:02:53 2020 kern.err kernel: [ 1860.723059] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Fri Sep 18 14:02:56 2020 kern.err kernel: [ 1863.724105] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Fri Sep 18 14:02:59 2020 kern.err kernel: [ 1866.730465] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Fri Sep 18 14:03:02 2020 kern.err kernel: [ 1869.722735] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Fri Sep 18 14:03:08 2020 kern.err kernel: [ 1875.732333] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Fri Sep 18 14:03:17 2020 kern.err kernel: [ 1884.730983] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Fri Sep 18 14:03:20 2020 kern.err kernel: [ 1887.786509] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Fri Sep 18 14:03:23 2020 kern.err kernel: [ 1890.723413] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Fri Sep 18 14:03:26 2020 kern.err kernel: [ 1893.722994] ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Fri Sep 18 14:03:26 2020 kern.err kernel: [ 1893.731992] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-110
Fri Sep 18 14:03:29 2020 kern.err kernel: [ 1896.728736] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Fri Sep 18 14:03:29 2020 kern.err kernel: [ 1896.739436] brcmfmac: brcmf_sdio_read_control: last control frame is being processed.
Fri Sep 18 14:03:31 2020 kern.err kernel: [ 1899.297497] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
Fri Sep 18 14:03:31 2020 kern.err kernel: [ 1899.303340] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-110
Fri Sep 18 14:03:35 2020 kern.err kernel: [ 1902.723229] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Fri Sep 18 14:03:38 2020 kern.err kernel: [ 1905.727712] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Fri Sep 18 14:03:44 2020 kern.err kernel: [ 1911.723974] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Fri Sep 18 14:03:47 2020 kern.err kernel: [ 1914.722168] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Fri Sep 18 14:03:50 2020 kern.err kernel: [ 1917.723908] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Fri Sep 18 14:03:53 2020 kern.err kernel: [ 1920.723910] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Fri Sep 18 14:03:56 2020 kern.err kernel: [ 1923.728359] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Fri Sep 18 14:03:59 2020 kern.err kernel: [ 1926.723185] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Fri Sep 18 14:04:02 2020 kern.err kernel: [ 1929.723011] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Fri Sep 18 14:04:05 2020 kern.err kernel: [ 1932.723633] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Fri Sep 18 14:04:08 2020 kern.err kernel: [ 1935.725258] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Fri Sep 18 14:04:11 2020 kern.err kernel: [ 1938.728872] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Fri Sep 18 14:04:11 2020 kern.err kernel: [ 1938.738482] brcmfmac: brcmf_sdio_read_control: last control frame is being processed.
Fri Sep 18 14:04:13 2020 kern.err kernel: [ 1941.298356] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
Fri Sep 18 14:04:13 2020 kern.err kernel: [ 1941.304198] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-110
Fri Sep 18 14:04:14 2020 kern.err kernel: [ 1941.723718] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Fri Sep 18 14:04:17 2020 kern.err kernel: [ 1944.733715] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Fri Sep 18 14:04:20 2020 kern.err kernel: [ 1947.723591] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Fri Sep 18 14:04:23 2020 kern.err kernel: [ 1950.723765] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Fri Sep 18 14:04:26 2020 kern.err kernel: [ 1953.722346] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Fri Sep 18 14:04:29 2020 kern.err kernel: [ 1956.722501] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Fri Sep 18 14:04:35 2020 kern.err kernel: [ 1962.723309] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Fri Sep 18 14:04:38 2020 kern.err kernel: [ 1965.723434] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Fri Sep 18 14:04:41 2020 kern.err kernel: [ 1968.722125] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Fri Sep 18 14:04:47 2020 kern.err kernel: [ 1974.723948] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Fri Sep 18 14:04:50 2020 kern.err kernel: [ 1977.730424] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Fri Sep 18 14:04:50 2020 kern.err kernel: [ 1977.740193] brcmfmac: brcmf_sdio_read_control: last control frame is being processed.
Fri Sep 18 14:04:50 2020 kern.err kernel: [ 1977.748046] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Fri Sep 18 14:04:53 2020 kern.err kernel: [ 1980.781697] brcmfmac: brcmf_cfg80211_dump_survey: Could not get noise (-512)
Fri Sep 18 14:04:56 2020 kern.err kernel: [ 1983.729976] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Fri Sep 18 14:04:56 2020 kern.err kernel: [ 1983.740242] brcmfmac: brcmf_sdio_read_control: last control frame is being processed.
Fri Sep 18 14:04:58 2020 kern.err kernel: [ 1986.259194] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
Fri Sep 18 14:04:58 2020 kern.err kernel: [ 1986.265037] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-110
Fri Sep 18 14:05:02 2020 kern.err kernel: [ 1989.734205] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Fri Sep 18 14:05:08 2020 kern.err kernel: [ 1995.722813] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Fri Sep 18 14:05:14 2020 kern.err kernel: [ 2001.725250] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Fri Sep 18 14:05:17 2020 kern.err kernel: [ 2004.729423] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Fri Sep 18 14:05:20 2020 kern.err kernel: [ 2007.722848] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Fri Sep 18 14:05:23 2020 kern.err kernel: [ 2010.730336] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
[... cut ...]
Fri Sep 18 14:29:58 2020 kern.err kernel: [ 3486.418116] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Fri Sep 18 14:30:01 2020 kern.err kernel: [ 3489.419549] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512

Modello	Raspberry Pi 4 Model B Rev 1.1
Architecture	ARMv7 Processor rev 3 (v7l)
Versione Firmware	openmptcprouter v0.55.3 r0+14198-da98603597 / LuCI Master git-20.219.57394-35e2735
Versione del Kernel	5.4.58

Ysurac, 2020-09-18:
Wifi is used ?

ascomputer, 2020-09-18:
no wifi, I'm using a mobile phone and a dongle

darthclide, 2020-09-21:
Fresh install. Same here. Not using WiFi and I guess I have to disable wifi at bootup again? I wish I didn't have to do that :/

Ysurac, 2020-09-21:
Did you update RPI4 firmware ? https://github.com/Ysurac/openmptcprouter/wiki/Update-RPI4-firmware

kevin39, 2020-09-21:
Hi @Ysurac,
FYI, I got the same error on my rpi4 last week. I didn't save the log.

On my side, the firmware is up-to-date (but not OMR which is 0.55.1) and wifi is never used.

Ysurac, 2020-09-21:
Firmware can only be up to date with latest OMR release :)

I have the same issue now, so I can make some test and try to fix that for next release.

darthclide, 2020-09-21:
Yeah, I just disabled the wifi in the /boot/config.txt and I no longer get this spam of ieee80211 phy0: brcmf_cfg80211 errors. I really am afraid to update my other raspberry pi now... (since it had stable wifi last time I used it)

kevin39, 2020-10-01:
Any update about this issue ? I got this error again on my rpi.

Ysurac, 2020-10-01:
This will be fixed in next release.

Mutanter, 2020-10-27:
Yay, got them too - even on the latest release (r56.0).
In my case I suspect they relate to some really short WiFi freezes, anyway I'am not complete sure about it.

Did you find [this](https://github.com/raspberrypi/linux/issues/2453) related post on you research?
The old (2015) broadcom firmware mentioned might worth a try?

Ysurac, 2020-10-27:
You updated RPI4 firmware ?
And what is the error ? Because in this issue wifi was disabled.

Mutanter, 2020-10-27:
Yes, updated according to your Post (https://github.com/Ysurac/openmptcprouter/wiki/Update-RPI4-firmware).

Error continues like this every 3-4s:
`Tue Oct 27 17:33:37 2020 kern.err kernel: [ 1084.798121] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-110
Tue Oct 27 17:33:40 2020 kern.err kernel: [ 1087.894968] ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Tue Oct 27 17:33:40 2020 kern.err kernel: [ 1087.903972] ieee80211 phy0: brcmf_cfg80211_get_station: GET STA INFO failed, -110
Tue Oct 27 17:33:40 2020 kern.err kernel: [ 1087.912480] ieee80211 phy0: brcmf_cfg80211_get_station: GET STA INFO failed, -512
Tue Oct 27 17:33:40 2020 kern.err kernel: [ 1087.913059] brcmfmac: brcmf_sdio_read_control: last control frame is being processed.
Tue Oct 27 17:33:40 2020 kern.err kernel: [ 1087.928493] brcmfmac: brcmf_cfg80211_dump_survey: chanspec failed (-512)`

Often without any impact but I still got about 1-2 complete random WiFi drops so far, without any other log-notification, while the wired connection seems to be OK all the time. Still under investigation, as the drops seem to be very short and sporadic.

Ysurac, 2020-10-27:
When wifi is used, this is in most case due to a not supported wifi configuration (check country, channel, power,...).

Mutanter, 2020-10-27:
Thanks for the quick response! Might have been the country code. The nasty thing is that a change seems to become only active after a complete reboot. By resetting the value to driver default the log looks much better!

ascomputer, 2020-10-28:
in this regard I want to write an experience of mine made these days, using a wifi router with the 4g connection I always had errors in the log file and at some point orm closed my wifi connection, then hotspot with the mobile phone the log file me  from some error but i notice that it doesn't close the connection.  so I think it may also depend on the wifi router you use.  obviously i set the wifi driver in the country i am hello to all.

closed 2021-01-31 by github-actions[bot]

weekends, 2021-03-03:
Is anybody resolve that issue?

darthclide, 2021-03-03:
> Is anybody resolve that issue?

A. If you are trying to get wifi to work, I know this isn't what you want to hear, but basically I followed Ysurac's instructions on this page: https://github.com/Ysurac/openmptcprouter/wiki/Configure-Wifi
Yes, you are seriously limiting the power and range of the wifi, but for me it works since I use it as a mobile bonder in my backpack.

B. If you don't care about having wifi, I resolved it by disabling the wifi completely
edit /boot/config.txt and add the following line:
dtoverlay=disable-wifi

Bumpalert, 2021-07-18:
I got this issue in 0.58.5 in rpi4b

Sun Jul 18 16:39:35 2021 kern.err kernel: [ 4961.750963] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Sun Jul 18 16:39:38 2021 kern.err kernel: [ 4964.751694] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Sun Jul 18 16:39:41 2021 kern.err kernel: [ 4967.751853] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Sun Jul 18 16:39:44 2021 kern.err kernel: [ 4970.762769] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Sun Jul 18 16:39:47 2021 kern.err kernel: [ 4973.751413] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Sun Jul 18 16:39:50 2021 kern.err kernel: [ 4976.771299] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Sun Jul 18 16:39:53 2021 kern.err kernel: [ 4979.753732] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Sun Jul 18 16:39:56 2021 kern.err kernel: [ 4982.755507] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Sun Jul 18 16:39:59 2021 kern.err kernel: [ 4985.751068] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Sun Jul 18 16:40:05 2021 kern.err kernel: [ 4991.750314] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Sun Jul 18 16:40:08 2021 kern.err kernel: [ 4994.750082] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Sun Jul 18 16:40:11 2021 kern.err kernel: [ 4997.752109] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Sun Jul 18 16:40:14 2021 kern.err kernel: [ 5000.751584] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Sun Jul 18 16:40:17 2021 kern.err kernel: [ 5003.751626] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Sun Jul 18 16:40:20 2021 kern.err kernel: [ 5006.752101] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Sun Jul 18 16:40:23 2021 kern.err kernel: [ 5009.753418] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Sun Jul 18 16:40:26 2021 kern.err kernel: [ 5012.761058] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Sun Jul 18 16:40:29 2021 kern.err kernel: [ 5015.749326] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Sun Jul 18 16:40:32 2021 kern.err kernel: [ 5018.753410] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Sun Jul 18 16:40:35 2021 kern.err kernel: [ 5021.754080] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Sun Jul 18 16:40:38 2021 kern.err kernel: [ 5024.754611] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Sun Jul 18 16:40:38 2021 daemon.err omr-tracker[14892]: Segmentation fault
Sun Jul 18 16:40:41 2021 kern.err kernel: [ 5027.751306] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Sun Jul 18 16:40:44 2021 kern.err kernel: [ 5030.755220] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Sun Jul 18 16:40:47 2021 kern.err kernel: [ 5033.764032] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512

Ysurac, 2021-07-18:
I had the same errors, I played in Network->Wireless and it's gone. I need to find what is new correct settings for that...

Ysurac, 2021-07-18:
Even if wifi is not used, these errors appears. This is my config :
```
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.path='platform/soc/fe300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
wireless.radio0.disabled='1'
wireless.radio0.country='FR'
wireless.radio0.cell_density='0'
wireless.radio0.hwmode='11g'
wireless.radio0.channel='3'
wireless.radio0.txpower='10'
wireless.default_radio0=wifi-iface
wireless.default_radio0.device='radio0'
wireless.default_radio0.ssid='OpenWrt'
wireless.default_radio0.encryption='none'
wireless.default_radio0.skip_inactivity_poll='1'
wireless.default_radio0.mode='sta'
```
I will make tests to find exactly what setting is needed.