Ysurac/openmptcprouter #1234: ip route list table 5(omrvpn) empty

terbonium, 2020-09-18:
## Expected Behavior

OMR bypass to OMRVPN (tun0/glorytun-udp) fails. route table 5 is empty
## Current Behavior

Bypassing traffic to omrvpn is marked with 0x5395 correctly, however traffic is forwarded to table 5 which is empty.

root@OpenMPTCProuter:~# ip route list table 5
Dump terminated

## Possible Solution

Adding a default route manually to table 5 fixes the problem:

root@OpenMPTCProuter:~# ip route add default dev tun0 table 5
root@OpenMPTCProuter:~# ip route list table 5
default dev tun0 scope link## Steps to Reproduce the Problem

  1. default config does not work, even after reboot
## Context (Environment)

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter platform:

terbonium, 2020-09-19:
I commented out from

/etc/init.d/mptctp (line 94)

```
# [ "$config" = "omrvpn" ] && return 0
```

This allows table 5 route to be updated, it also adds routes for tun0 to the main table as well.  I'm not sure what this might break:

```
root@OpenMPTCProuter:~# ip route list table 5
default via 10.255.255.1 dev tun0
10.255.255.2 dev tun0 scope link
```

Not sure why I'm getting 2 routes for the VPS on the same external interface with 2 metrics:
```
root@OpenMPTCProuter:~# ip route list
default via 10.255.255.1 dev tun0
default via 198.18.40.1 dev eth0.40 metric 3
default via 198.18.50.1 dev eth0.50 metric 4
default via 10.255.255.1 dev tun0 metric 5
default via 10.255.252.5 dev tun3 metric 7
10.255.252.1 via 10.255.252.5 dev tun3
10.255.252.5 dev tun3 proto kernel scope link src 10.255.252.6
10.255.252.6 dev tun3 scope link metric 7
10.255.255.1 dev tun0 proto kernel scope link src 10.255.255.2
10.255.255.2 dev tun0 scope link metric 5
5x.xx.xx.xx via 198.18.50.1 dev eth0.50 metric 1
5x.xx.xx.xx via 198.18.50.1 dev eth0.50 metric 4
5x.xx.xx.xx via 10.255.252.5 dev tun3 metric 7
127.0.0.0/8 dev lo proto static scope link metric 1
198.18.40.0/24 dev eth0.40 scope link metric 3
198.18.50.0/24 dev eth0.50 scope link metric 4
```

Ysurac, 2020-09-19:
I don't think this can break anything, I already modified that.

For metric, it's default configuration when not in master balancing mode. Interface with metric 1 is master interface.

closed 2021-01-04 by github-actions[bot]