Ysurac/openmptcprouter #1235: ipv6 on vps

skyfr54, 2020-09-19:
Bonjour, que faut-il faire pour activer ipv6 sur un serveur vps debian 10??? Je n'ai rien trouvé à ce sujet. Merci

Ysurac, 2020-09-19:
L'IPv6 doit être activé avant de lancer le script d'installation du VPS (même si uniquement des liens locaux type fe80 sont alors configurés).
Ensuite l'IPv6 doit être fonctionnel sur le VPS suivant ce qu'indique le fournisseur pour le configurer si ce n'est pas fait automatiquement (il doit pouvoir aller sur un site en IPv6).
Enfin dans le Système->OpenMPTCProuter sur le routeur, il faut cocher la case "Paramètres avancés" puis "activer IPv6".

skyfr54, 2020-09-30:
Merci, ça marche...je rencontre maintenant un pb de debit : voici le speedtest (le resultat est le meme que sur l'application) Pour l'upload c'est ok mais pour le download on est à la moitié de ce que je devrais avoir avec l'agrégation.J'ai la fibre orange 1gb qui plafonne à 800-850 mb et la 4g free qui m'envoie 120mb quand tout va bien....Mon vps est chez ohv est a une bande passante de 1gbOù est le problème?? J'ai suivi toutes vos recommandations mais rien n'y fait....
[image]

Adorfer, 2020-10-01:
as far as i understand, onkla-speedtest only uses ipv4, so whatever speedtest.net shows, it's NOT related to ipv6!

skyfr54, 2020-10-01:
I know but when I use only wan 1 connection (only my fiber router) I have more than 800mb in download, so I must have at least this value with my agregation, whatever ipv4 or ipv6

kevin39, 2020-10-01:
There may be some reasons : 
- Hardware limitations on OMR side
- Bandwidth limitations on VPS side (are you sure you 1gb bandwidth is guaranteed) ?
Maybe @Ysurac will have more tips to help you.

Ysurac, 2020-10-01:
Hardware limitation on VPS side is also possble.
Check load average during a speed test using top, htop,... on both side.

Adorfer, 2020-10-01:
> I know but when I use only wan 1 connection (only my fiber router) I have more than 800mb in download, so I must have at least this value with my agregation, whatever ipv4 or ipv6

But how is this related to IPv6 on the VPS? Does it change if you disable V6 on the VPS and rerun the installscript?

skyfr54, 2020-10-02:
No it changes nothing Provenance : Courrier pour Windows 10 De : AdorferEnvoyé le :jeudi 1 octobre 2020 19:28À : Ysurac/openmptcprouterCc : jean-philippe sartori; AuthorObjet :Re: [Ysurac/openmptcprouter] ipv6 on vps (#1235) I know but when I use only wan 1 connection (only my fiber router) I have more than 800mb in download, so I must have at least this value with my agregation, whatever ipv4 or ipv6But how is this related to IPv6 on the VPS? Does it change if you disable V6 on the VPS and rerun the installscript?

skyfr54, 2020-10-02:
I will do it Provenance : Courrier pour Windows 10 De : Ycarus (Yannick Chabanois)Envoyé le :jeudi 1 octobre 2020 10:28À : Ysurac/openmptcprouterCc : jean-philippe sartori; AuthorObjet :Re: [Ysurac/openmptcprouter] ipv6 on vps (#1235) Hardware limitation on VPS side is also possble.Check load average during a speed test using top, htop,... on both side.

Adorfer, 2020-10-05:
for ipv6 issues on the vps: 

eventhough the script has just been installed, after the installation of the new kernel and the reboot: 
I had 2 times on a "new vps host" the problem: ipv6 was NOT working on the OMR (as reported on status page, and as seen by real LAN clients)

**Just re-run the install script on the vps a second time!**
then (without reboot on the OMR!) ipv6 became working.

closed 2021-01-09 by github-actions[bot]