Ysurac/openmptcprouter #1236: ip issues

zulu14, 2020-09-19:
hello, i have reinstall an another omr instance on a two ip vps, an i have an issues, 

i have install the script vps after check the correct working of the two interface of each ip (ens18 && ens19) 

i have setup the master ip ***.***.***.227 into the wizard 
the second ip is the same with ***.***.***.228 at the end, 

my omr not create 'the tunnel", its not retryve the token, (and its show me a direct ouput with the  ***.***.***.228 ip ) 

i think he has in issues with the 227 and 228 (my public ip on browser are the master modem , but ont the 
openmptcprouter.omr.detected_ss_ipv4 and openmptcprouter.omr.detected_public_ipv4 its the ***.***.***.228)

## Specifications

  - OpenMPTCProuter version: 55.3
  - OpenMPTCProuter VPS version: 0.1017
  - OpenMPTCProuter platform: rpi4

config:

hello, i have reinstall an another omr instance on a two ip vps, an i have an issues, 

i have install the script vps after check the correct working of the two interface of each ip (ens18 && ens19) 

i have setup the master ip ***.***.***.227 into the wizard 
the second ip is the same with ***.***.***.228 at the end, 

my omr not create 'the tunnel", its not retryve the token, (and its show me a direct ouput with the  ***.***.***.228 ip ) 

i think he has in issues with the 227 and 228 (my public ip on browser are the master modem , but ont the 
openmptcprouter.omr.detected_ss_ipv4 and openmptcprouter.omr.detected_public_ipv4 its the ***.***.***.228)

## Specifications

  - OpenMPTCProuter version: 55.3
  - OpenMPTCProuter VPS version: 0.1017
  - OpenMPTCProuter platform: rpi4

dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].boguspriv='1'
dhcp.@dnsmasq[0].filterwin2k='0'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].resolvfile='/tmp/resolv.conf.d/resolv.conf.auto'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].server='127.0.0.1#5353' '/lan/' '/use-application-dns.net/'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].rebind_domain='plex.direct'
dhcp.@dnsmasq[0].ipset='/googlevideo.com/omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/steamcontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.lan.force='1'
dhcp.lan.ra_slaac='1'
dhcp.lan.ra_flags='managed-config' 'other-config'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
[... 1523 lines cut ...]
upnpd.@perm_rule[0].ext_ports='1024-65535'
upnpd.@perm_rule[0].int_addr='0.0.0.0/0'
upnpd.@perm_rule[0].int_ports='1024-65535'
upnpd.@perm_rule[0].comment='Allow high ports'
upnpd.@perm_rule[1]=perm_rule
upnpd.@perm_rule[1].action='deny'
upnpd.@perm_rule[1].ext_ports='0-65535'
upnpd.@perm_rule[1].int_addr='0.0.0.0/0'
upnpd.@perm_rule[1].int_ports='0-65535'
upnpd.@perm_rule[1].comment='Default deny'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='wan1' 'wan2'
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.channel='36'
wireless.radio0.hwmode='11a'
wireless.radio0.path='platform/soc/fe300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
wireless.radio0.htmode='VHT80'
wireless.radio0.disabled='1'
wireless.default_radio0=wifi-iface
wireless.default_radio0.device='radio0'
wireless.default_radio0.network='lan'
wireless.default_radio0.mode='ap'
wireless.default_radio0.ssid='OpenWrt'
wireless.default_radio0.encryption='none'

zulu14, 2020-09-19:
[image]

Ysurac, 2020-09-19:
> my omr not create 'the tunnel", its not retryve the token
I don't understand this. VPN seems to be up and you output via a VPS IP.

I only see a display error as it's not a direct output. It's the issue ?

zulu14, 2020-09-19:
the uci values show public ip vps, but its wrong  
[image]

[image]

my vps ip are xxx.xxxx.xxx.227 and  xxx.xxxx.xxx.228 i have set the first ip of vps ens18 interface (xxx.xxxx.xxx.227) in wizard

Ysurac, 2020-09-19:
OpenMPTCProuter is used as gateway ? What is the result of ```curl -4 ifconfig.co``` via ssh on the router ?

zulu14, 2020-09-19:
[image]
[image]

yes the fateway are omr ip (192.168.100.1) 

no host resolv for ifconfig.co

zulu14, 2020-09-19:
[image]

zulu14, 2020-09-19:
[image]

Ysurac, 2020-09-19:
There is no issue in the fact that .228 ip is used as output. It's the default output IP on your VPS.

But there is a strange issue with "ifconfig.co" that can't be resolved... Can you check State->System log ?

zulu14, 2020-09-19:
its has an most important issues,
1- my public ip are not one of the two vps ip, 
2- the port of my two ip not working if i want to redirect the omr interface on world for example,
3- the status show an direct output, 
4- my trafic are no agregated

zulu14, 2020-09-19:
LOG:

Sat Sep 19 17:53:53 2020 user.notice post-tracking-post-tracking: omrvpn down. Replace default route by 192.168.0.1 dev wan1
Sat Sep 19 17:53:54 2020 user.notice Shadowsocks: Rules UP
Sat Sep 19 17:53:54 2020 user.notice Shadowsocks: Reload omr-bypass rules
Sat Sep 19 17:53:54 2020 user.notice omr-bypass: Starting OMR-ByPass...
Sat Sep 19 17:53:54 2020 daemon.err glorytun[27139]: 51.178.122.227.65001: connected
Sat Sep 19 17:53:55 2020 daemon.err glorytun[27139]: 51.178.122.227.65001: key exchange failed
Sat Sep 19 17:53:55 2020 daemon.info glorytun[27139]: STOPPED tun0
Sat Sep 19 17:53:55 2020 user.notice omr-bypass: Reload dnsmasq...
Sat Sep 19 17:53:55 2020 user.notice post-tracking-post-tracking: Set server vps (51.178.122.227) default route 51.178.122.227 nexthop via 192.168.0.1 dev wan1 weight 10 nexthop via 192.168.1.1 dev wan2 weight 1
Sat Sep 19 17:53:55 2020 daemon.info dnsmasq[1460]: read /etc/hosts - 4 addresses
Sat Sep 19 17:53:55 2020 daemon.info dnsmasq[1460]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Sat Sep 19 17:53:55 2020 daemon.info dnsmasq-dhcp[1460]: read /etc/ethers - 0 addresses
Sat Sep 19 17:53:55 2020 user.notice omr-bypass: OMR-ByPass is running
Sat Sep 19 17:53:55 2020 user.info : using tcp fast open
Sat Sep 19 17:53:55 2020 daemon.info /usr/bin/ss-redir[28934]: enable TCP no-delay
Sat Sep 19 17:53:55 2020 daemon.info /usr/bin/ss-redir[28934]: resolving hostname to IPv6 address first
Sat Sep 19 17:53:55 2020 daemon.info /usr/bin/ss-redir[28934]: initializing ciphers... chacha20-ietf-poly1305
Sat Sep 19 17:53:55 2020 daemon.info /usr/bin/ss-redir[28934]: listening at 0.0.0.0:1100
Sat Sep 19 17:53:55 2020 daemon.info /usr/bin/ss-redir[28934]: tcp port reuse enabled
Sat Sep 19 17:53:55 2020 daemon.info /usr/bin/ss-redir[28934]: UDP relay enabled
Sat Sep 19 17:53:55 2020 daemon.err /usr/bin/ss-redir[28934]: [udp] getaddrinfo: System error
Sat Sep 19 17:53:55 2020 daemon.info /usr/bin/ss-redir[28934]: running from root user
Sat Sep 19 17:53:55 2020 daemon.info /usr/bin/ss-local[28932]: using tcp fast open
Sat Sep 19 17:53:55 2020 daemon.info /usr/bin/ss-local[28932]: enable TCP no-delay
Sat Sep 19 17:53:55 2020 daemon.info /usr/bin/ss-local[28932]: initializing ciphers... chacha20-ietf-poly1305
Sat Sep 19 17:53:55 2020 daemon.info /usr/bin/ss-local[28932]: listening at 127.0.0.1:1111
Sat Sep 19 17:53:55 2020 daemon.info /usr/bin/ss-local[28932]: tcp port reuse enabled
Sat Sep 19 17:53:55 2020 daemon.info /usr/bin/ss-local[28932]: udprelay enabled
Sat Sep 19 17:53:55 2020 daemon.err /usr/bin/ss-local[28932]: [udp] getaddrinfo: System error
Sat Sep 19 17:53:55 2020 daemon.info /usr/bin/ss-local[28932]: running from root user
Sat Sep 19 17:53:55 2020 user.info : using tcp fast open
Sat Sep 19 17:53:55 2020 daemon.info /usr/bin/ss-redir[28935]: enable TCP no-delay
Sat Sep 19 17:53:55 2020 daemon.info /usr/bin/ss-redir[28935]: initializing ciphers... chacha20-ietf-poly1305
Sat Sep 19 17:53:55 2020 daemon.info /usr/bin/ss-redir[28935]: listening at 0.0.0.0:1101
Sat Sep 19 17:53:55 2020 daemon.info /usr/bin/ss-redir[28935]: tcp port reuse enabled
Sat Sep 19 17:53:55 2020 daemon.info /usr/bin/ss-redir[28935]: UDP relay enabled
Sat Sep 19 17:53:55 2020 daemon.err /usr/bin/ss-redir[28935]: [udp] getaddrinfo: System error
Sat Sep 19 17:53:55 2020 daemon.info /usr/bin/ss-redir[28935]: running from root user
Sat Sep 19 17:53:55 2020 user.info : using tcp fast open
Sat Sep 19 17:53:55 2020 daemon.info /usr/bin/ss-redir[28936]: enable TCP no-delay
Sat Sep 19 17:53:55 2020 daemon.info /usr/bin/ss-redir[28936]: initializing ciphers... chacha20-ietf-poly1305
Sat Sep 19 17:53:55 2020 daemon.info /usr/bin/ss-redir[28936]: listening at 0.0.0.0:1101
Sat Sep 19 17:53:55 2020 daemon.info /usr/bin/ss-redir[28936]: tcp port reuse enabled
Sat Sep 19 17:53:55 2020 daemon.info /usr/bin/ss-redir[28936]: UDP relay enabled
Sat Sep 19 17:53:55 2020 daemon.err /usr/bin/ss-redir[28936]: [udp] getaddrinfo: System error
Sat Sep 19 17:53:55 2020 daemon.info /usr/bin/ss-redir[28936]: running from root user
Sat Sep 19 17:53:55 2020 user.info : using tcp fast open
Sat Sep 19 17:53:55 2020 daemon.info /usr/bin/ss-redir[28933]: enable TCP no-delay
Sat Sep 19 17:53:55 2020 daemon.info /usr/bin/ss-redir[28933]: resolving hostname to IPv6 address first
Sat Sep 19 17:53:55 2020 daemon.info /usr/bin/ss-redir[28933]: initializing ciphers... chacha20-ietf-poly1305
Sat Sep 19 17:53:55 2020 daemon.info /usr/bin/ss-redir[28933]: listening at 0.0.0.0:1100
Sat Sep 19 17:53:55 2020 daemon.info /usr/bin/ss-redir[28933]: tcp port reuse enabled
Sat Sep 19 17:53:55 2020 daemon.info /usr/bin/ss-redir[28933]: UDP relay enabled
Sat Sep 19 17:53:55 2020 daemon.err /usr/bin/ss-redir[28933]: [udp] getaddrinfo: System error
Sat Sep 19 17:53:55 2020 daemon.info /usr/bin/ss-redir[28933]: running from root user
Sat Sep 19 17:53:55 2020 daemon.err glorytun[27139]: getpeername: Socket not connected
Sat Sep 19 17:53:55 2020 daemon.notice netifd: Network device 'tun0' link is down
Sat Sep 19 17:53:55 2020 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Sat Sep 19 17:53:55 2020 daemon.notice netifd: Interface 'omrvpn' is now down
Sat Sep 19 17:53:55 2020 daemon.info glorytun: starting glorytun vpn instance vpn
Sat Sep 19 17:53:55 2020 user.notice ModemManager: hotplug: remove network interface tun0: event processed
[... cut ...]
Sat Sep 19 18:42:13 2020 daemon.info dnsmasq-dhcp[12161]: DHCPOFFER(eth0) 192.168.100.135 54:ec:2f:15:56:f0
Sat Sep 19 18:42:13 2020 daemon.info dnsmasq-dhcp[12161]: DHCPDISCOVER(eth0) 94:9f:3e:aa:c5:3c
Sat Sep 19 18:42:13 2020 daemon.info dnsmasq-dhcp[12161]: DHCPOFFER(eth0) 192.168.100.157 94:9f:3e:aa:c5:3c
Sat Sep 19 18:42:15 2020 daemon.info dnsmasq-dhcp[12161]: DHCPDISCOVER(eth0) 94:9f:3e:aa:c5:3c
Sat Sep 19 18:42:15 2020 daemon.info dnsmasq-dhcp[12161]: DHCPOFFER(eth0) 192.168.100.157 94:9f:3e:aa:c5:3c

zulu14, 2020-09-19:
[image]

Ysurac, 2020-09-19:
No related errors in log.

Hum this resolve here ? strange...

Can you give me the result of ```ip r``` and ```ip a``` via SSH on the router ?

zulu14, 2020-09-19:
[image]

zulu14, 2020-09-19:
on the sys log he has:

Sat Sep 19 17:53:55 2020 daemon.err glorytun[27139]: XXX.XXX.XXX.227.65001: key exchange failed

Sat Sep 19 17:53:57 2020 daemon.err glorytun[28996]: XXX.XXX.XXX.65001: key exchange failed
Sat Sep 19 17:53:58 2020 daemon.err glorytun[28996]: XXX.XXX.XXX.65001: key exchange failed
Sat Sep 19 17:54:00 2020 daemon.err glorytun[28996]: XXX.XXX.XXX.227.65001: key exchange failed
and so on...

Sat Sep 19 17:54:09 2020 user.notice OMR-VPS: Can't get token, try later
Sat Sep 19 17:54:09 2020 user.notice OMR-VPS: Can't get token, try later
Sat Sep 19 17:54:10 2020 user.notice OMR-VPS: Can't get token, try later (can't ping server, no server API answer)
Sat Sep 19 17:54:10 2020 user.notice OMR-VPS: Can't get token, try later
Sat Sep 19 17:54:10 2020 user.notice OMR-VPS: Can't get token, try later
Sat Sep 19 17:54:10 2020 user.notice OMR-VPS: Can't get token, try later

zulu14, 2020-09-19:
and this : Sat Sep 19 18:53:12 2020 daemon.err omr-tracker[9369]: Failed to send request: Operation not permitted
Sat Sep 19 18:53:18 2020 daemon.err omr-tracker[9369]: Failed to send request: Operation not permitted

Ysurac, 2020-09-19:
So there is a problem with the tunnel. Try to reboot the VPS and then the router.
If no changes, check on the VPS log in ```/var/log/daemon.log``` related to glorytun.

zulu14, 2020-09-19:
i have reboot the omr and the vps, 
now the tunnel seem "ok" 

[image]

when i ping a domain : 
[image]

when i ping an ip : 

[image]

zulu14, 2020-09-19:
( i need to switch to static local ip of one of my modem to send message to you )

zulu14, 2020-09-19:
var log message : https://codeshare.io/2Bov6y

Ysurac, 2020-09-19:
Strange glorytun-udp restart in log. But this doesn't look like daemon.log

Can you ```ping 10.255.255.1``` ?

zulu14, 2020-09-19:
i can 
[image]

i have try to restart shorewall ( systemctl restart shorewall) 
and 
[image]

/etc/shorewall/snat :
#
MASQUERADE		10.255.0.0/16,\
			169.254.0.0/16,\
			172.16.0.0/12,\
			192.168.0.0/16		$NET_IFACE
# SNAT from VPN server for all VPN clients
#SNAT($VPS_ADDR)		0.0.0.0/0		$VPS_IFACE
SNAT(xxx.xxx.xxx.227)	10.255.249.0/30	ens18 # OMR GRE for public IP xxx.xxx.xxx.227 for user openmptcprouter
SNATxxx.xxx.xxx.228)	10.255.249.4/30	ens19 # OMR GRE for public IP xxx.xxx.xxx.228 for user openmptcprouter

zulu14, 2020-09-19:
codeshare about ifconfig if you need : https://codeshare.io/2jy7jD

Ysurac, 2020-09-20:
ok, the issue is that the IPs are on different interfaces, this is not a tested case and so it's the problem.
You need in ```/etc:shorewall/interfaces``` to add ```net    ens18   dhcp,tcpflags,routefilter,nosmurfs,sourceroute=0``` and restart shorewall and all should work.

zulu14, 2020-09-20:
hello, 
my ip are now correct, but i have an dns issues message on the status page (dns of my computer set to 192.168.100.1) (i can reach 1.1.1.1 of direct server ip)

Sun Sep 20 11:22:13 2020 daemon.notice netifd: Interface 'omrvpn' has link connectivity
Sun Sep 20 11:22:13 2020 daemon.notice netifd: Interface 'omrvpn' is setting up now
Sun Sep 20 11:22:13 2020 daemon.info glorytun[5409]: INITIALIZED tun0
Sun Sep 20 11:22:13 2020 daemon.notice netifd: Interface 'omrvpn' is now up
Sun Sep 20 11:22:13 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty62, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty62
Sun Sep 20 11:22:13 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty62, subsystem=tty
Sun Sep 20 11:22:13 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun Sep 20 11:22:13 2020 user.notice ModemManager: hotplug: add network interface tun0: event processed
Sun Sep 20 11:22:13 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tun0, subsystem=net
Sun Sep 20 11:22:13 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty24, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty24
Sun Sep 20 11:22:13 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty24, subsystem=tty
Sun Sep 20 11:22:13 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun Sep 20 11:22:14 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun Sep 20 11:22:14 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty8, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty8
Sun Sep 20 11:22:14 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty8, subsystem=tty
Sun Sep 20 11:22:14 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun Sep 20 11:22:14 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty52, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty52
Sun Sep 20 11:22:14 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty52, subsystem=tty
Sun Sep 20 11:22:14 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun Sep 20 11:22:14 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty14, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty14
Sun Sep 20 11:22:14 2020 user.notice firewall: Reloading firewall due to ifup of wan2 (wan2)
Sun Sep 20 11:22:14 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty14, subsystem=tty
Sun Sep 20 11:22:14 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun Sep 20 11:22:14 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty42, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty42
Sun Sep 20 11:22:14 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty42, subsystem=tty
Sun Sep 20 11:22:14 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun Sep 20 11:22:14 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty32, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty32
Sun Sep 20 11:22:14 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty32, subsystem=tty
Sun Sep 20 11:22:14 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun Sep 20 11:22:14 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty60, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty60
Sun Sep 20 11:22:14 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty60, subsystem=tty
Sun Sep 20 11:22:14 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun Sep 20 11:22:14 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty22, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty22
Sun Sep 20 11:22:14 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty22, subsystem=tty
Sun Sep 20 11:22:14 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun Sep 20 11:22:14 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty6, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty6
Sun Sep 20 11:22:14 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty6, subsystem=tty
Sun Sep 20 11:22:14 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun Sep 20 11:22:14 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty50, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty50
Sun Sep 20 11:22:14 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty50, subsystem=tty
Sun Sep 20 11:22:14 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun Sep 20 11:22:14 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty12, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty12
Sun Sep 20 11:22:14 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty12, subsystem=tty
Sun Sep 20 11:22:14 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
[... cut ...]
Sun Sep 20 11:32:37 2020 daemon.info dnsmasq-dhcp[12575]: DHCPOFFER(eth0) 192.168.100.158 94:9f:3e:aa:c5:3c
Sun Sep 20 11:32:37 2020 daemon.info dnsmasq-dhcp[12575]: DHCPDISCOVER(eth0) 94:9f:3e:aa:c5:3c
Sun Sep 20 11:32:37 2020 daemon.info dnsmasq-dhcp[12575]: DHCPOFFER(eth0) 192.168.100.158 94:9f:3e:aa:c5:3c
Sun Sep 20 11:32:37 2020 daemon.info dnsmasq-dhcp[12575]: DHCPDISCOVER(eth0) 94:9f:3e:aa:c5:3c
Sun Sep 20 11:32:37 2020 daemon.info dnsmasq-dhcp[12575]: DHCPOFFER(eth0) 192.168.100.158 94:9f:3e:aa:c5:3c

zulu14, 2020-09-20:
my network/interface file: 
[image]

zulu14, 2020-09-20:
if the fact that there are two interfaces is too much of a problem in the vps (although i don't think that is the cause of the dns,) we can put them together on one interface?

zulu14, 2020-09-20:
i have activated the proxy dns and its work

zulu14, 2020-09-20:
i have try multiple ip feature and he has broken the tunnel,  (after setup the SS config, i have restart omr and vps ) 

[image]

[image]

[image]

[image]

sys log: https://codeshare.io/5O1Xx7

zulu14, 2020-09-20:
if i disable the local instance, the server and the redirect and the system not re UP

zulu14, 2020-09-20:
[image]
[image]
[image]
[image]

zulu14, 2020-09-20:
i have try to re exe the controller of omr wizard with the wizard form, and its not fix the issues

my all uci data after disable all shadowsock i have enabled with the wiki tutorial, https://codeshare.io/G6Ow1o

Ysurac, 2020-09-20:
You made changes in shorewall config ?

zulu14, 2020-09-20:
yes

zulu14, 2020-09-20:
[image]

Ysurac, 2020-09-20:
And you restarted shorewall and no more error on restart ?

zulu14, 2020-09-20:
[image]

zulu14, 2020-09-20:
[image]

Ysurac, 2020-09-20:
ok this seems good but shadowsocks and VPN can't connect now ? What do you have in VPS log and router log now ?

zulu14, 2020-09-20:
my var log message: https://codeshare.io/21gXRn
my sys log omr : https://codeshare.io/5glPQ8

Ysurac, 2020-09-20:
Many martian source error, you know what are these 37.187.* IPs ?

For OMR it seems to say that shadowsocks and tunnel is up at end of the log.

zulu14, 2020-09-20:
37.187 is the first part of the hypervisor of the kvm mv running the vps script

zulu14, 2020-09-21:
i re ping up about the multiple ip issues, when i start the ss local and ss remote the status page are green, when i turn up the ss redirect all down

zulu14, 2020-09-21:
[image]
[image]
[image]

Ysurac, 2020-09-22:
Before using multiples IPs, default is working now ?

zulu14, 2020-09-22:
yes now its working , the 37.187 martian error is on datacenter hardware , i will tell us

zulu14, 2020-09-23:
[image]

before enable shadowsocks remote and local instance for IP2 https://codeshare.io/296nb7
after: https://codeshare.io/5ekvP9
(as that did not work I deliberately did not activate the ss redir)

Ysurac, 2020-09-23:
I'm not able to reproduce the issue.
Can you check if you have error about shadowsocks in State->System log and via SSH, the file ```/tmp/etc/shadowsocks-libev/ss_redir.omrip1.json```?

zulu14, 2020-09-23:
[image]
(my fist ip end by 227 ans my second ip by 228 , its not 228 but 227, i think is normal )

zulu14, 2020-09-23:
(the key are correct)

zulu14, 2020-09-23:
i need to check the redir H1 H2 H3 H4 ?

Ysurac, 2020-09-23:
You can check one.
You can also try to enable verbose mode on redir.
What is the result of ```ps w | grep ss``` on the router via SSH ?

zulu14, 2020-09-23:
[image]
(i have actualy disable the local and remote instance for ip2, i its activated i d'ont have internet)
"You can also try to enable verbose mode on redir." on all redir or only h1 ?

zulu14, 2020-09-23:
(i have actualy disable the local and remote instance for ip2, i its activated i d'ont have internet)
"You can also try to enable verbose mode on redir." on all redir or only h1 ?
[image]

Ysurac, 2020-09-23:
on all or enable only hi1.
I want the ps when it's not working.

zulu14, 2020-09-23:
okay i do it

zulu14, 2020-09-23:
[image]
[image]
ss_redir.omrip1.json
[image]

zulu14, 2020-09-23:
do you want an remote control?

Ysurac, 2020-09-23:
No, I will go to bed ;)
What do you have in State->System log about shadowsocks/ss-redir ?

zulu14, 2020-09-23:
okay i wish you a good night :)   
the sys log:
Wed Sep 23 22:11:36 2020 daemon.info /usr/bin/ss-redir[28897]: redir to 54.69.57.43:443, len=819, recv=819
Wed Sep 23 22:11:37 2020 daemon.info /usr/bin/ss-redir[28896]: redir to 104.123.23.222:80, len=213, recv=213
Wed Sep 23 22:11:37 2020 daemon.info /usr/bin/ss-redir[28897]: redir to 35.227.237.213:443, len=98, recv=98
Wed Sep 23 22:11:37 2020 daemon.info /usr/bin/ss-redir[28897]: redir to 35.227.237.213:443, len=31, recv=31
Wed Sep 23 22:11:37 2020 daemon.info /usr/bin/ss-redir[28896]: redir to 35.227.237.213:443, len=180, recv=180
Wed Sep 23 22:11:37 2020 daemon.info /usr/bin/ss-redir[28896]: redir to 35.227.237.213:443, len=75, recv=75
Wed Sep 23 22:11:37 2020 daemon.info /usr/bin/ss-redir[28896]: redir to 35.227.237.213:443, len=51, recv=51
Wed Sep 23 22:11:37 2020 daemon.info /usr/bin/ss-redir[28896]: redir to 35.227.237.213:443, len=450, recv=450
Wed Sep 23 22:11:37 2020 daemon.info /usr/bin/ss-redir[28898]: redir to 35.190.43.134:443, len=517, recv=517
Wed Sep 23 22:11:37 2020 daemon.info /usr/bin/ss-redir[28898]: redir to 35.190.43.134:443, len=64, recv=64
Wed Sep 23 22:11:37 2020 daemon.info /usr/bin/ss-redir[28896]: redir to 35.227.237.213:443, len=92, recv=92
Wed Sep 23 22:11:37 2020 daemon.info /usr/bin/ss-redir[28896]: redir to 35.227.237.213:443, len=2116, recv=2116
Wed Sep 23 22:11:37 2020 daemon.info /usr/bin/ss-redir[28896]: redir to 52.18.137.75:443, len=2006, recv=2006
Wed Sep 23 22:11:38 2020 daemon.info /usr/bin/ss-redir[28897]: redir to 54.69.57.43:443, len=1424, recv=1424
Wed Sep 23 22:11:38 2020 daemon.info /usr/bin/ss-redir[28896]: redir to 35.227.237.213:443, len=98, recv=98
Wed Sep 23 22:11:38 2020 daemon.info /usr/bin/ss-redir[28896]: redir to 35.227.237.213:443, len=31, recv=31
Wed Sep 23 22:11:38 2020 daemon.info /usr/bin/ss-redir[28899]: redir to 35.227.237.213:443, len=180, recv=180
Wed Sep 23 22:11:38 2020 daemon.info /usr/bin/ss-redir[28899]: redir to 35.227.237.213:443, len=75, recv=75
Wed Sep 23 22:11:38 2020 daemon.info /usr/bin/ss-redir[28899]: redir to 35.227.237.213:443, len=51, recv=51
Wed Sep 23 22:11:38 2020 daemon.info /usr/bin/ss-redir[28899]: redir to 35.227.237.213:443, len=450, recv=450
Wed Sep 23 22:11:38 2020 daemon.info /usr/bin/ss-redir[28899]: redir to 35.227.237.213:443, len=92, recv=92
Wed Sep 23 22:11:38 2020 daemon.info /usr/bin/ss-redir[28896]: redir to 52.18.137.75:443, len=1963, recv=1963
Wed Sep 23 22:11:38 2020 daemon.info /usr/bin/ss-redir[28899]: redir to 35.227.237.213:443, len=2116, recv=2116
Wed Sep 23 22:11:38 2020 daemon.info /usr/bin/ss-redir[28897]: redir to 54.69.57.43:443, len=456, recv=456
Wed Sep 23 22:11:39 2020 daemon.info /usr/bin/ss-local[28895]: connection from 127.0.0.1:56869
Wed Sep 23 22:11:39 2020 daemon.info /usr/bin/ss-local[28895]: connect to 1.1.1.1:80
Wed Sep 23 22:11:39 2020 daemon.info /usr/bin/ss-local[28895]: remote: xx.xxx.xxx.227:65101
Wed Sep 23 22:11:41 2020 daemon.info /usr/bin/ss-redir[28896]: redir to 52.18.137.75:443, len=2006, recv=2006
Wed Sep 23 22:11:41 2020 daemon.info /usr/bin/ss-redir[28899]: redir to 35.227.237.213:443, len=2116, recv=2116
Wed Sep 23 22:11:43 2020 daemon.info /usr/bin/ss-redir[28899]: redir to 35.227.237.213:443, len=2117, recv=2117
Wed Sep 23 22:11:43 2020 daemon.info /usr/bin/ss-redir[28896]: redir to 52.18.137.75:443, len=1961, recv=1961
Wed Sep 23 22:11:43 2020 daemon.info /usr/bin/ss-redir[28896]: redir to 35.186.197.135:443, len=596, recv=596
Wed Sep 23 22:11:43 2020 daemon.info /usr/bin/ss-redir[28896]: redir to 35.186.197.135:443, len=64, recv=64
Wed Sep 23 22:11:43 2020 daemon.info /usr/bin/ss-redir[28896]: redir to 35.186.197.135:443, len=92, recv=92
Wed Sep 23 22:11:43 2020 daemon.info /usr/bin/ss-redir[28896]: redir to 35.186.197.135:443, len=31, recv=31
Wed Sep 23 22:11:43 2020 daemon.info /usr/bin/ss-redir[28899]: redir to 54.197.108.67:443, len=517, recv=517
Wed Sep 23 22:11:44 2020 daemon.info /usr/bin/ss-redir[28899]: redir to 54.197.108.67:443, len=126, recv=126
Wed Sep 23 22:11:44 2020 daemon.info /usr/bin/ss-local[28895]: connection from 127.0.0.1:56877
Wed Sep 23 22:11:44 2020 daemon.info /usr/bin/ss-local[28895]: connect to 1.0.0.1:80
Wed Sep 23 22:11:44 2020 daemon.info /usr/bin/ss-local[28895]: remote: xx.xxx.xxx.227:65101
Wed Sep 23 22:11:44 2020 daemon.info /usr/bin/ss-redir[28899]: redir to 54.197.108.67:443, len=1483, recv=1483
Wed Sep 23 22:11:45 2020 daemon.info /usr/bin/ss-redir[28897]: redir to 54.236.86.96:443, len=170, recv=170
Wed Sep 23 22:11:45 2020 daemon.info /usr/bin/ss-redir[28897]: redir to 54.236.86.96:443, len=126, recv=126
Wed Sep 23 22:11:46 2020 daemon.info /usr/bin/ss-redir[28897]: redir to 54.236.86.96:443, len=806, recv=806
Wed Sep 23 22:11:46 2020 daemon.info /usr/bin/ss-redir[28899]: redir to 51.124.78.146:443, len=210, recv=210
Wed Sep 23 22:11:46 2020 daemon.info /usr/bin/ss-redir[28899]: redir to 51.124.78.146:443, len=126, recv=126
Wed Sep 23 22:11:46 2020 daemon.info /usr/bin/ss-redir[28899]: redir to 51.124.78.146:443, len=947, recv=947
Wed Sep 23 22:11:46 2020 daemon.info /usr/bin/ss-redir[28897]: redir to 54.236.86.96:443, len=31, recv=31
Wed Sep 23 22:11:46 2020 daemon.info /usr/bin/ss-redir[28896]: redir to 52.18.137.75:443, len=72, recv=72
Wed Sep 23 22:11:47 2020 daemon.info /usr/bin/ss-redir[28898]: redir to 13.82.93.245:443, len=186, recv=186
Wed Sep 23 22:11:47 2020 daemon.info /usr/bin/ss-redir[28899]: redir to 216.58.215.42:443, len=39, recv=39
[... cut ...]
Wed Sep 23 22:14:20 2020 daemon.info /usr/bin/ss-local[28895]: remote: xx.xxx.xxx.227:65101
Wed Sep 23 22:14:20 2020 daemon.info /usr/bin/ss-redir[28898]: redir to 172.67.69.162:443, len=31, recv=31
Wed Sep 23 22:14:21 2020 daemon.info /usr/bin/ss-redir[28897]: redir to 35.186.224.25:443, len=184, recv=184
Wed Sep 23 22:14:21 2020 daemon.info /usr/bin/ss-redir[28897]: redir to 35.186.224.25:443, len=93, recv=93
Wed Sep 23 22:14:21 2020 daemon.info /usr/bin/ss-redir[28897]: redir to 35.186.224.25:443, len=175, recv=175

zulu14, 2020-09-23:
if you want to go to bed no problem we will resume tomorrow if you can and if you want, good night to you!

[42 more replies]

closed 2021-01-09 by github-actions[bot]