Ysurac/openmptcprouter #1237: Can't get public IP adress from ShadowSocks

Nethary, 2020-09-20:
Hello there!

First, thank you so much Ysurac for your work, that change my life ^^

## Context (Environment)
I have an issue with my OpenMPTCP, that cause some lag in my online games and web use.
I have this on the Etat page :
[image]

And this error flood my routeur log :
[image]

## Specifications

  - OpenMPTCProuter version: v0.55.3 
  - OpenMPTCProuter VPS version: V0.1017
  - OpenMPTCProuter platform: Raspberry Pi 3 Model B Plus Rev 1.3 x86_64

My router configuration:
dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].boguspriv='1'
dhcp.@dnsmasq[0].filterwin2k='0'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].server='127.0.0.1#5353' '/use-application-dns.net/'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].rebind_domain='plex.direct'
dhcp.@dnsmasq[0].resolvfile='/tmp/resolv.conf.d/resolv.conf.auto'
dhcp.@dnsmasq[0].ipset='/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/googlevideo.com/omr_dscp-cs4,omr_dscp6-cs4' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/s3.ll.dash.row.aiv-cdn.net/omr_dscp-cs4,omr_dscp6-cs4' '/steamcontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/d25xi40x97liuc.cloudfront.net/omr_dscp-cs4,omr_dscp6-cs4' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/aiv-delivery.net/omr_dscp-cs4,omr_dscp6-cs4' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/audio-fa.scdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1' '/jeuxvideo.com/omr_dst_bypass_wan2,omr6_dst_bypass_wan2' '/static.jvc.gg/omr_dst_bypass_wan2,omr6_dst_bypass_wan2' '/gcsip.com/omr_dst_bypass_wan2,omr6_dst_bypass_wan2' '/secure.square-enix.com/omr_dst_bypass_wan2,omr6_dst_bypass_wan2' '/pass.canal-plus.com/omr_dst_bypass_wan2,omr6_dst_bypass_wan2' '/decouvrir.mycanal.fr/omr_dst_bypass_wan2,omr6_dst_bypass_wan2' '/www.mycanal.fr/omr_dst_bypass_wan2,omr6_dst_bypass_wan2' '/e8933.b.akamaiedge.net/omr_dst_bypass_wan2,omr6_dst_bypass_wan2' '/reddit.com/omr_dst_bypass_wan2,omr6_dst_bypass_wan2' '/go.ocs.fr/omr_dst_bypass_wan2,omr6_dst_bypass_wan2' '/ocs.fr/omr_dst_bypass_wan2,omr6_dst_bypass_wan2' '/netflix.com/omr_dst_bypass_wan2,omr6_dst_bypass_wan2' '/bigfernand.com/omr_dst_bypass_wan2,omr6_dst_bypass_wan2' '/eu-west419.discord.gg/omr_dst_bypass_wan2,omr6_dst_bypass_wan2' '/crunchyroll.com/omr_dst_bypass_wan2,omr6_dst_bypass_wan2' '/cloudflare.com/omr_dst_bypass_wan2,omr6_dst_bypass_wan2' '/mafreebox.freebox.fr/omr_dst_bypass_wan2,omr6_dst_bypass_wan2' '/store.steampowered.com/omr_dst_bypass_wan2,omr6_dst_bypass_wan2' '/particulier.edf.fr/omr_dst_bypass_wan2,omr6_dst_bypass_wan2' '/tpeweb.paybox.com/omr_dst_bypass_wan2,omr6_dst_bypass_wan2' '/paybox.com/omr_dst_bypass_wan2,omr6_dst_bypass_wan2' '/google.com/omr_dst_bypass_wan2,omr6_dst_bypass_wan2' '/instant-gaming.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/vod.canalplus.com/omr_dst_bypass_all,omr6_dst_bypass_all' '/ecoledirecte.com/omr_dst_bypass_all,omr6_dst_bypass_all' '/eurofins.mesanalyses.fr/omr_dst_bypass_all,omr6_dst_bypass_all' '/all/omr_dst_bypass_all,omr6_dst_bypass_all' '/nflxext.com/omr_dst_bypass_all,omr6_dst_bypass_all' '/nflximg.com/omr_dst_bypass_all,omr6_dst_bypass_all' '/nflximg.net/omr_dst_bypass_all,omr6_dst_bypass_all' '/nflxvideo.net/omr_dst_bypass_all,omr6_dst_bypass_all' '/nflxso.net/omr_dst_bypass_all,omr6_dst_bypass_all' '/nintendo.net/omr_dst_bypass_wan2,omr6_dst_bypass_wan2' '/nintendo.com/omr_dst_bypass_wan2,omr6_dst_bypass_wan2'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.lan.ra_default='0'
dhcp.lan.dhcpv6='disabled'
dhcp.lan.force='1'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dropbear.@dropbear[0]=dropbear
dropbear.@dropbear[0].PasswordAuth='on'
dropbear.@dropbear[0].Port='22'
dscp.@classify[0]=classify
dscp.@classify[0].direction='both'
dscp.@classify[0].proto='icmp'
dscp.@classify[0].class='cs5'
dscp.@classify[0].comment='ICMP'
dscp.@classify[1]=classify
dscp.@classify[1].direction='both'
dscp.@classify[1].proto='udp'
dscp.@classify[1].class='cs5'
dscp.@classify[1].src_port='53,123,5353'
dscp.@classify[1].comment='DNS udp and NTP'
[... 1434 lines cut ...]
upnpd.@perm_rule[1].comment='Default deny'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='wan1' 'wan2'
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.channel='36'
wireless.radio0.hwmode='11a'
wireless.radio0.path='platform/soc/3f300000.mmc/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
wireless.radio0.htmode='VHT80'
wireless.radio0.disabled='1'
wireless.radio1=wifi-device
wireless.radio1.type='mac80211'
wireless.radio1.channel='36'
wireless.radio1.hwmode='11a'
wireless.radio1.path='platform/soc/3f300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
wireless.radio1.htmode='VHT80'
wireless.radio1.disabled='1'
wireless.default_radio1=wifi-iface
wireless.default_radio1.device='radio1'
wireless.default_radio1.network='lan'
wireless.default_radio1.mode='ap'
wireless.default_radio1.ssid='OpenWrt'
wireless.default_radio1.encryption='none'

Thank you for your help! :)

Ysurac, 2020-09-20:
Shadowsocks is not connected. Check shadowsocks key and that it's running on the VPS via ```ps aux | grep ss```

Nethary, 2020-09-20:
My shadowsocks config seem ok:
[image]

The result of my ps aux | grep ss:
[image]

Ysurac, 2020-09-20:
ss-server is not running on the VPS, so there is an issue.
Try to do a ```systemctl restart shadowsocks-libev-manager@manager```

Nethary, 2020-09-20:
The result:
[image]

Ysurac, 2020-09-20:
So it's not working. Check in ```/var/log/daemon.log``` if there is an error.

Nethary, 2020-09-20:
My log:

Sep 20 00:00:15 vps650959 systemd[1]: logrotate.service: Succeeded.
Sep 20 00:00:15 vps650959 systemd[1]: Started Rotate log files.
Sep 20 00:00:16 vps650959 systemd[1]: man-db.service: Succeeded.
Sep 20 00:00:16 vps650959 systemd[1]: Started Daily man-db regeneration.
Sep 20 00:30:40 vps650959 dhclient[378]: DHCPREQUEST for xxxxxxx on ens3 to xxxxxxx port 67
Sep 20 00:30:40 vps650959 dhclient[378]: DHCPACK of xxxxxxxx from xxxxxxxx
Sep 20 00:30:40 vps650959 dhclient[378]: bound to xxxxxxxx -- renewal in 42033 seconds.
Sep 20 01:52:42 vps650959 systemd[1]: Starting Daily apt download activities...
Sep 20 01:52:44 vps650959 systemd[1]: apt-daily.service: Succeeded.
Sep 20 01:52:44 vps650959 systemd[1]: Started Daily apt download activities.
Sep 20 06:27:08 vps650959 systemd[1]: Starting Daily apt upgrade and clean activities...
Sep 20 06:27:12 vps650959 systemd[1]: apt-daily-upgrade.service: Succeeded.
Sep 20 06:27:12 vps650959 systemd[1]: Started Daily apt upgrade and clean activities.
Sep 20 07:07:05 vps650959 glorytun-tcp-run[442]: read: Connection timed out
Sep 20 07:07:06 vps650959 glorytun-tcp-run[442]: xxxxxxxxxxxxxx: connected
Sep 20 07:07:06 vps650959 glorytun-tcp-run[442]: bad packet [E85A309CE85A309DE85A309E] !
Sep 20 07:07:06 vps650959 glorytun-tcp-run[442]: xxxxxxxxxxxx: key exchange failed
Sep 20 07:07:06 vps650959 glorytun-tcp-run[442]: xxxxxxxxxxx: connected
Sep 20 11:14:52 vps650959 systemd[1]: Stopping Shadowsocks-Libev Custom Manager Service for manager...
Sep 20 11:14:52 vps650959 systemd[1]: shadowsocks-libev-manager@manager.service: Succeeded.
Sep 20 11:14:52 vps650959 systemd[1]: Stopped Shadowsocks-Libev Custom Manager Service for manager.
Sep 20 11:14:52 vps650959 systemd[1]: Started Shadowsocks-Libev Custom Manager Service for manager.
Sep 20 11:14:52 vps650959 ss-manager[17628]:  2020-09-20 11:14:52 INFO: using tcp fast open
Sep 20 11:14:52 vps650959 ss-manager[17628]:  2020-09-20 11:14:52 INFO: using tcp no-delay
Sep 20 11:14:52 vps650959 ss-manager[17628]:  2020-09-20 11:14:52 INFO: using MPTCP
Sep 20 11:14:52 vps650959 ss-manager[17628]:  2020-09-20 11:14:52 INFO: running from root user
Sep 20 11:14:52 vps650959 ss-manager[17628]:  2020-09-20 11:14:52 INFO: working directory points to /root/.shadowsocks
Sep 20 11:14:52 vps650959 ss-manager[17628]:  2020-09-20 11:14:52 INFO: try to bind interface: ::0, port: 65101
Sep 20 11:14:52 vps650959 ss-manager[17628]:  2020-09-20 11:14:52 INFO: try to bind interface: 0.0.0.0, port: 65101
Sep 20 11:14:52 vps650959 ss-manager[17628]:  2020-09-20 11:14:52 INFO: initializing acl...
Sep 20 11:14:52 vps650959 ss-manager[17628]:  2020-09-20 11:14:52 INFO: enable TCP no-delay
Sep 20 11:14:52 vps650959 ss-manager[17628]:  2020-09-20 11:14:52 INFO: enable multipath TCP
Sep 20 11:14:52 vps650959 ss-server[17632]: resolving hostname to IPv6 address first
Sep 20 11:14:52 vps650959 ss-server[17632]: using tcp fast open
Sep 20 11:14:52 vps650959 ss-server[17632]: UDP relay enabled
Sep 20 11:14:52 vps650959 ss-server[17632]: enable TCP no-delay
Sep 20 11:14:52 vps650959 ss-server[17632]: initializing ciphers... chacha20-ietf-poly1305
Sep 20 11:14:52 vps650959 ss-server[17632]: Invalid key for your chosen cipher!
Sep 20 11:14:52 vps650959 ss-server[17632]: It requires a 32-byte key encoded with URL-safe Base64
Sep 20 11:14:52 vps650959 ss-server[17632]: Generating a new random key: 1q8uGDJs3lXJNf9mGNnnRQEzkBxS9YzN7rmMOPmMmX4=
Sep 20 11:14:52 vps650959 ss-server[17632]: Please use the key above or input a valid key
Sep 20 11:20:07 vps650959 systemd[1]: Stopping Shadowsocks-Libev Custom Manager Service for manager...
Sep 20 11:20:07 vps650959 systemd[1]: shadowsocks-libev-manager@manager.service: Succeeded.
Sep 20 11:20:07 vps650959 systemd[1]: Stopped Shadowsocks-Libev Custom Manager Service for manager.
Sep 20 11:20:07 vps650959 systemd[1]: Started Shadowsocks-Libev Custom Manager Service for manager.
Sep 20 11:20:07 vps650959 ss-manager[21322]:  2020-09-20 11:20:07 INFO: using tcp fast open
Sep 20 11:20:07 vps650959 ss-manager[21322]:  2020-09-20 11:20:07 INFO: using tcp no-delay
Sep 20 11:20:07 vps650959 ss-manager[21322]:  2020-09-20 11:20:07 INFO: using MPTCP
Sep 20 11:20:07 vps650959 ss-manager[21322]:  2020-09-20 11:20:07 INFO: running from root user
Sep 20 11:20:07 vps650959 ss-manager[21322]:  2020-09-20 11:20:07 INFO: working directory points to /root/.shadowsocks
Sep 20 11:20:07 vps650959 ss-manager[21322]:  2020-09-20 11:20:07 INFO: try to bind interface: ::0, port: 65101
Sep 20 11:20:07 vps650959 ss-manager[21322]:  2020-09-20 11:20:07 INFO: try to bind interface: 0.0.0.0, port: 65101
Sep 20 11:20:07 vps650959 ss-manager[21322]:  2020-09-20 11:20:07 INFO: initializing acl...
Sep 20 11:20:07 vps650959 ss-manager[21322]:  2020-09-20 11:20:07 INFO: enable TCP no-delay
Sep 20 11:20:07 vps650959 ss-manager[21322]:  2020-09-20 11:20:07 INFO: enable multipath TCP
Sep 20 11:20:07 vps650959 ss-server[21326]: resolving hostname to IPv6 address first
Sep 20 11:20:07 vps650959 ss-server[21326]: using tcp fast open
Sep 20 11:20:07 vps650959 ss-server[21326]: UDP relay enabled
Sep 20 11:20:07 vps650959 ss-server[21326]: enable TCP no-delay
Sep 20 11:20:07 vps650959 ss-server[21326]: initializing ciphers... chacha20-ietf-poly1305
Sep 20 11:20:07 vps650959 ss-server[21326]: Invalid key for your chosen cipher!
Sep 20 11:20:07 vps650959 ss-server[21326]: It requires a 32-byte key encoded with URL-safe Base64
[... cut ...]
Sep 20 16:50:19 vps650959 systemd[1]: Started Shorewall IPv6 firewall.
Sep 20 16:50:19 vps650959 systemd[1]: Startup finished in 3.674s (kernel) + 10.349s (userspace) = 14.024s.
Sep 20 16:50:20 vps650959 chronyd[428]: Selected source 162.xxxxxx
Sep 20 16:50:27 vps650959 systemd-networkd[213]: mlvpn0: Configured
Sep 20 17:05:13 vps650959 systemd[1]: Starting Cleanup of Temporary Directories...
Sep 20 17:05:13 vps650959 systemd[1]: systemd-tmpfiles-clean.service: Succeeded.
Sep 20 17:05:13 vps650959 systemd[1]: Started Cleanup of Temporary Directories.

Ysurac, 2020-09-20:
There is a problem with shadowsocks key.
What do you have in ```/etc/shadowsocks-libev/manager.json``` ?

Nethary, 2020-09-20:
{
    "mptcp": true,
    "ipv6_first": true,
    "no_delay": true,
    "ebpf": false,
    "server": [
        "[::0]",
        "0.0.0.0"
    ],
    "method": "chacha20-ietf-poly1305",
    "fast_open": true,
    "timeout": 600,
    "port_key": {
        "65101": "WVfzzoCbul2N",
    },
    "local_port": 1081,
    "verbose": 0,
    "acl": "/etc/shadowsocks-libev/local.acl",
    "mode": "tcp_and_udp",
    "reuse_port": true,
    "prefer_ipv6": false
}

Ysurac, 2020-09-20:
Really short key. You put it manually in the wizard ? Check in /root/openmptcprouter_config.txt what the key should be and replace the "WVfzzoCbul2N" that you have in the wizard. If this still don't work, replace it also in this config file. and restart shadowsocks

Nethary, 2020-09-20:
Not at all.

So, i had change my key on the /root/openmptcprouter_config.txt, /etc/shadowsocks-libev/manager.json and on the shadowsocks-libev menu of my router.

The invalid key error (maybe the + digit on my old key?) disapear, but i have this error now:
Sep 20 20:11:19 vps650959 ss-server[8797]: failed to handshake with 91.xxx.xxx.xxx: authentication error

Ysurac, 2020-09-20:
There is still a key problem.
Generate a new key: ```head -c 32 /dev/urandom | base64 -w0``` and use it everywhere. I think it's the best solution :)
And paste here old key, I will check what is the problem and fix my script.

Nethary, 2020-09-20:
Still have authentication error :(

My old key was 0UZ1F+EXPnrF5qqdHa9/aCzdHeFjEikz7maNUIlupnc=

Ysurac, 2020-09-20:
Strange, it's not at all related to "WVfzzoCbul2N"...
Changing the key in the OMR wizard should be enough (this should change key on the VPS too).
You can check key on the VPS (this should be the same as the wizard, but with "+" replaced by "-" and "\" replaced by "_"), and check that method is the same on both.
Else you can reinstall the VPS from scratch.

Nethary, 2020-09-20:
Ok, the shadowsocks error seem to be done.
Now i have this error on the router:
[image]

But if i check on the VPN menu, only the Gloryturn is enable.

Ysurac, 2020-09-20:
It's an update ?
Go to the wizard, check advanced setting box, choose a VPN and validate the wizard again.
And don't use VPN menu to enable a VPN, always use the wizard (or set it to none in the wizard first).

Nethary, 2020-09-20:
That work with Gloryturn, but not with OpenVPN. Is it ok?

[image]

Thank you so much for your help :)

closed 2020-09-24 by Nethary