Ysurac/openmptcprouter #1239: Sendbackup - Multiple servers

dragon2611, 2020-09-21:
## Expected Behavior

On Page cgi-bin/luci/admin/system/openmptcprouter/backup#

If there are multiple servers
Send backup should send to the server next to the button or at least the current primary server.

## Current Behavior
The Backup is sent to the first server on the list

openmptcprouter v0.55.3 r0+14198-da98603597

Ysurac, 2020-09-22:
Backup should be sent to all servers.

dragon2611, 2020-09-22:
Looks like it did but the UI isn't updated as logging in today shows it on both servers.

Sending a new backup makes the date change on the top server but not the 2nd one

Ysurac, 2020-09-22:
True, date is not updated. I will fix that.

Adorfer, 2020-09-22:
hups, observed that as well, forgot to report.

closed 2021-01-04 by github-actions[bot]