Ysurac/openmptcprouter #1240: WhatsApp Call Not Working

mack1726, 2020-09-22:
## Expected Behavior

## Current Behavior
WhatsApp Call Not Working. When i stop Glorytun, whatsapp starts working

## Possible Solution

## Steps to Reproduce the Problem

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: v0.55.3 r0+14198-da98603597
  - OpenMPTCProuter VPS version: V0.1017
  - OpenMPTCProuter platform: RPI4

Ysurac, 2020-09-22:
Whatsapp doesn't have a call test, so it's difficult to really test.
It's possible that they banned IP range of your VPS provider.
This may also be a MTU problem, you can try with only one connection and check if this work or not.

mack1726, 2020-09-23:
I ran tests, whatsapp call works when i disable vpn and set ss-redir for udp - unset. Can I fix this somehow? Thank you so much

Ysurac, 2020-09-23:
Without VPN enabled, UDP use direct connection.

xabix, 2020-09-27:
Hello,
what would make sense it's to find the URL used by WhatsApp app to established the calls. If you can do that then you can put this URL in OMR bypass and get it working.

if you find, let us know as I have the same issue :)

Merci
XabiX

kevin39, 2020-09-27:
WhatsApp IP & ports are publicly available on facebook help center.

xabix, 2020-09-28:
The IPs are given here: https://developers.facebook.com/docs/whatsapp/guides/network-requirements/?locale=fr_FR but would be easier and better to have directly the alias.

xabix, 2020-10-08:
did anyone found a way to overcome this issue easily?

zicochaos, 2020-11-02:
same issue here inbound calls not working 

dedicated server with hyper v 
router: rpi4

Malaga82, 2020-11-04:
i'm using Google Duo and WhatsApp calling, no issues.

closed 2021-02-08 by github-actions[bot]