Ysurac/openmptcprouter #1243: Force certain IP/MAC to strictly use certain WAN

vistman, 2020-09-23:
## Expected Behavior
Force certain IP/MAC to strictly use certain WAN.

## Current Behavior
Maybe, possible but I"m not aware how to do it ...

## Possible Solution
If exists, please provide a short description how to do it ...

## Context (Environment)
I'd like to provide permanent/forced connection via just one WAN for one of my network devices as it doesn't use much bandwidth but requires a stable connection ...

## Specifications

  - OpenMPTCProuter version: 0.55.3
  - OpenMPTCProuter VPS version: 0.1017
  - OpenMPTCProuter platform: RPI4 x86_64

Ysurac, 2020-09-23:
This can be done via Services->OMR-ByPass

vistman, 2020-09-23:
Thanks. Although since my network is managed by router behind OMR I'll have to find another way to force this connection to actually see real source IP and/or MAC.

closed 2020-09-25 by vistman