Ysurac/openmptcprouter #1245: VPN for wan, keys not copied from server

PxxBxx, 2020-09-23:
## Expected Behavior
Activate the option `MPTCP over VPN`
It works

## Current Behavior
When you activate the option, you get in the router syslog : 
```
Wed Sep 23 18:56:48 2020 daemon.err openvpn(wan2)[9522]: VERIFY ERROR: depth=1, error=self signed certificate in certificate chain: CN=ChangeMe
Wed Sep 23 18:56:48 2020 daemon.err openvpn(wan2)[9522]: OpenSSL: error:1416F086:SSL routines:tls_process_server_certificate:certificate verify failed
Wed Sep 23 18:56:48 2020 daemon.err openvpn(wan2)[9522]: TLS_ERROR: BIO read tls_read_plaintext error
Wed Sep 23 18:56:48 2020 daemon.err openvpn(wan2)[9522]: TLS Error: TLS object -> incoming plaintext read error
Wed Sep 23 18:56:48 2020 daemon.err openvpn(wan2)[9522]: TLS Error: TLS handshake failed
```

And in the VPS syslog : 
```
Sep 23 21:09:07 xxxxxxx ovpn-tun1[396]: XX.XX.XX.XX:65301 TLS: Initial packet from [AF_INET]XX.XX.XX.XX:65301, sid=7d0ac250 7be6a215
Sep 23 21:09:07 xxxxxxx ovpn-tun1[396]: XX.XX.XX.XX:65301 VERIFY ERROR: depth=0, error=unable to get local issuer certificate: CN=openmptcprouter
Sep 23 21:09:07 xxxxxxx ovpn-tun1[396]: XX.XX.XX.XX:65301 OpenSSL: error:1417C086:SSL routines:tls_process_client_certificate:certificate verify failed
Sep 23 21:09:07 xxxxxxx ovpn-tun1[396]: XX.XX.XX.XX:65301 TLS_ERROR: BIO read tls_read_plaintext error
Sep 23 21:09:07 xxxxxxx ovpn-tun1[396]: XX.XX.XX.XX:65301 TLS Error: TLS object -> incoming plaintext read error
Sep 23 21:09:07 xxxxxxx ovpn-tun1[396]: XX.XX.XX.XX:65301 TLS Error: TLS handshake failed
Sep 23 21:09:07 xxxxxxx ovpn-tun1[396]: XX.XX.XX.XX:65301 SIGUSR1[soft,tls-error] received, client-instance restarting
```

## Possible Solution
=> The CA/CRT/KEY used by OpenVPN Client were not the same as on the VPS OpenVPN Server

=> I went to the VPS and copied the following files contents : 
`/etc/openvpn/ca/pki/private/ca.crt`
`/etc/openvpn/ca/pki/private/issued/openmptcprouter.crt`
`/etc/openvpn/ca/pki/private/private/openmptcprouter.key`
into the router files :
`/etc/luci-uploads/ca.crt`
`/etc/luci-uploads/client.crt`
`/etc/luci-uploads/client.key`

And the VPN was up on the next respawn

## Steps to Reproduce the Problem

  1. activate the `MPTCP over VPN` option on any wan (note: I used `wan2` in my case)

## Context (Environment)
SFR/Altice is blocking MPTCP on their 4G in my area. It may not be the case everywhere, but at my place, it is :-/

## Specifications

  - OpenMPTCProuter version: 0.55.3
  - OpenMPTCProuter VPS version: 0.1017
  - OpenMPTCProuter platform: x64 ext4 image (on Proxmox)

Ysurac, 2020-09-23:
I'm not able to reproduce the issue.
How did you check that MPTCP is blocked on SFR ?

PxxBxx, 2020-09-23:
Hello @Ysurac , the issue is not that SFR is blocked on my side, but that the MPTCP over VPN was not working.
FYI, I don't know how to read the MPTCP Support Check for my wan2, but here are 2 symptoms : 
* if wan2 (SFR) is master, aggregation doesn't work and all trafic stays on wan2
* if wan1 or wan3 is master, no trafic goes on wan2
Sorry, I already said that in #1090, but.... i forgot to add an answer telling I went with MPTCP over VPN since July, and it worked like a charm.
But with 0.55.3 the VPN (for wan2) was not up, I had to update the CA/CRT/KEY

So I think that on setup, the Router does not get CA/CRT/KEYs from Server. Maybe it's a bug ? (it worked with 0.54.x without editing any CA/CRT/KEY)

Ysurac, 2020-09-23:
Cert should be downloaded from the VPS at first run. Did you reinstall the VPS from scratch while the router was already configured from previous config ?
I will try to find in which case the key can be not downloaded or updated.

PxxBxx, 2020-09-23:
Hello, no, I always upgrade the VPS (keeping the keys) *then* I upgrade the Router (install from scratch + manual setup)

PxxBxx, 2020-09-23:
Note : I temporary switch to a second VPS during the nominal VPS upgrade.
And as soon as the VPS is upgraded, I upgrade the router (from scratch)

PxxBxx, 2020-09-23:
> Cert should be downloaded from the VPS at first run. Did you reinstall the VPS from scratch while the router was already configured from previous config ?

I confirm that the CA/CRT/KEY files on the router were all dated 11/09, the day I installed the router

The `.placeholder` file from your image is dated on 18/08, so these CA/CRT/KEY files were really created during the setup.

I don't know how/why they are different than the ones from the VPS (on the VPS, the files are dated 26/01/2020, so they are really the original ones)

PxxBxx, 2020-11-16:
Hello, did you have any clue on these VPN keys download ?
I just switched VPS, and keys are not updated :-/

Ysurac, 2020-11-16:
In System->OpenMPTCProuter, "advanced settings" checkbox, "Force retrieve settings" should work.

PxxBxx, 2020-11-16:
UPDATE : Fresh OMR install (Router) + initial config activating the VPN for WAN2 + advanced settings checking for force keys from server => it worked

PxxBxx, 2020-11-16:
I did the "force" while still running on 0.56.3 and it didn't work.
But on the fresh 0.56.5 install, it worked first try.

closed 2021-02-19 by github-actions[bot]