Ysurac/openmptcprouter #1246: Informations et questions

JeedHome44, 2020-09-23:
Bonjour,

J'aimerai me lancer mais j'ai quelques points que je ne pige pas trop. Déjà, je pensais qu'il fallait mettre le Pi avec ce logiciel entre le réseau LAN de la maison avec le port principal ethernet du Pi et acheter 2 USB vers ethernet pour chaque box. Mais, sur ton schéma, tu branches ton Pi seulement sur l'un des routeurs?

Deuxièment, je n'avais jamais entendu parlé de VPS avant aujour'dhui. Au début, j'ai cru que tu parlais de VPN. Ca sert à quoi exactement? Pas possible de s'en passer? Et peut on avoir un VPN ensuite si on a un VPS? Personnellement, j'ai une Freebox et donc une ip fixe externe. Si je fais ca, elle ne fonctionnera plus du tout?

Troisièmement, j'ai 2 Pi avec Jeedom, 1 Pi Octoprint et 1 Pi avec Home assistant. J'ai des DNS Jeedom avec le pack Power Jeedom. Ils continueront à marcher avec le VPS ou pas? Ca serait très embêtant car je remonte tous mes appareils domotiques de Jeedom vers Google Home avec le DNS https de Jeedom.

Et dernière question: vous parlez de bridage de réseau. Mon réseau LAN n'est fait qu'avec des cables cat 6, CPL 1000 Mpbs et je ne veux donc pas brider ma connection 4G qui monte à 120 Mbps avec le même routeur Huawei que vous. Je vais donc mettre votre logiciel sur un Pi4. C'est bien 1 Gbps pour le Pi4? Ca veut dire que mon réseau global restera à 1000 Mbps?

Encore une question, j'allais oublier! Le VPS à 200 Mbps va donc brider tous mon réseau LAN à 200 Mbps ou là, pour le coup, on parle seulement de vitesse internet et donc pas du LAN? Les 200 Mbps veulent donc dire qu'avec ma connection 4G à 120 et mon ADSL à 7, je plafonnerai donc à 127 Mbps (sous les 200 du VPS) c'est bien ca?

Ah, oui, encore une pardon: Vous avez marquez qu Netflix n'est pas compatible avec OVH. Nous regardons actuellement Netflix par la connexion 4G ainsi que Disney+. Quel VPS faut il donc prendre pour que ca soit toujours possible?

Merci et désolé pour ce gros pavé! :)

Ysurac, 2020-09-24:
1) Oui pas besoin de USB vers ethernet. Il suffit de brancher sur un switch externe ou intégré à l'un des routeurs et de tout relier dessus. Des interfaces de type MacVlan sont crées sur le RPI afin de partager son interface physique.

2) Un VPS correspond à une machine virtuelle sur un serveur. Un VPS étant souvent moins cher qu'un serveur, c'est ce qui est conseillé. Il permet de reconstituer les liens et donc d'aggréger réellement en profitant de sa connexion dans un centre de données. On peut avoir un VPN par dessus OpenMPTCProuter. Pour la Freebox elle fonctionnera, c'est une connection qui sera aggrégée.

3) Il est possible d'utiliser des DNS personnalisées. Donc surement même si je ne sais pas ce que c'est exactement.

4) Pour le Pi4 la connection agrégée n'atteindra pas 4Gbps, mais ça change rien à la vitesse du réseau local.

5) On parle bien que de la vitesse Internet. Qui aura surement un peu moins que 127Mbps (une partie est prise par le chiffrement, et la 4G n'est jamais très stable en débit).

6) Netflix bloque effectivement les IPs venant d'OVH. Donc il faut soit prendre un VPS non bloqué (il n'y a que Milkywan à ma connaissance), soit utilisé omr-bypass qui permet de sortir directement par une interface pour netflix ou autre mais dans ce cas sans profiter de l'agrégation.

JeedHome44, 2020-09-24:
Bonjour,

On me dit qu'il est possible dans les options de votre logiciel de dire que tel ou tel équipement ne passe que par la 4G par exemple. Il est donc possible d'éviter le VPS à ma TV qui a donc Netflix et ça marchera même avec VOS OVH c'est ça?

Ysurac, 2020-09-24:
Oui c'est ce que je dis en 6. On peut sortir directement par une connexion sans passer pour le VPS pour un domaine, ip, protocole, un équipement,...

JeedHome44, 2020-09-24:
Ah donc sans viser un équipement c'est donc possible de dire pour le site netflix c'est que par 4G?

Ysurac, 2020-09-24:
Oui.

JeedHome44, 2020-09-24:
Donc si je comprends bien, pas besoin d'un Pi 4B forcément vu que le 3B+ atteint 300 Mbps? C'est donc suffisant si je prends un abo VPS à 250 Mbps? Et en local j'aurai quand même tout mon réseau en 1000 Mbps.

Ysurac, 2020-09-24:
Le RPI3B+ n'atteindra pas 300Mbps. Le CPU est faible, et même sans le chiffrement activé, il dépasse pas les 100Mbps avec OpenMPTCProuter.
Avec un RPI4 un VPS 250Mbps est suffisant oui. Et en local tu auras bien ton réseau en 1000Mbps.

JeedHome44, 2020-09-24:
Donc quel modèle de Pi4 me conseillez vous? 2Go suffisent?

De plus, quel service me conseillez vous pour mettre un VPN dessus et sur le VPS? Je n'ai jamais fait ça. Jeedom a juste OpenVPN qui permet d'avoir le DNS Jeedom mais ça s'est configuré automatiquement.

JeedHome44, 2020-09-24:
Et encore une question, désolé:

Vaut il mieux mettre le système sur clé USB, carte SD ou SSD?

La SD ne va-t-elle pas vite cramer?

Ysurac, 2020-09-24:
2Go est suffisant oui.
Je conseille de pas mettre de VPN, mais je ne sais pas trop ton besoin.

Sur carte SD. Et si la carte est de qualité elle devrait tenir pas mal de temps.

JeedHome44, 2020-09-24:
En fait, certains sites sont bloqués si IP FR. J'avais lu qu'il faut mettre un VPN pour contourner ce problème.
Le VPN ne protège pas plus du piratage? Je pensais aussi que ça cryptait les données entre le Pi et le VPN. Sans VPN les données avec le VPS sont quand même cryptés?

Ysurac, 2020-09-24:
L'utilisation d'un VPN pour contourner ce type de blocage géographique est indépendant d'OpenMPTCProuter.
Le VPN protège de pas grand chose.
Les données entre le PI et le VPS sont chiffrées par défaut avec OpenMPTCProuter.

JeedHome44, 2020-09-24:
Ok, je vais donc déjà me concentrer sur l'installation de votre logiciel. Par contre, comment faire pour basculer de mes 2 LAN actuels à un seul avec votre logiciel? Je peux configurer le logiciel en branchant d'abord le Pi sur l'un des réseaux, ensuite, enregistrer les IP fixe de tous les appareils, désactiver les 2 DHCP de mes routeurs, tout éteindre, relier les 2 LAN entre eux, tout rallumer.
C'est bien comme ça?

JeedHome44, 2020-09-24:
Je pense que je vais mettre à jour avant mon fichier Excel avec mes équipements pour bien choisir leur IP avant et faciliter le boulot.

Ysurac, 2020-09-24:
Oui, après il est aussi possible de garder les 2 LANs, mais je ne sais pas comment est exactement la configuration de ton réseau.

JeedHome44, 2020-09-24:
Je préfère autant tout mettre sur un seul. Ça sera bien plus facile à gérer et je vais virer mes routeur TP link et mes répéteurs pour mettre 5 point d'accès Mesh TP Link.

Dernière question: je vois qu'il y a plusieurs lignes différentes pour Pi 4B pour télécharger le fichier. Lequel dois-je prendre pour éviter les erreurs?

Encore merci pour toutes ces réponses. Ça va changer complètement mon réseau LAN tout ça pour quelque chose de bien plus simple a l'utilisation et qui ne coupera plus internet si l'une des connexions lache.

Ysurac, 2020-09-25:
L'image ext4 factory.

JeedHome44, 2020-09-26:
Salut!

J'ai lancé l'installation sur mon VPS OVH fraichement créé.
J'ai eu ceci à la fin:

After reboot, check with uname -a that the kernel name contain mptcp.
 Else, you may have to modify GRUB_DEFAULT in /etc/default/grub

Je ne comprends pas vraiment la commande exacte que je dois taper pour faire la vérification.
Merci

Ysurac, 2020-09-27:
```uname -a```

JeedHome44, 2020-09-27:
Bonjour,

J'ai mis la version ext4 factory 64 bits sur ma carte SD 64 Go avec Etcher. Je la mets dans le Pi et rien ne se passe.

Je précise que j'ai branché le Pi sur l'ethernet de mon PC portable car j'aimerai dans un premier temps y ajouter tous les IP fixes de tous mes appareils chez moi. J'ai récupéré toutes les adresses MAC de chaque appareil et prévu un IP pour chacun.

Je veux donc avoir accès à votre logiciel pour les ajouter avant de mettre le Pi sur le réseau.
Ce n'est pas possible?

Je précise que l'eeprom du Pi a été mis à jour en stable avec la version du 03-09-2020.
Est-ce ca le problème? Votre logiciel ne prend pas en charge la dernière version de l'eeprom?

kevin39, 2020-09-27:
> rien ne se passe.

C'est à dire  ? Tu as tenté de faire quoi ?

JeedHome44, 2020-09-27:
Comment ca tenté? Normalement, un Pi qui boot bien a ses LED verte et orange de l'ethernet qui clignote. La, ca ne fait rien.
Est-ce la bonne version que j'ai mis sur la carte SD? J'ai un Pi 4B 2Go eeprom du 03-09-2020.
Je vois sur le lien de téléchargement qu'il y a aussi une version sysupgrade. c'est quoi la différence?

JeedHome44, 2020-09-27:
Mais comme dit plus haut, le Pi n'est pas branché à un routeur mais juste à mon PC portable en ethernet. Est-ce ca le soucis?

kevin39, 2020-09-27:
Tu as essayé d'accéder au routeur sur 192.168.100.1 ?

JeedHome44, 2020-09-27:
oui ca ne marche pas. il faut obligatoirement qu'il soit branché au routeur? y a pas un fichier dans la clé usb qu'on peut éditer pour configurer tous les ip fixés?

JeedHome44, 2020-09-27:
Bon, j'ai pris un Pi 3B+ que je garde pour me faire une Recallbox. Ca marche parfaitement sur le 3B+ mais pas sur le Pi 4B 2Go. Il y a donc un problème avec le l'img pour le 4B je pense!

Ysurac, 2020-09-27:
Le 4B 2Go est la plateforme que j'utilise quotidiennement donc elle est testée très régulièrement et de façon intensive.

JeedHome44, 2020-09-27:
avez vous mis l'eeprom a jour du 03-09-2020 dessus?

JeedHome44, 2020-09-27:
Comment se fait-il alors que ca ne marche pas sur mon Pi 4B 2Go mais que ca démarre normalement sur le 3B+? C'est peut être moi qui prend le mauvais chier dans toute la liste. j'ai pris comme vous m'avez dit celui pour le Pi 4B ext4 j'ai testé le 64 et le 32 rien n'y fait

Ysurac, 2020-09-27:
Pas encore, elle est encore à la précédente stable du 31 juillet.
Celui la : https://download.openmptcprouter.com/release/v0.55.3/rpi4/targets/bcm27xx/bcm2711/openmptcprouter-v0.55.3-r0+14198-da98603597-bcm27xx-bcm2711-rpi-4-ext4-factory.img.gz fonctionne, après gunzip puis dd sur la carte SD.

JeedHome44, 2020-09-27:
vous me conseillez donc de mettre l'eeprom en version du 31 juillet sur ce Pi 4B? on peut faire un downgrade de l'eeprom?

JeedHome44, 2020-09-27:
De plus, je suis dans la section DHCP de votre logiciel. j'ai l'impression qu'on ne peut pas définir manuellement l'ip qu'on souhaite mettre en fixe à un appareil. on a le menu déroulant mais pas de choix possible?

JeedHome44, 2020-09-27:
On ne peut pas non plus attribuer une plage d'ip dynamique par exemple seulement de 230 à 255? comme ca, les nouveaux appareils ne prennent un IP que sur cette plage?

Ysurac, 2020-09-27:
On peut définir l'IP que l'on veut en bas de la liste de choix, et on peut bien entendu définir la plage IP dynamique que l'on veut dans Réseau->Interface puis LAN puis onglet DHCP serveur.

JeedHome44, 2020-09-27:
Ok merci bien. Je vais donc retenté dans le Pi 4B. Vous me conseillez de mettre l'eeprom en date du 31 juillet?

JeedHome44, 2020-09-27:
Je viens de tester de mettre une carte SD dans le même Pi 4B avec Raspbian d'installé dessus qui m'a permis de mettre à jour l'eeprom et il boot bien.

Ysurac, 2020-09-27:
L'eeprom devrait pas changer grand chose.

JeedHome44, 2020-09-27:
c'est quand même étonnant que le Pi marche très bien avec Raspbian Buster mais pas avec l'img de votre logiciel alors. je vais changer l'eeprom et tester

Ysurac, 2020-09-27:
Je n'ai eu aucun retour sur un défaut avec l'image RPI4.

JeedHome44, 2020-09-27:
peut être que personne n'a mis à jour l'eeprom au 03-09-2020! j'ai appliqué la version du 31-07 la et je teste de suite pour vous dire

JeedHome44, 2020-09-27:
bon bah ca fait pareil... les led ethernet du pi 4B sont fixes et rien ne boot... je n'y comprends rien...

JeedHome44, 2020-09-27:
même en changeant de carte SD, rien ne se passe... donc il y a quelque chose que je ne pige pas...

JeedHome44, 2020-09-27:
si j'ai pu le faire marcher sur un 3B+, c'est que je fais les choses correctement pour les mettre sur la carte SD.... je sèche complètement la

Ysurac, 2020-09-27:
Tu as vérifier que l'image a été téléchargé correctement avec sa signature sha256sum ?

JeedHome44, 2020-09-27:
j'ai téléchargé l'image plusieurs fois cet aprem. mais on vérifie ca comment? et j'ai utilisé aussi l'image de la version compatible aussi pour le 3B+ sur la 4B pour tester et idem. Ca veut dire que l'image marche sur mon 3B+ mais pas sur mon 4B...

Ysurac, 2020-09-27:
```sha256sum -c <fichier.sha256sum>```

JeedHome44, 2020-09-27:
Désolé mais je ne comprends pas.

Ysurac, 2020-09-27:
C'est la commande pour vérifier l'image télécharger en utilisant le fichier signature avec sha256sum.

JeedHome44, 2020-09-27:
Je suis sous Windows et je ne vois pas où taper ça en fait. Sachant que ça boot pas ça ne doit pas être en ssh

Ysurac, 2020-09-27:
Je ne connais pas Windows. Je ne sais pas si sha256sum est dispo sous windows ou pas.

JeedHome44, 2020-09-27:
Ok mais l'image doit être bonne sinon ça ne marcherait pas sur le Pi 3B+.

Donc si on résume.
L'image marche sur un Pi 3B+.
Raspbian lite marche sans problème sur SD avec mon Pi 4B.
L'image ne marche pas sur 2 SD différente avec le Pi 4B.

Ysurac, 2020-09-27:
T

Ysurac, 2020-09-27:
Oups, j'ai glissé ;)
Tu as test l'image squashfs ?

Ysurac, 2020-09-27:
L'image ne peut pas marcher sur le PI 3B+. Elle est spécifique RPI4. Sauf si tu prends pas la bonne image.

Ysurac, 2020-09-27:
Essaye : https://download.openmptcprouter.com/release/v0.55.3/rpi4/targets/bcm27xx/bcm2711/openmptcprouter-v0.55.3-r0+14198-da98603597-bcm27xx-bcm2711-rpi-4-squashfs-factory.img.gz

JeedHome44, 2020-09-27:
Je me suis mal exprimé. L'image 32 bits marche sur 3B+ et 4B c'est bien ça?
Aujourd'hui j'ai donc essayé plusieurs fois sur le 4B celle pour 3B+ et celle que pour 4B.

Donc celle pour 3B+ marche sur 3B+ mais pas sur 4B

JeedHome44, 2020-09-27:
Non je n'ai pas essayé que les versions ext4 comme vous me disiez de prendre celles-ci

Ysurac, 2020-09-27:
Je viens de télécharger https://download.openmptcprouter.com/release/v0.55.3/rpi4/targets/bcm27xx/bcm2711/openmptcprouter-v0.55.3-r0+14198-da98603597-bcm27xx-bcm2711-rpi-4-ext4-factory.img.gz de la décompresser puis de la mettre sur une carte SD et dans le RPI4 et aucun soucis.

JeedHome44, 2020-09-27:
Est-ce possible que ce soit l'alimentation du P4 qui supporte pas le logiciel ?￼

Ysurac, 2020-09-27:
Si elle est vraiment limite c'est possible. Mais normalement une alimentation 3A est nécessaire pour le RPI4.

[9 more replies]

closed 2020-09-30 by Ysurac