Ysurac/openmptcprouter #1247: info

ascomputer, 2020-09-24:
if i remove a dongle and then put it back why do i have to restart my raspberry pi 4 to get the dongle working again?
the dongle is rejoined but not used,
how can I do?
I saw that if I save in the wizard it starts but the problem is that if I'm seeing a video it stops and then starts again.
thank you

Modello	Raspberry Pi 4 Model B Rev 1.1
Architecture	ARMv7 Processor rev 3 (v7l)
Versione Firmware	openmptcprouter v0.55.3 r0+14198-da98603597 / LuCI Master git-20.219.57394-35e2735
Versione del Kernel	5.4.58

Ysurac, 2020-09-24:
What is the dongle ? What is the protocol used ?

ascomputer, 2020-09-25:
the dongle is a 4g usb modem ... I attach photos.
[image]

Ysurac, 2020-09-25:
What is the protocol used ? QMI, ModemManager,... or it create an ethernet port ?

ascomputer, 2020-09-25:
yes creates an ethernet port

Ysurac, 2020-09-25:
I'm not able to reproduce the issue.
What is the master selection used in System->OpenMPTCProuter advanced settings tab ?

ascomputer, 2020-09-25:
is balancing

closed 2021-01-04 by github-actions[bot]