Ysurac/openmptcprouter #1248: DNS issue

tlkelley70, 2020-09-25:
## Expected Behavior
DNS is not resolving on the PI4 I have this installed on.  It worked fine for a few weeks.  I can ping the internet by IP but nslookup gets no resolution.  This causes the server to show down.

## Current Behavior
DNS does not resolve with nslookup 

## Possible Solution
Easy way to restart DNS or whatver process is controlling what is failing.

## Steps to Reproduce the Problem
server shows down.  Go to diagnostics and do nslookup on anything, it fails.  Ping 1.1.1.1 or anything by IP, it works.  If I go to the master interface and set a custom DNS the server comes up and resolution then works. which is a bandaid but has me back to functional.  

I did a full backup restore of the PI4 to a previous backup that worked fine just encase something happened.  No help.  Did a snapshot restore of the server, same situation as above no help.

## Specifications

PI4 running r v0.55.3 r0+14198-da98603597

Ysurac, 2020-09-25:
What do you have in status page without the custom DNS added ?
By DNS is done via unbound that send requests to root DNS servers. Check in State->System log if you have anything about that.

tlkelley70, 2020-09-25:
[image]
[image]
[image]
[image]

I am not sure how it occurred, but ubound was disabled....magic it works when I enable.  Please consider this resolved, user error I guess but I did not login to the box between when it worked and when it stopped but I am the only one with access...

Thanks for your help and the product is awesome.

Ysurac, 2020-09-25:
Maybe a crash of unbound, I will add a check that will restart unbound if needed.

closed 2020-09-25 by Ysurac