Ysurac/openmptcprouter #1249: How to use vps like a web server ?

launux, 2020-09-26:
I just discover OpenMPTCProuter, it seems fantastic ! I whished use my webserver to test your solution.

OMR works fine, but all my websites are unavailable anymore.

How can I open ports on my vps ?

Thank you for your help.

Ysurac, 2020-09-26:
On the VPS a firewall is configured by default. This use shorewall. You can check in ```/etc/shorewall/rules``` current rules and open port 80/443.

launux, 2020-09-26:
Thank you very much, I didn't know Sharewall ! But I work with Docker and the opening of the ports is not enough...

Ysurac, 2020-09-26:
I don't know what you mean, but if you installed OpenMPTCProuter under docker this can't work without using a kernel with MPTCP patch.
For Docker there is some configuration needed that you can see in Shorewall doc.

launux, 2020-09-26:
I'm sorry OpenMPTCProuter works fine. On my vps, I had a apache server with redirect from http (80) to https (443) next a reverse proxy on a Docker container on another port (8001).

I am going to continue to investigate it.

Is there an option to uninstall the vps part ?

Ysurac, 2020-09-26:
No, there is no option to uninstall VPS part.

closed 2021-01-04 by github-actions[bot]