Ysurac/openmptcprouter #1250: TPLink T9UH USB WiFi Adapter Not Working

shrieram, 2020-09-26:
## Expected Behavior
Added the USB adapter T9UH to one of the usb3.0 ports in my RPi4. I've installed the driver kmod-rtl8192cu (hope this is the correct one). I'm supposed to see the new WiFi radios in the wireless screen (along with the inbuilt radio).
## Current Behavior
I'm seeing only the inbuilt radio of my RPi4
## Possible Solution
Am I using the correct driver ? I see more rtl drivers in the software section & in fact installed all of them.
## Steps to Reproduce the Problem

  1. Add T9UH to one of the usb ports
  2. Install the driver
  3. Look for the new radio in wireless section

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: v0.55.3
  - OpenMPTCProuter VPS version: 0.1017 5.4.52-mptcp
  - OpenMPTCProuter platform: RPI4

Below is the kernel log

_[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 5.4.58 (circleci@default-2a7790e6-62ca-4901-8399-62d30064fe6c) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r0+14198-da98603597)) #0 SMP Tue Aug 18 16:14:54 2020
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.2
[    0.000000] Reserved memory: created CMA memory pool at 0x000000002ac00000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] On node 0 totalpages: 1012736
[    0.000000]   DMA zone: 3792 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 242688 pages, LIFO batch:63
[    0.000000]   DMA32 zone: 12032 pages used for memmap
[    0.000000]   DMA32 zone: 770048 pages, LIFO batch:63
[    0.000000] percpu: Embedded 21 pages/cpu s45144 r8192 d32680 u86016
[    0.000000] pcpu-alloc: s45144 r8192 d32680 u86016 alloc=21*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: EL2 vector hardening
[    0.000000] ARM_SMCCC_ARCH_WORKAROUND_1 missing from firmware
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 996912
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_headphones=1 bcm2708_fb.fbwidth=0 bcm2708_fb.fbheight=0 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32:99:93:8B vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=squashfs,ext4 rootwait
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x37400000-0x3b400000] (64MB)
[    0.000000] Memory: 3828792K/4050944K available (7038K kernel code, 430K rwdata, 1836K rodata, 512K init, 539K bss, 156616K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: 	CONFIG_RCU_FANOUT set to non-default value of 32.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] random: get_random_bytes called from start_kernel+0x2dc/0x3dc with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000005] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000156] Console: colour dummy device 80x25
[    0.000395] printk: console [tty1] enabled
[    0.000428] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=216000)
[    0.000450] pid_max: default: 32768 minimum: 301
[    0.000643] Mount-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.000727] Mountpoint-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.001371] Disabling memory control group subsystem
[    0.002064] ASID allocator initialised with 32768 entries
[    0.002153] rcu: Hierarchical SRCU implementation.
[    0.002783] smp: Bringing up secondary CPUs ...
[    0.003171] Detected PIPT I-cache on CPU1
[    0.003223] CPU1: Booted secondary processor 0x0000000001 [0x410fd083]
[    0.003660] Detected PIPT I-cache on CPU2
[    0.003695] CPU2: Booted secondary processor 0x0000000002 [0x410fd083]
[... 713 lines cut ...]
[    9.792655] brcmfmac mmc1:0001:3: no of_node; not parsing pinctrl DT
[    9.792763] usbcore: registered new interface driver brcmfmac
[    9.802779] usbcore: registered new interface driver cdc_mbim
[    9.809423] usbcore: registered new interface driver ipw
[    9.814824] usbserial: USB Serial support registered for IPWireless converter
[    9.822816] usbcore: registered new interface driver option
[    9.828454] usbserial: USB Serial support registered for GSM modem (1-port)
[    9.837602] usbcore: registered new interface driver qcserial
[    9.843447] usbserial: USB Serial support registered for Qualcomm USB modem
[    9.855760] usbcore: registered new interface driver rt2800usb
[    9.869811] usbcore: registered new interface driver rtl8192cu
[    9.904064] kmodloader: done loading kernel modules from /etc/modules.d/*
[   10.531109] usb 1-1.4: USB disconnect, device number 4
[   10.759829] usb 1-1.4: new high-speed USB device number 5 using xhci_hcd
[   10.869096] usb 1-1.4: New USB device found, idVendor=12d1, idProduct=14db, bcdDevice= 1.02
[   10.877598] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   10.884943] usb 1-1.4: Product: HUAWEI Mobile
[   10.889310] usb 1-1.4: Manufacturer: HUAWEI
[   10.894248] usb 1-1.4: no of_node; not parsing pinctrl DT
[   10.953010] cdc_ether 1-1.4:1.0: no of_node; not parsing pinctrl DT
[   10.987330] cdc_ether 1-1.4:1.0 eth1: register 'cdc_ether' at usb-0000:01:00.0-1.4, CDC Ethernet Device, 00:1e:10:1f:00:00
[   12.105380] bcmgenet: Skipping UMAC reset
[   12.111141] bcmgenet fd580000.ethernet: configuring instance for external RGMII
[   12.119051] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   12.127433] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready_

Ysurac, 2020-09-26:
This is not a correct driver for this USB key. I'm not sure this key can be supported.

shrieram, 2020-09-27:
> This is not a correct driver for this USB key. I'm not sure this key can be supported.

Oh that’s bad. Can you recommend some dual band usb WiFi adapters that is supported please ? The onboard WiFi module in RPI doesn’t give me more than 100mbps.

shrieram, 2020-09-30:
> > This is not a correct driver for this USB key. I'm not sure this key can be supported.
> 
> Oh that’s bad. Can you recommend some dual band usb WiFi adapters that is supported please ? The onboard WiFi module in RPI doesn’t give me more than 100mbps.

@Ysurac can you help with this information please ?

Ysurac, 2020-09-30:
I have no idea at all. 
RPI have low USB power, and it's often a problem with USB dongle.
Personally I use an external wifi router.

shrieram, 2020-10-01:
> I have no idea at all.
> RPI have low USB power, and it's often a problem with USB dongle.
> Personally I use an external wifi router.

Alright. Thanks for the response. I'll go with an external wifi router as well. Also, wanted to tell you that, this is an excellent project.. fabulous job, appreciate it @Ysurac

closed 2020-10-01 by shrieram