Ysurac/openmptcprouter #1251: Add option to MASQUERADE outbound IPv6

dragon2611, 2020-09-26:
Whilst ideally this wouldn't be needed with a public subnet, as doing SNAT on IPv6 is usually frowned upon due to the way shadowsocks is a proxy this could be useful in some scenarios.

I'm specifically thinking of the scenario where an applications passes both TCP/UDP traffic and the Proxy is set to shadowsocks.

This ensures the IPv6 address of both the TCP and UDP packet comes from the VPS IP

Adorfer, 2020-09-26:
what kind of nat6 do you want to disable by this option?

dragon2611, 2020-09-26:
It would to be to enable nat6 not disable.

Looking at shorewall6 it looks like it might nat some ula prefixes but it doesn't do it for public prefixes

Adorfer, 2020-09-27:
on my fresh install of v0.55.3 this does not happen. Perhaps try an update.

closed 2021-01-04 by github-actions[bot]