Ysurac/openmptcprouter #1252: Request: wireguard as VPN

Malaga82, 2020-09-26:
## Expected Behavior
It would be great if wireguard could be used as VPN for traffic instead of OpenVPN, since wireguard is very fast, compared to OpenVPN
## Current Behavior
There is no wireguard VPN

## Possible Solution
Implement wireguard server on VPS and client part in the router.
The only thing is, since it's an exchange of public keys, that priv key on router side must be created ( on first boot?  ) and there should be an input field to fill the related pub key.
Server side it's like all other keys, just output the pub key.

## Steps to Reproduce the Problem
No problems

## Context (Environment)
https://www.wireguard.com/performance/
wireguard is definitely fast ( on both ping and bandwidth ) than OpenVPN, and with bonding 4G connections save some bandwidth and ping is really nice.

Ysurac, 2020-09-26:
Wireguard is UDP VPN and can't be bind to an interface. It's not usable for OpenMPTCProuter.

closed 2020-09-26 by Ysurac

Adorfer, 2020-09-27:
Network namespaces should provide "interface binding" functionality. 

https://www.wireguard.com/netns/

Malaga82, 2020-09-27:
Thanks @Adorfer .
@Ysurac maybe Softether VPN can be used if wireguard is not usable?

Ysurac, 2020-09-27:
In all case wireguard is an UDP based VPN. Can't work over MPTCP. May be usable with MPTCP over wireguard after a configuration nightmare.

Softether is also a configuration nightmare. I didn't see what it can provide.

Wyk72, 2022-08-16:
> In all case wireguard is an UDP based VPN. Can't work over MPTCP. May be usable with MPTCP over wireguard after a configuration nightmare.
> 
> Softether is also a configuration nightmare. I didn't see what it can provide.

I disagree: wireguard w namespaces is not a "configuration nightmare". 

Softehter IS a nightmare, on that I agree.

It boils down to a few commands with wireguard , easily scriptable:

Example: restrict a specific wireguard config/endpoint to the interface  "eth0.4":

ip netns add ethlink -> add the "ethlink" namespace
ip link set eth0.4 netns ethlink -> move specified interface to the namespace previously created 
ip -n ethlink link set eth0.4 up -> wake up the interface
ip netns exec ethlink udhcpc -i eth0.4 -> get an ip address w udhcpc (or static or whatever)
ip netns exec ethlink ping -c2 8.8.8.8 -> make sure eth0.4 link  is up and running
ip -n ethlink link add ethwg0 type wireguard -> create wg interface 
ip -n ethlink link set ethwg0 netns 1 -> move back wg interface to "init" namespace
wg setconf ethwg0 "wgconf.conf" -> configure wg interface w keys/etc
ip addr add 172.32.100.2/32 dev ethwg0 -> give an address to wg interface
ip link set ethwg0 up -> fire it up
ip -4 route add 172.32.100.254/32 dev ethwg0 -> route to the endpoint (assuming it's 172.32.100.254)
ping -c2 172.32.100.254 -> make sure it's ok

It's super-convenient, imho much, much better than the route hacks like fwmarks etc. 
Bulletproof, elegant and 300% faster than openvpn, maybe more.
If you want L2 connectivity (pretty pointless in this use-case)  just fire up a VXLAN tunnel into the wireguard tunnel.

Ysurac, 2022-08-17:
I hope you know it's an old closed issue and wireguard is since available in OpenMPTCProuter as a choice for MPTCP over VPN ? The problem is never to set a VPN up but to be able to set it in all case for all configurations without too much user configuration.
fwmark is used because it's supported by OpenWRT and easy. MPTCP over VPN is only a dirty hack for MPTCP support...