Ysurac/openmptcprouter #1254: 0.55 versions breaking compatibility with my SFR ADSL box

juliencampy, 2020-09-27:
## Expected Behavior
My SFR ADSL box is up and detected by OpenMPTCProutern multipath working.
Everything should work as it did with the 0.54 versions.

## Current Behavior
My SFR ADSL box is detected by OpenMPTCProuter but is marked as Gateway DOWN and is not used by OpenMPTCProuter.
[image]

## Possible Solution
Rollback to 0.54

## Steps to Reproduce the Problem

  1. Upgrade to any 0.55.X versions
  2. That's it !

## Context (Environment)
Everything was working until I upgraded to 0.55.X versions.
No change were made to my network hardware nor configs.

## Specifications

  - OpenMPTCProuter version: v0.55.3 r0+14198-da98603597
  - OpenMPTCProuter VPS version: 0.1015 4.19.104-mptcp
  - OpenMPTCProuter platform: Raspberry Pi 4 Model B Rev 1.1 / 64 bits

Thanks for your help :)

Ysurac, 2020-09-27:
Check that gateway answer to ping.

juliencampy, 2020-09-27:
No answer, I don't get it, OpenMTCPRouter can display the wan address, so there must be some kind of communication there, no ?

juliencampy, 2020-09-27:
I get a weird message when I try to ping the gateway from the RPI4 :

juliencampy, 2020-09-27:
After reapplying my 0.54 backup configuration, all is fine.
[image]

Ysurac, 2020-09-27:
There is no reason that gateway doesn't answer to ping from the router.
You can check route config on the router via SSH with ```ip r``` and network config with ```ip a```

closed 2021-01-04 by github-actions[bot]