Ysurac/openmptcprouter #1255: Configuration VPN

JeedHome44, 2020-09-27:
Bonjour,

J’ai commandé hier un VPS OVH. J’ai reçu les identifiants dans la soirée. J’ai pu y configurer sans problème Openmctprouteur.

J’ai mis aussi sur un Pi 4B 2 Go l’image du logiciel Openmctprouteur. Je vais le configurer aujourd’hui.

J’ai vu qu’il était possible de mettre un VPN sur le VPS OVH. Je n’ai jamais configuré de VPN, surtout sur un VPS. C’est une grande première pour moi.

J’aimerai donc savoir comment configurer ce VPN pour que mes connexions soit basées hors de France. 

On peut choisir et changer facilement de changer de pays avec un VPN? On peut aussi facilement l’activer ou le désactiver ou pas?

OpenVPN est un service de VPN ou juste un logiciel qui permet d’y lier un VPN qu’on choisit après?

Quel prix mensuel pour un VPN?

Netflix, Spotify, Disney+... marche avec VPN?

Désolé pour toutes ces questions mais je préfère bien me renseigner avant de faire quelque chose qui me bloquerait ensuite.

kevin39, 2020-09-27:
Bonjour,
Ces demandes sont hors du périmètre d'assistance d'OpenMTCPRouter.

Le VPS ne sert, dans l'idéal, qu'à l'agrégation de tes connexions entre ton domicile et internet. Chacun de tes box ouvre une connexion vers le VPS (OVH dans ton cas).

Si tu souhaites par dessus rajouter un VPN sur le VPS, cela risque de te causer des ennuis si tu n'es pas administrateur système, donc je ne vais pas rentrer dans ces détails.

Si tu souhaites utiliser un VPN pour avoir une sortie internet hors de France, rien t'empeche d'utiliser un VPN grand public sur ton ordinateur (NordVPN par exemple ou n'importe quel concurrent). Ce VPN passera à travers OMR sans problème et bénéficiera de l'agrégation de tes connexions.

Pour répondre plus précisement à chacune des questions : 

> J’aimerai donc savoir comment configurer ce VPN pour que mes connexions soit basées hors de France.

Le script d'installation du VPS configure tout automatiquement et je te déconseille de rajouter (côté VPS) une VPN.
Si tu souhaites délocaliser ta sortie Internet tu as 2 solutions : 
- Soit tu choisis un VPS OVH dans un autre pays
- Soit tu mets le VPN sur ton ordinateur (et seul ton ordinateur bénéficiera de ce VPN)

> On peut choisir et changer facilement de changer de pays avec un VPN? On peut aussi facilement l’activer ou le désactiver ou pas?

Ca dépend de ton fournisseur VPN et c'est pas lié à OpenMTCPRouter.

> OpenVPN est un service de VPN ou juste un logiciel qui permet d’y lier un VPN qu’on choisit après?

OpenVPN est un outil libre. Ce n'est pas un fournisseur de VPN en effet. 

 > Quel prix mensuel pour un VPN?

Ca dépend du fournisseur.

> Netflix, Spotify, Disney+... marche avec VPN?

Ca dépend du fournisseur.

JeedHome44, 2020-09-27:
Ah d'accord.
J'avais donc bien mal compris comment utiliser un VPN.
Donc pas besoin de VPN sur le VPS.
J'ai juste besoin de prendre un abonnement VPN chez un grand du nom et de l'installer sur mon PC.

Merci beaucoup pour tes réponses rapides. J'ai juste un problème avec le boot sur le Pi la mais j'ai demandé sur l'autre sujet déjà ouvert.

JeedHome44, 2020-09-27:
Bon, le Pi doit avoir un vrai défaut d'alimentation.
J'ai voulu tester de lui mettre un hub USB alimenté sur l'un de ses ports USB 3.0 pour lui donner un coup de boost en plus.
Si le Hub USB est branché quand j'allume le Pi, celui-ci ne s'allume même pas et le ventilateur ne tourne pas... jamais vu ça...
Désolé encore je pense donc bien que c'est ce Pi 4 qui doit avoir un défaut.

closed 2021-01-04 by github-actions[bot]