Ysurac/openmptcprouter #1256: vmap allocation for size <n> failed: use vmalloc=<size> to increase

cr3ative, 2020-09-27:
## Expected Behavior

Tidier dmesg output :)

## Current Behavior

At the moment, when I run `dmesg`, I get a lot of vmalloc failures. Here's a snippet:

```
[857958.982070] vmap allocation for size 36864 failed: use vmalloc=<size> to increase size
[857962.292302] vmap allocation for size 36864 failed: use vmalloc=<size> to increase size
[857965.532131] vmap allocation for size 36864 failed: use vmalloc=<size> to increase size
[857968.782684] vmap allocation for size 36864 failed: use vmalloc=<size> to increase size
[857968.790990] warn_alloc: 3 callbacks suppressed
[857968.790997] tracebox: vmalloc: allocation failure: 32768 bytes, mode:0xcc0(GFP_KERNEL), nodemask=(null),cpuset=/,mems_allowed=0
[857968.807121] CPU: 0 PID: 23622 Comm: tracebox Tainted: G         C        5.4.58 #0
[857968.814775] Hardware name: BCM2711
[857968.818275] [<c020fe6c>] (unwind_backtrace) from [<c020b338>] (show_stack+0x10/0x14)
[857968.826105] [<c020b338>] (show_stack) from [<c082602c>] (dump_stack+0xa4/0xc4)
[857968.833415] [<c082602c>] (dump_stack) from [<c0324df4>] (warn_alloc+0xb4/0x13c)
[857968.840810] [<c0324df4>] (warn_alloc) from [<c0320340>] (__vmalloc_node_range+0x1b0/0x294)
[857968.849159] [<c0320340>] (__vmalloc_node_range) from [<c020c108>] (module_alloc+0x44/0x4c)
[857968.857509] [<c020c108>] (module_alloc) from [<c02b8348>] (bpf_jit_binary_alloc+0x6c/0x104)
[857968.865947] [<c02b8348>] (bpf_jit_binary_alloc) from [<c021a174>] (bpf_int_jit_compile+0x138/0x314)
[857968.875078] [<c021a174>] (bpf_int_jit_compile) from [<c02b94d8>] (bpf_prog_select_runtime+0x100/0x1ac)
[857968.884472] [<c02b94d8>] (bpf_prog_select_runtime) from [<c06ee19c>] (bpf_prepare_filter+0x79c/0x858)
[857968.893778] [<c06ee19c>] (bpf_prepare_filter) from [<c06ee608>] (sk_attach_filter+0xc/0x44)
[857968.902216] [<c06ee608>] (sk_attach_filter) from [<c06a896c>] (sock_setsockopt+0xbd8/0xc68)
[857968.910653] [<c06a896c>] (sock_setsockopt) from [<c06a1b50>] (__sys_setsockopt+0x158/0x168)
[857968.919088] [<c06a1b50>] (__sys_setsockopt) from [<c0201000>] (ret_fast_syscall+0x0/0x4c)
[857968.927348] Exception stack(0xe21effa8 to 0xe21efff0)
[857968.932481] ffa0:                   00000008 00000000 00000005 00000001 0000001a bef5e3b0
[857968.940741] ffc0: 00000008 00000000 00fb4c98 00000126 00000001 0000000b 00000006 00000c30
[857968.949001] ffe0: b6d6bf10 bef5e38c b6d43934 b6edc644
[857968.954141] Mem-Info:
[857968.956501] active_anon:3425 inactive_anon:76 isolated_anon:0
[857968.956501]  active_file:12594 inactive_file:8906 isolated_file:0
[857968.956501]  unevictable:0 dirty:0 writeback:0 unstable:0
[857968.956501]  slab_reclaimable:2526 slab_unreclaimable:6368
[857968.956501]  mapped:5704 shmem:516 pagetables:174 bounce:0
[857968.956501]  free:940663 free_pcp:1611 free_cma:16009
[857968.990221] Node 0 active_anon:13700kB inactive_anon:304kB active_file:50376kB inactive_file:35624kB unevictable:0kB isolated(anon):0kB isolated(file):0kB mapped:22816kB dirty:0kB writeback:0kB shmem:2064kB writeback_tmp:0kB unstable:0kB all_unreclaimable? no
[857969.013262] DMA free:604736kB min:16384kB low:20480kB high:24576kB active_anon:0kB inactive_anon:0kB active_file:4676kB inactive_file:5940kB unevictable:0kB writepending:0kB present:786432kB managed:660804kB mlocked:0kB kernel_stack:1056kB pagetables:0kB bounce:0kB free_pcp:1968kB local_pcp:176kB free_cma:64036kB
[857969.041070] lowmem_reserve[]: 0 0 3188 3188
[857969.045343] HighMem free:3157916kB min:512kB low:20744kB high:40976kB active_anon:13700kB inactive_anon:304kB active_file:45700kB inactive_file:29684kB unevictable:0kB writepending:0kB present:3264512kB managed:3264512kB mlocked:0kB kernel_stack:0kB pagetables:696kB bounce:0kB free_pcp:4476kB local_pcp:548kB free_cma:0kB
[857969.073847] lowmem_reserve[]: 0 0 0 0
[857969.077593] DMA: 54*4kB (UMC) 67*8kB (UME) 33*16kB (UMEC) 3*32kB (EC) 6*64kB (UMC) 12*128kB (UMEC) 13*256kB (UMEC) 10*512kB (UME) 11*1024kB (UME) 8*2048kB (UMC) 138*4096kB (UMC) = 604640kB
[857969.094477] HighMem: 385*4kB (UM) 189*8kB (UM) 85*16kB (UM) 47*32kB (UM) 14*64kB (UM) 2*128kB (UM) 2*256kB (UM) 1*512kB (U) 2*1024kB (UM) 1*2048kB (U) 768*4096kB (M) = 3157916kB
[857969.110404] 19558 total pagecache pages
[857969.114323] 1012736 pages RAM
[857969.117370] 816128 pages HighMem/MovableOnly
[857969.121723] 31407 pages reserved
[857969.125031] 16384 pages cma reserved
[857972.302407] vmap allocation for size 36864 failed: use vmalloc=<size> to increase size
```

## Possible Solution

Unknown

## Steps to Reproduce the Problem

  1. Boot on Raspberry Pi 4B+, using the v0.55.3 r0+14198-da98603597 64-bit image
  2. Leave running for some time
  3. Observe vmap allocation failures

## Context (Environment)

I only went looking because "Shadowsocks is not running" was on my info screen.

## Specifications

  - OpenMPTCProuter version: v0.55.3 r0+14198-da98603597
  - OpenMPTCProuter VPS version: 0.1017
  - OpenMPTCProuter platform: RPI4B+

Ysurac, 2020-09-28:
Did you install any additional modules or do you use USB ?

cr3ative, 2020-09-28:
No USB devices - 

```
root@OpenMPTCProuter  ~ lsusb
Bus 001 Device 001: ID 1d6b:0002
Bus 001 Device 002: ID 2109:3431
Bus 002 Device 001: ID 1d6b:0003
```

Those are all the onboard RPi controllers for USB, rather than plugged in devices

Modules, I'm unsure how to check - I think installing/configuring NextDNS is the only major change made to the image. Let me know a command to run and I can get you that information.

cr3ative, 2020-09-28:
Hm, sorry, I think something must be wrong with my setup. I checked `/etc/opkg/distfeeds.conf` and my installation is aimed at RPI2 rather than RPI4:

```
root@OpenMPTCProuter  opkg cat distfeeds.conf 
src/gz openmptcprouter_core http://download.openmptcprouter.com:80/release/v0.55.3/rpi2/targets/bcm27xx/bcm2709/packages
src/gz openmptcprouter_base http://download.openmptcprouter.com:80/release/v0.55.3/rpi2/packages/arm_cortex-a7_neon-vfpv4/base
src/gz openmptcprouter_luci http://download.openmptcprouter.com:80/release/v0.55.3/rpi2/packages/arm_cortex-a7_neon-vfpv4/luci
src/gz openmptcprouter_openmptcprouter http://download.openmptcprouter.com:80/release/v0.55.3/rpi2/packages/arm_cortex-a7_neon-vfpv4/openmptcprouter
src/gz openmptcprouter_packages http://download.openmptcprouter.com:80/release/v0.55.3/rpi2/packages/arm_cortex-a7_neon-vfpv4/packages
```

I'll redownload and reimage my RPI.

closed 2020-09-28 by cr3ative

Ysurac, 2020-09-28:
Because it's the 32bits image that is the same for all RPI.
The 64bits image is dedicated to RPI4.

moham96, 2021-11-20:
I'm facing the same problem where upon bootup the wifi doesn't work because the kernel fails to load the wifi kernel module.
According to [this](https://forum.openwrt.org/t/vmap-allocation-for-size-442368-failed-use-vmalloc-size-to-increase-size/34545/6) enabling `CONFIG_ARM_MODULE_PLTS=y` in the kernel config should resolve the problem, haven't tried it but i hope @Ysurac  would enable it in new releases to test

santosh07bec, 2022-01-17:
@Ysurac I have the same issue on my 32 bit RSP 4B version. The wifi driver randomly loads, but most of the times it doesn't (and prints below error and stack trace). Is there any plan to address this?
Also, is there any workaround available for this? It would be really great for newbie like myself if you could provide any workaround in the mean time.

```
[  426.051053] vmap allocation for size 245760 failed: use vmalloc=<size> to increase size
[  426.060064] modprobe: vmalloc: allocation failure: 241664 bytes, mode:0xcc0(GFP_KERNEL), nodemask=(null),cpuset=/,mems_allowed=0
[  426.072618] CPU: 3 PID: 15617 Comm: modprobe Tainted: G         C        5.4.106 #0
[  426.081239] Hardware name: BCM2711
[  426.085118] [<c020fe84>] (unwind_backtrace) from [<c020b360>] (show_stack+0x10/0x14)
[  426.093808] [<c020b360>] (show_stack) from [<c08446b8>] (dump_stack+0xa4/0xc4)
[  426.101987] [<c08446b8>] (dump_stack) from [<c0326460>] (warn_alloc+0xb4/0x13c)
[  426.110260] [<c0326460>] (warn_alloc) from [<c032194c>] (__vmalloc_node_range+0x1b0/0x294)
[  426.119513] [<c032194c>] (__vmalloc_node_range) from [<c020c130>] (module_alloc+0x44/0x4c)
[  426.128766] [<c020c130>] (module_alloc) from [<c029d0cc>] (load_module+0x944/0x2254)
[  426.137498] [<c029d0cc>] (load_module) from [<c029eb1c>] (sys_init_module+0x140/0x160)
[  426.146431] [<c029eb1c>] (sys_init_module) from [<c0201000>] (ret_fast_syscall+0x0/0x4c)
[  426.155581] Exception stack(0xe3573fa8 to 0xe3573ff0)
[  426.161178] 3fa0:                   00000000 00000000 b6e67010 0004c89c 000129d1 0000b302
[  426.170437] 3fc0: 00000000 00000000 00000003 00000080 0004c89c 00000000 013b8d10 00000000
[  426.179689] 3fe0: beac4bf4 beac4bd8 00011e14 b6f18ce8
[  426.185291] Mem-Info:
[  426.188108] active_anon:2797 inactive_anon:87 isolated_anon:0
[  426.188108]  active_file:8051 inactive_file:5446 isolated_file:0
[  426.188108]  unevictable:0 dirty:0 writeback:0 unstable:0
[  426.188108]  slab_reclaimable:1824 slab_unreclaimable:4528
[  426.188108]  mapped:3759 shmem:174 pagetables:151 bounce:0
[  426.188108]  free:950836 free_pcp:1432 free_cma:14202
[  426.224455] Node 0 active_anon:11188kB inactive_anon:348kB active_file:32204kB inactive_file:21784kB unevictable:0kB isolated(anon):0kB isolated(file):0kB mapped:15036kB dirty:0kB writeback:0kB shmem:696kB writeback_tmp:0kB unstable:0kB all_unreclaimable? no
[  426.249370] DMA free:616700kB min:16384kB low:20480kB high:24576kB active_anon:0kB inactive_anon:0kB active_file:4208kB inactive_file:592kB unevictable:0kB writepending:0kB present:786432kB managed:660804kB mlocked:0kB kernel_stack:912kB pagetables:0kB bounce:0kB free_pcp:1852kB local_pcp:624kB free_cma:56808kB
[  426.278935] lowmem_reserve[]: 0 0 3188 3188
[  426.283615] HighMem free:3186644kB min:512kB low:20744kB high:40976kB active_anon:11188kB inactive_anon:348kB active_file:27996kB inactive_file:21192kB unevictable:0kB writepending:0kB present:3264512kB managed:3264512kB mlocked:0kB kernel_stack:0kB pagetables:604kB bounce:0kB free_pcp:3876kB local_pcp:844kB free_cma:0kB
[  426.314458] lowmem_reserve[]: 0 0 0 0
[  426.318585] DMA: 91*4kB (UMC) 76*8kB (UME) 21*16kB (UMEC) 11*32kB (UEC) 4*64kB (UEC) 7*128kB (UMEC) 8*256kB (UM) 7*512kB (UME) 6*1024kB (UMC) 4*2048kB (UMC) 145*4096kB (MC) = 616700kB
[  426.336325] HighMem: 185*4kB (UM) 134*8kB (UM) 56*16kB (UM) 28*32kB (UM) 7*64kB (U) 4*128kB (UM) 2*256kB (U) 2*512kB (UM) 0*1024kB 1*2048kB (U) 776*4096kB (M) = 3186644kB
[  426.352981] 13670 total pagecache pages
[  426.357296] 1012736 pages RAM
[  426.360744] 816128 pages HighMem/MovableOnly
[  426.365497] 31407 pages reserved
[  426.369199] 16384 pages cma reserved
```

moham96, 2022-01-17:
The main issue is that the kernel for the raspberry pi platform load a
plethora of unneeded drivers like sound drivers and usb wireless
drivers(which should only be loaded if the user wants to connect such
adapter)
This is causing the kernel to quickly run out of memory allocated for the
drivers and sometimes it runs out before it loads the brcmfmac driver which
is for the built in wireless.
The solution in my opinion is to remove the plethora of files in
/etc/modules.d/ (or comment them) to prevent the kernel from loading the
huge number of drivers and the user can enable a said driver at will if
needed.

Regards

frank087234, 2026-10-29:
root@sf8008:~# cat /proc/meminfo
MemTotal:         948824 kB
MemFree:          304444 kB
MemAvailable:     706648 kB
Buffers:          121428 kB
Cached:            69932 kB
SwapCached:         1360 kB
Active:           122312 kB
Inactive:         145816 kB
Active(anon):      37264 kB
Inactive(anon):    40628 kB
Active(file):      85048 kB
Inactive(file):   105188 kB
Unevictable:           0 kB
Mlocked:               0 kB
HighTotal:        441848 kB
HighFree:         292832 kB
LowTotal:         506976 kB
LowFree:           11612 kB
SwapTotal:        524284 kB
SwapFree:         512968 kB
Dirty:                32 kB
Writeback:             0 kB
AnonPages:         76432 kB
Mapped:            41380 kB
Shmem:              1124 kB
Slab:             264580 kB
SReclaimable:     249708 kB
SUnreclaim:        14872 kB
KernelStack:        2904 kB
PageTables:         1856 kB
NFS_Unstable:          0 kB
Bounce:                0 kB
WritebackTmp:          0 kB
CommitLimit:      998696 kB
Committed_AS:     550156 kB
VmallocTotal:     507904 kB
VmallocUsed:           0 kB
VmallocChunk:          0 kB
CmaTotal:          77824 kB
CmaFree:               0 kB
root@sf8008:~#