Ysurac/openmptcprouter #1258: 302 Moved Temporarily Location : params.vpn

Didyme33, 2020-09-28:
## Expected Behavior
Mise à jour complétée.

## Current Behavior
Part en boucle lors de la mise à jour.

## Steps to Reproduce the Problem

  1. Update the VPS part (Debian 10)
  2. : HTTP request sent, awaiting response... 302 Moved Temporarily
Location: https://www.openmptcprouter.com/server/shorewall6/params.vpn [following]
20 redirections exceeded.

Le message s'affiche une bonne dizaine de fois, et le script s'arrête.

## Specifications

  - OpenMPTCProuter version: 0.55.3
  - OpenMPTCProuter VPS version: 0.1015  ---> 0.1017
  - OpenMPTCProuter platform:RPI3

gwarf, 2020-09-28:
Tried to report the same error on IRC this week end.
New VPS install works fine, but update is broken.

Ysurac, 2020-09-28:
Fixed. This was a CDN problem.

closed 2020-09-28 by Ysurac