Ysurac/openmptcprouter #1259: Changement des ports des services OpenMPTCProuter

poti56, 2020-09-28:
Bonsoir,

J'ai installé un VPS sur une machine fixe chez moi, qui se trouve derrière une Box SFR.
Malheureusement, les ports utilisés par OpenMPTCProuter (65***) sont réservés sur ma box, donc je n'arrive pas à ouvrir et rediriger correctement les ports de ma box. J'ai pensé du coup utiliser les ports 5*** à la place. Au niveau de la box, ça fonctionne bien, j'arrive par exemple à me connecter en ssh à ma Debian spécifiquement installée pour le VPS via le port 5222.

J'aimerais savoir comment faire les réglages adaptés sur le routeur pour que les données soient envoyées sur les ports 5*** de ma box, et non sur les ports 65***. Pouvez-vous svp me guider ? Je suis en train de découvrir ce type de configurations et ne suis pas très à l'aise pour le moment.

En vous remerciant, bien cordialement,

Patricia.

kevin39, 2020-09-28:
Bonsoir,
La partie VPS ne doit pas être chez toi mais chez un prestataire tiers. Sinon ça servira à rien pour tester l’agrégation ;)

poti56, 2020-09-28:
Bonsoir,
Ah mais j'agrège ailleurs que chez moi en plaçant le VPS chez moi.
Patricia.

Ysurac, 2020-09-29:
C'est possible de changer les ports normalement (je n'ai jamais testé), par contre ce n'est pas évident. Il faut modifier sur le VPS manuellement la configuration du/des VPNs, de Shadowsocks, de l'API et du firewall. Il faut aussi modifier le port de l'API sur le routeur.

closed 2021-01-04 by github-actions[bot]